esphome/issues #5067: Wrong id in Prometheus metrics

LinFor, 2023-11-02:
### The problem

Prometheus export ignores sensor ids, using sensor names instead.

For example, i declare some sensor with different id (**test_heater_1**) and name (**dallas 08**):
```
  - platform: dallas
    id: test_heater_1
    address: 0xbc3ce1d44463fd28
    name: "dallas 08"
    resolution: 12
    accuracy_decimals: 2
```

But on /metrics endpoint it appears with different id (**dallas_08**), probably generated from name:
```
esphome_sensor_failed{id="dallas_08",name="dallas 08"} 0
esphome_sensor_value{id="dallas_08",name="dallas 08",unit="°C"} 31.19
```

### Which version of ESPHome has the issue?

2023.10.1 Nov 2 2023, 14:26:06

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

none

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

prometheus

### Example YAML snippet

```yaml
prometheus:
  include_internal: true
sensor:
  - platform: dallas
    id: test_heater_1
    address: 0xbc3ce1d44463fd28
    name: "dallas 08"
    resolution: 12
    accuracy_decimals: 2
```

### Anything in the logs that might be useful for us?

```txt
esphome_sensor_failed{id="dallas_08",name="dallas 08"} 0
esphome_sensor_value{id="dallas_08",name="dallas 08",unit="°C"} 31.19
```

jesserockz, 2023-11-02:
Please read the documentation =) https://esphome.io/components/prometheus#prometheus-relabel

closed 2023-11-02 by jesserockz

LinFor, 2023-11-03:
Hello. I know about relabelling, and it seems that it is not needed in my task, although yes, I know that with its help the problem can be solved (at the cost of complexity in the configuration). Just in case - in the example above relabelling is not configured, these are not its effects.

However, the documentation states that

> ESPHome allows you to do some basic relabeling of Prometheus metrics. This is useful **if you want to have different metric names or IDs** than those shown in Home Assistant or the web interface.

I don't want different IDs, I want to see exactly the same ones that are specified in the sensor configuration.

If the behavior of ignoring an explicitly specified sensor ID is considered correct, let’s at least clearly reflect this in the documentation.