esphome/issues #5069: Pause does not work in i2s media player

7h30n3, 2023-11-03:
### The problem

I think I have the same issue as described in #4447 however the merged PR esphome/esphome#4796 didn't fix the problem for me.

Playing a webstream (e.g. mp3) and pausing it renders the media player unusable until restart. Meaning you can select other sources (as it can be seen in the log), but the media player doesn't play any sound anymore, not even (local) TTS.

The issue first arises with ESPhome 2023.4.0

### Which version of ESPHome has the issue?

2023.10.6

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

i2s

### Example YAML snippet

```yaml
i2s_audio:
  i2s_lrclk_pin: 17
  i2s_bclk_pin: 18

media_player:
  - platform: i2s_audio
    name: "Media Player"
    dac_type: external
    i2s_dout_pin: 19
    mode: mono
```

### Anything in the logs that might be useful for us?

```txt
[23:16:11][D][media_player:059]: 'ESP32 1 Media Player' - Setting
[23:16:11][D][media_player:066]:   Media URL: http://streams.radiobob.de/bob-national/mp3-192/mediaplayer
[23:16:13][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.97 s).
[23:16:13][W][component:215]: Components should block for at most 20-30ms.
[23:16:32][D][media_player:059]: 'ESP32 1 Media Player' - Setting
[23:16:32][D][media_player:063]:   Command: PAUSE
[23:16:49][D][media_player:059]: 'ESP32 1 Media Player' - Setting
[23:16:49][D][media_player:066]:   Media URL: https://listener2.mp3.tb-group.fm/clt.mp3
```

mondbaron, 2023-11-26:
Same issue here. After pausing a radio stream, playing it again or any other source (e.g. TTS) is not possible until a device restart. 

Additional input:
- The ESPHome device receives a "PLAY" command but does not react (`Command: PLAY` in the debug logs)
- The `on_play` section of the ESPHome config is not triggered, I did place a `logger.log: "Playback started"` there, but nothing appears in the logs
- Pause and continue of a TTS file works and triggers `on_pause` and `on_play` correctly multiple times

I'm running ESPHome 2023.11.4

YahyaAlfayad, 2023-12-23:
I am having the same issue. Once radio stream is paused, nothing works anymore till I disconnect the esp32 from electricity and connect it again

stewe93, 2024-03-20:
2024.3.0

Same issue here, if you need any other info I would happily provide it to solve this.

closed 2024-03-31 by jesserockz