esphome/issues #5076: Pi Pico W bootloops when connecting to a WPA3 secured network

Honza-m, 2023-11-05:
### The problem

When trying to connect to a WPA3 secured network, the Pico briefly connects but immediately bootloops, and prints out "serial port closed" in the logs. This happens on the most barebone .yaml file. I've created a guest WPA2 network and the pico connects just fine.

When I flashed MicroPython to the same board, I was able to connect and maintain connection to the WPA3 network, so I thought this was not the problem, but it turned out it was. If WPA3 is indeed not supported on this board with ESPHome, it would be nice if the firmware printed out an error or refused to connect altogether, as the bootloop was very confusing. Or if this is not easy to do, maybe I can raise a documentation MR to document that this is not supported to let others who will inevitably try this know.

### Which version of ESPHome has the issue?

2023.10.6

### What type of installation are you using?

pip

### What platform are you using?

RP2040

### Board

Pi Pico W

### Example YAML snippet

```yaml
esphome:
  name: pico
  friendly_name: pico

rp2040:
  board: rpipicow
  framework:
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

thenextbutton, 2023-11-06:
Hi,

I am running a PiCOW connecting to an SSID on WPA3, I have not experienced that issue.

under my WiFi section I have:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  power_save_mode: none

Honza-m, 2023-11-06:
So I've changed the wifi credentials back to my WPA3 network, applied the `power_save_mode: none` setting, and it worked! The issue is, I can't get it back to the broken state! Whathever settings I use that I was using before now work and the device does not bootloop anymore...

I have two theories. I've either messed up my wifi secrets on the first try (although I did triple check), or something changed on my network (my router did need to restart to create a WPA2 guest network). Either way, thank you for pointing out that it is possible to run the pico on a WPA3 network with ESPHome, works like a charm now.

closed 2023-11-06 by Honza-m