esphome/issues #5077: Unable to compile code with last Pillow 10.1.0

homiodev, 2023-11-05:
### The problem

Error: please update your pillow installation to 10.0.1. (pip install "pillow==10.0.1") while pip install pillow and it has 
10.1.0

### Which version of ESPHome has the issue?

2023.10.6

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

geerlingguy, 2023-11-12:
Same here...

```
$ pip3 freeze
...
Pillow==10.1.0
```

But getting:

```
  Please update your pillow installation to 10.0.1. (pip install "pillow==10.0.1").
```

geerlingguy, 2023-11-12:
If I explicitly install 10.0.1 instead of 10.1.0, it works. Is it an incompatibility with future versions of the library?

JacksonBarker, 2023-11-15:
This has been an issue for me as well, since Arch now uses pacman for managing python dependencies and pinning a pacman package is not recommended.

hostcc, 2023-11-21:
Originally the PR https://github.com/esphome/esphome/pull/5489 has not-less-then version constraint to accommodate for future package version, but then it has been decided to pin the exact version, similar to how it is done in [requirements.txt](https://github.com/esphome/esphome/blob/dev/requirements.txt). Most reasonable explanation would be predictable installs, though I do see a rationale behind the extra package having relaxed constraints.
Will cut another PR and let's hear from @jesserockz ;)

jesserockz, 2023-11-21:
If not using docker which has the correct versions installed, You should be installing ESPHome using a dedicated python venv. This is the only supported installation method and we are not going to relax the pin so other methods work as it causes even more issues for us as maintainers.

As commented on the PR, we welcome bumps to the dependency versions if its tested to work.

closed 2023-11-23 by jesserockz

Heiner1967, 2024-01-07:
Hi,

whilst on my old ESPHOME pillow 10.0.1 worked, i updated now to ESPHOME 2023.12.5 which requested me afterward to update pillow to 10.1.0 which i did.

However ESPHOME still state that pillow 10.1.0 is needed even i did install it successfully as mentioned.
Even a "pip show pillow" shows that 10.1.0 is installed.

Reboot of my Rasberry PI is not bringing any change.

ssieb, 2024-01-07:
Come ask for help on the esphome discord server.

hostcc, 2024-01-08:
> whilst on my old ESPHOME pillow 10.0.1 worked, i updated now to ESPHOME 2023.12.5 which requested me afterward to update pillow to 10.1.0 which i did.
>
> However ESPHOME still state that pillow 10.1.0 [...]

@Heiner1967 , if you have multiple Python versions installed on the system it might be that the `pillow` package got installed into one differs from what ESPHome uses. Under the hood the usage of Pillow, and version check in particular, is built upon standard Python import mechanism - the fragment below essentially mimics that:
```
$ python3
>>> import PIL
>>> PIL.__version__
'10.1.0'
```
You can try it yourself and verify if the Pillow version matches the expected one.

To isolate the issue from a potential multiple Python versions symptom you could use Python virtual environment, which gives you predictable Python version and set of packages, for example the fragment below will create such under `.venv` directory, install latest ESPhome and attempt compiling your configration:
```
$ cd <directory with ESPHome configuration(s) 
$ python3 -m venv .venv
$ . .venv/bin/activate
$ pip install esphome
$ esphome compile <your ESPHome configuration>
```

Heiner1967, 2024-01-09:
Thanks, great thought,  I tried:

$ python3
>>> import PIL
>>> PIL.__version__
which bring me the correct result '10.1.0'

Nowi strigle to find the folder for ESPHOME......
neither i know how to check if other Pyton instances are installed, but in principle I am fine to remove others if there are any.

Heiner1967, 2024-01-09:
stupid me i installed pillow without "sudo" upfront.
it seems to work and even `pip3 show pillow` told me the right Version is there which is not the case.

So `sudo pip3 install pillow==10.1.0` did the trick

ssieb, 2024-01-09:
You generally should never run pip as root.  It's a good way to mess up your system.