esphome/issues #5078: ESPCam : Setup Failed: ERROR

upamanyudas, 2023-11-05:
### The problem

Getting the ESPCam Setup Failed: ERROR again with the latest update. 
Everything else is fine with the device, connects to the WiFi and the flashlight on the board works too.
This has happened a couple of times in the past and has been solved by subsequent updates. Something went wrong again with the latest update.

### Which version of ESPHome has the issue?

2023.10.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.0

### What platform are you using?

ESP32

### Board

ESP32-CAM

### Component causing the issue

esp32_camera

### Example YAML snippet

```yaml
esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
```

### Anything in the logs that might be useful for us?

```txt
[10:14:30][E][esp32_camera:123]: Setup Failed: ERROR
```

twsl, 2023-12-18:
altough i have a different hpio config, i encoutered the same issue with a m5cam dev board. the camera used to work in the past but some update and/or config change broke it

randybb, 2024-02-14:
Can confirm. With ESP-IDF it throws that error, with Arduino it works, but for this module I needed to remove `power_down_pin:` and pull-down that GPIO32 manually to get it working, otherwise it throws `[W][esp32_camera:190]: Got invalid frame from camera!`.

RaveGun, 2025-02-09:
I did not updated my camera in the last 3 years (it is a TTGO T-Journal ESP32 Camera) and it worked fine.
No I want to repurpose the camera to add some additional features an I just recompiled, with the latests esphome I have on the computer (Version: 2024.12.4), the same yaml file.

After the upload I am getting this errors:
```
[11:20:48][E][esp32_camera:024]: esp_camera_init failed: ESP_FAIL
[11:20:48][E][component:119]: Component esp32_camera was marked as failed.
[11:20:48][E][component:164]: Component esp32_camera set Error flag: unspecified
```

On the other hand I just downloaded the official Platformio examples and flashed the device with the webserver example. The device works and I streams images.

How can I roll back to an older version IDF, Esphome?

RaveGun, 2025-02-09:
Oh, now I've seen something new from the serial log:
```
[14:11:18][D][esp-idf:000]: E (463) cam_hal: cam_dma_config(306): frame buffer malloc failed
[14:11:18][D][esp-idf:000]: E (463) cam_hal: cam_config(390): cam_dma_config failed
[14:11:18][D][esp-idf:000]: E (473) camera: Camera config failed with error 0xffffffff

```