esphome/issues #5082: `remote_transmitter` crashing ESP on boot

Steve-Tech, 2023-11-07:
### The problem

Hi, I'm trying to make an IR ceiling light smart with an ESP-01, but it seems to hang for a couple seconds after scanning for WiFi networks, and then gets reset by its watchdog. Safe mode works fine, and it works fine without the `remote_transmitter` stuff in the YAML file, so I'm pretty sure the board is fine.

### Which version of ESPHome has the issue?

2023.10.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.1

### What platform are you using?

ESP8266

### Board

ESP-01S

### Component causing the issue

remote_receiver

### Example YAML snippet

```yaml
esphome:
  name: alexs-room-esp
  friendly_name: Alex's Room ESP

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "REMOVED"

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

remote_transmitter:
  pin: GPIO0
  carrier_duty_percent: 50%

output:
  - platform: template
    id: light1
    type: binary
    write_action:
      - output.turn_on:
          id: light1
      - output.turn_off:
          id: light1

light:
  - platform: binary
    name: "Alex's Bedroom Light"
    output: light1
    on_turn_on:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [1125, -375, 1125, -375, 375, -1125, 1125, -375, 1125, -375, 375, -1125, 1125, -375, 375, -1125, 375, -1125, 375, -1125, 375, -1125, 375, -1125]
    on_turn_off:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [1125, -375, 1125, -375, 375, -1125, 1125, -375, 1125, -375, 375, -1125, 375, -1125, 1125, -375, 375, -1125, 375, -1125, 375, -1125, 375, -1125]
```

### Anything in the logs that might be useful for us?

```txt
[I][logger:326]: Log initialized
[C][ota:473]: There have been 9 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[C][light:035]: Setting up light 'Alex's Bedroom Light'...
[D][light:036]: 'Alex's Bedroom Light' Setting:
[D][light:041]:   Color mode: 
[D][remote_transmitter:075]: Sending remote code...
[C][wifi:038]: Setting up WiFi...
[C][wifi:051]: Starting WiFi...
[C][wifi:052]:   Local MAC: D8:BF:C0:0E:E3:51
[D][wifi:428]: Starting scan...

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00072f10
~ld
```

ssieb, 2023-11-07:
The remote transmitter isn't initialized before the light calls it.  Try adding `setup_priority: 599` to the light.

Steve-Tech, 2023-11-07:
Like this? It says invalid option.

```yaml
light:
  - platform: binary
    setup_priority: 599
    name: "Alex's Bedroom Light"
    output: light1
    setup_priority: 599
    on_turn_on:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [1125, -375, 1125, -375, 375, -1125, 1125, -375, 1125, -375, 375, -1125, 1125, -375, 375, -1125, 375, -1125, 375, -1125, 375, -1125, 375, -1125]
    on_turn_off:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [1125, -375, 1125, -375, 375, -1125, 1125, -375, 1125, -375, 375, -1125, 375, -1125, 1125, -375, 375, -1125, 375, -1125, 375, -1125, 375, -1125]
```

I can add it to `remote_transmitter`, but it still has the issue.

ssieb, 2023-11-07:
ok, then add `setup_priority: 601` to the transmitter instead.

Steve-Tech, 2023-11-07:
It still does the same thing.
```yaml
remote_transmitter:
  setup_priority: 601
  pin: GPIO0
  carrier_duty_percent: 50%
```

ssieb, 2023-11-07:
Can you please come to the esphome discord server?  It will be a lot easier to do this there.

Steve-Tech, 2023-11-07:
Issue is fixed, I created an infinite loop in the `output` section.
Working config:
```yaml
esphome:
  name: alexs-room-esp
  friendly_name: Alex's Room ESP

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "REMOVED"

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

remote_transmitter:
  pin: GPIO0
  carrier_duty_percent: 50%

output:
  - platform: template
    id: light1
    type: binary
    write_action:
      - delay: 0s

light:
  - platform: binary
    name: "Alex's Bedroom Light"
    output: light1
    on_turn_on:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [1125, -375, 1125, -375, 375, -1125, 1125, -375, 1125, -375, 375, -1125, 1125, -375, 375, -1125, 375, -1125, 375, -1125, 375, -1125, 375, -1125]
    on_turn_off:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [1125, -375, 1125, -375, 375, -1125, 1125, -375, 1125, -375, 375, -1125, 375, -1125, 1125, -375, 375, -1125, 375, -1125, 375, -1125, 375, -1125]
```

closed 2023-11-07 by Steve-Tech