esphome/issues #5083: Cannot compile with v2023.10.6

rafafuenza, 2023-11-07:
### The problem

No Issues before but with the last few updates I havent been able to compile this particular node.

`src/esphome/components/api/api_server.h:115:96: error: invalid use of '::'
   Trigger<std::string, std::string> *client_connected_trigger_ = new Trigger<std::string, std::string>();
                                                                                                ^
src/esphome/components/api/api_server.h:115:96: error: expected ';' at end of member declaration
src/esphome/components/api/api_server.h:115:102: error: expected unqualified-id before '>' token`

and the list of errors goes on... always with `api_server.h`.

### Which version of ESPHome has the issue?

2023.10.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.5

### What platform are you using?

ESP32

### Board

nodemcuv2

### Component causing the issue

api_server.h

### Example YAML snippet

```yaml
esphome:
  name: wordclock
  friendly_name: wordclock_1
  includes:
    - wordclock.h
  on_boot:
    priority: 800
    then:
      - light.turn_on: 
          id: fastledlight

esp8266:
  board: nodemcuv2
  framework:
    version: 2.7.4

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "qpyJxtctmirRbNR3SXDT1f/7P99zxrLq3Ue7o/tz5KA="

ota:
  password: "67506f30187199231f751e134f0e593a"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

[... 189 lines cut ...]
                 from src/esphome/components/api/subscribe_state.cpp:2:
src/esphome/components/api/api_server.h:115:96: error: invalid use of '::'
   Trigger<std::string, std::string> *client_connected_trigger_ = new Trigger<std::string, std::string>();
                                                                                                ^
src/esphome/components/api/api_server.h:115:96: error: expected ';' at end of member declaration
src/esphome/components/api/api_server.h:115:102: error: expected unqualified-id before '>' token
   Trigger<std::string, std::string> *client_connected_trigger_ = new Trigger<std::string, std::string>();
                                                                                                      ^
src/esphome/components/api/api_server.h:116:99: error: invalid use of '::'
   Trigger<std::string, std::string> *client_disconnected_trigger_ = new Trigger<std::string, std::string>();
                                                                                                   ^
src/esphome/components/api/api_server.h:116:99: error: expected ';' at end of member declaration
src/esphome/components/api/api_server.h:116:105: error: expected unqualified-id before '>' token
   Trigger<std::string, std::string> *client_disconnected_trigger_ = new Trigger<std::string, std::string>();
                                                                                                         ^
src/esphome/components/api/api_server.h:115:83: error: template argument 1 is invalid
   Trigger<std::string, std::string> *client_connected_trigger_ = new Trigger<std::string, std::string>();
                                                                                   ^
src/esphome/components/api/api_server.h:116:86: error: template argument 1 is invalid
   Trigger<std::string, std::string> *client_disconnected_trigger_ = new Trigger<std::string, std::string>();
                                                                                      ^
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [.pioenvs/wordclock/src/esphome/components/api/subscribe_state.cpp.o] Error 1
========================== [FAILED] Took 1.85 seconds ==========================
```

ssieb, 2023-11-07:
I expect that's because you're overriding the arduino platform version.  That version is not compatible.

rafafuenza, 2023-11-07:
> ing the arduino platform version. That version is not compatible.

Since the version 2023.10.x? It runs fine on 2023.9.3!

ssieb, 2023-11-07:
Things change.  There's a big warning right at the top of your compile log.

rafafuenza, 2023-11-07:
> Things change. There's a big warning right at the top of your compile log.

How can I fix it?

ssieb, 2023-11-07:
You have to remove the arduino version from your config.  If that makes fastled not work, then you can either try https://github.com/esphome/esphome/pull/4257 or switch to using neopixel instead.

rafafuenza, 2023-11-07:
> You have to remove the arduino version from your config. If that makes fastled not work, then you can either try [esphome/esphome#4257](https://github.com/esphome/esphome/pull/4257) or switch to using neopixel instead.

I will try to do that! thaks

closed 2023-11-08 by rafafuenza

sebasgra39, 2023-11-12:
@rafafuenza have you been able to solve this? I'm facing the same issue and https://github.com/esphome/esphome/pull/4257 didn't change anything. 
BR
Sebastian

nemeier, 2023-11-12:
I'm using fastled_clockless that requires 2.7.4, is this project dead?

ssieb, 2023-11-12:
@sebasgra39 what's the problem? Using that PR had to have changed something.  Come ask on discord if you need more help or else use the neopixel component instead.
@nemeier is what project dead?

Necrosaurus, 2024-03-05:
I had same problem, I had been stuck on framework: version: 2.7.4 for about a year, waiting for progress on fastled. 
Then after a recent esphome update, I couldn't install anything on my esp8266. so switched to neopixel, removed framework: version: 2.7.4 
and problem went away.