esphome/issues #5084: "Visit" link on ESPHome device does not load

jamespreedy, 2023-11-07:
### The problem

I have 23x ESP home devices. Depending on device type, some of them have options to "Visit", "Edit" or view "Logs".
Some simply have "Edit" and "Logs". This bug relates to those that show the option to "Visit" the device, which when clicked should launch the self hosted web page of the device.

For me, this doesn't work consistently. In this example, all devices are the same device - a LocalBytes ESPHome flashed power monitoring smart plug.

For some devices I click "Visit" and my browser opens a new window with the IP address of the device - the page loads successfully.

For other device I click "Visit" and my browser opens a new window with a hassio_ingress URL (example below). This doesn't load successfully and eventually times out.

Example url - http://homeassistant.local:8123/api/hassio_ingress/j_5oPDT_9oCfTtzpcp_eDPUADuPHIJcIPmiSLpkyXFg/

http://homeassistant.local:8123 loads fine if you pop it into my browser.

In each case, ESPHome can connect to the devices just fine if you click on "Logs" - it's going to that device's web portal that doesnt work.

Screenshots showing working and not working below.

[image]

[image]

### Which version of ESPHome has the issue?

2023.10.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.1

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
Example of device config below.
There's no real difference in config between the ones where "Visit" works and the ones where it gets stuck on the hassio_ingress page.

substitutions:
  name: "kitchen-island-2"
  friendly_name: Kitchen Island 2
packages:
  localbytes.plug-pm: github://LocalBytes/esphome-localbytes-plug/localbytes-plug-pm.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: redacted
wifi:
  power_save_mode: none
  domain: .home
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.1.41
    gateway: 192.168.1.1
    subnet: 255.255.254.0
    dns1: 192.168.1.2
    dns2: 192.168.1.3
```

FruitJelliesGD, 2024-02-22:
add in YAML
```
web_server:
 port: 80
```

CyberFlexie, 2024-12-02:
> add in YAML
> 
> ```
> web_server:
>  port: 80
> ```

I'm not the OP but was having same issue and this solved it!

jamespreedy, 2024-12-02:
Sadly I already have this in place, hasn't resolved this issue for me.

[image]