esphome/issues #5085: Sonoff M5 3 gang US Version: relay 3 randomly stops working until I power cycle the device

santibur06, 2023-11-07:
### The problem

Hi, I have flashed a couple of Sonoff M5 3 gang US Version devices with ESPHome. I recently noticed that after a while (maybe hours, maybe days, but mostly days) the 3rd relay stops working. I have a light linked to that output (GPIO22) and the lights status gets updated when i press the button or change its status in HA (you'll see that in the logs), however, the relay does not turn on. This is happening to all my M5 3 gang sonoffs, the rest of the devices work fine (the rest are 1 gang and 2 gang).
The only way to bring the relay back to life is to cut power to the device and turn it back on.
I've tried restarting the device, no luck. Re-flashing (ota), no luck.

### Which version of ESPHome has the issue?

2023.10.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.0

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

output platform: gpio

### Example YAML snippet

```yaml
substitutions:
  brand_code: snff
  model_code: m5-3c
  platform_name: esp32
  board_type: esp32dev
  status_led_pin: GPIO05
  button1_pin: GPIO4
  button2_pin: GPIO0
  button3_pin: GPIO15
  relay1_pin: GPIO23
  relay2_pin: GPIO19
  relay3_pin: GPIO22
  entity1_name: luz1
  entity2_name: luz2
  entity3_name: luz3
  device_name: toilette-snff-m5-3c
  area_name: Toilette
esphome:
  name: toilette-snff-m5-3c
  build_path: build/toilette-snff-m5-3c
  friendly_name: ''
  platformio_options: {}
  includes: []
  libraries: []
  name_add_mac_suffix: false
  min_version: 2023.10.6
esp32:
  board: esp32dev
  framework:
    version: 2.0.5
    source: ~3.20005.0
    platform_version: platformio/espressif32@5.4.0
    type: arduino
  variant: ESP32
wifi:
  domain: .lan.turkonet.duckdns.org[redacted]
  power_save_mode: NONE
[... 347 lines cut ...]
[18:53:38][C][api:140]:   Address: toilette-snff-m5-3c:6053 
[18:53:38][C][api:144]:   Using noise encryption: NO
[18:53:38][C][wifi_info:009]: WifiInfo IPAddress 'IP Address'
[18:53:38][C][wifi_signal.sensor:009]: WiFi Signal 'RSSI'
[18:53:38][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[18:53:38][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[18:53:38][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[18:53:38][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[18:53:48][D][binary_sensor:036]: 'button3': Sending state OFF
[18:53:49][D][binary_sensor:036]: 'button3': Sending state ON
[18:53:49][D][light:036]: 'luz3 Toilette' Setting:
[18:53:49][D][light:047]:   State: ON
[18:53:51][D][binary_sensor:036]: 'button3': Sending state OFF
[18:53:51][D][binary_sensor:036]: 'button3': Sending state ON
[18:53:51][D][light:036]: 'luz3 Toilette' Setting:
[18:53:51][D][light:047]:   State: OFF
[18:53:57][D][binary_sensor:036]: 'button3': Sending state OFF
[18:53:57][D][binary_sensor:036]: 'button3': Sending state ON
[18:53:57][D][light:036]: 'luz3 Toilette' Setting:
[18:53:57][D][light:047]:   State: ON
[18:53:58][D][binary_sensor:036]: 'button3': Sending state OFF
[18:53:58][D][binary_sensor:036]: 'button3': Sending state ON
[18:53:58][D][light:036]: 'luz3 Toilette' Setting:
[18:53:58][D][light:047]:   State: OFF
```

ssieb, 2023-11-07:
Probably not the issue, but you need to invert all those button inputs.

santibur06, 2023-11-07:
> Probably not the issue, but you need to invert all those button inputs.

Hey. Yeah I saw a couple of code snippets that had those inputs inverted, however, I used esphome's yaml example (https://devices.esphome.io/devices/Sonoff-M5). Buttons work fine, not sure what would change if I inverted them. Can you elaborate on the why? thanks!

ssieb, 2023-11-08:
That config is wrong.  You're using a pullup and the button pulls to GND.  So the state is inverted.  You can see it in your log.  When you push the button, it goes off and when you release, it goes on.

santibur06, 2023-11-08:
> That config is wrong. You're using a pullup and the button pulls to GND. So the state is inverted. You can see it in your log. When you push the button, it goes off and when you release, it goes on.

I didn't pay attention to that in the logs, good catch, thanks!

santibur06, 2023-11-28:
Hi! So I tried something. Left a light on for a couple of days (relay 3 which is the one causing trouble) and I tried turning it off now and it won't do anything. Also from HA, it believes it's turning it off but the relay doesn't turn off. I suspect the esp32 goes into some kind of sleep state and stops changing the output of gpio15. Does this ring a bell to anyone? I would also like to do some logging for a couple of days, any ideas on how to log everything to a file? I'd change logging to very verbose and check if I see something weird there.

ssieb, 2023-11-28:
If the ESP is responding and logging, then it's not asleep.  What do the logs show?

santibur06, 2023-11-29:
Hi, here are the logs:

> INFO ESPHome 2023.11.2
> INFO Reading configuration /config/esphome/living-4-snff-m5-3c.yaml...
> WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
> Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
> See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
> WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
> Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
> See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
> WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
> Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
> See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
> INFO Starting log output from living-4-snff-m5-3c.lan.turkonet.duckdns.org using esphome API
> INFO Successfully connected to living-4-snff-m5-3c.lan.turkonet.duckdns.org in 0.190s
> INFO Successful handshake with living-4-snff-m5-3c.lan.turkonet.duckdns.org in 0.028s
> [13:19:01][I][app:102]: ESPHome version 2023.11.2 compiled on Nov 20 2023, 19:21:34
> [13:19:01][C][status_led:065]: Status Led Light:
> [13:19:01][C][status_led:066]:   Pin: GPIO5
> [13:19:01][C][wifi:559]: WiFi:
> [13:19:01][C][wifi:391]:   Local MAC: 54:43:B2:70:F2:A4
> [13:19:01][C][wifi:396]:   SSID: 'turkonet'[redacted]
> [13:19:01][C][wifi:397]:   IP Address: 10.0.5.164
> [13:19:01][C][wifi:399]:   BSSID: 74:B6:B6:46:79:65[redacted]
> [13:19:01][C][wifi:400]:   Hostname: 'living-4-snff-m5-3c'
> [13:19:01][C][wifi:402]:   Signal strength: -47 dB ▂▄▆█
> [13:19:01][C][wifi:406]:   Channel: 6
> [13:19:01][C][wifi:407]:   Subnet: 255.255.255.0
> [13:19:01][C][wifi:408]:   Gateway: 10.0.5.1
> [13:19:01][C][wifi:409]:   DNS1: 10.0.5.1
> [13:19:01][C][wifi:410]:   DNS2: 8.8.8.8
> [13:19:01][C][logger:416]: Logger:
> [13:19:01][C][logger:417]:   Level: DEBUG
> [13:19:01][C][logger:418]:   Log Baud Rate: 115200
> [13:19:01][C][logger:420]:   Hardware UART: UART0
> [13:19:01][C][uptime.sensor:031]: Uptime Sensor 'Uptime'
> [13:19:01][C][uptime.sensor:031]:   Device Class: 'duration'
> [13:19:01][C][uptime.sensor:031]:   State Class: 'total_increasing'
> [13:19:01][C][uptime.sensor:031]:   Unit of Measurement: 's'
> [13:19:01][C][uptime.sensor:031]:   Accuracy Decimals: 0
> [13:19:01][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
> [13:19:01][C][ledc.output:164]: LEDC Output:
> [13:19:01][C][ledc.output:165]:   Pin GPIO18
> [13:19:01][C][ledc.output:166]:   LEDC Channel: 0
> [13:19:01][C][ledc.output:167]:   PWM Frequency: 1000.0 Hz
> [13:19:01][C][ledc.output:168]:   Bit depth: 16
> [13:19:01][C][gpio.output:010]: GPIO Binary Output:
> [13:19:01][C][gpio.output:011]:   Pin: GPIO23
> [13:19:01][C][gpio.output:010]: GPIO Binary Output:
> [13:19:01][C][gpio.output:011]:   Pin: GPIO19
> [13:19:01][C][gpio.output:010]: GPIO Binary Output:
> [13:19:01][C][gpio.output:011]:   Pin: GPIO22
> [13:19:01][C][template.binary_sensor:028]: Template Binary Sensor 'API connected'
> [13:19:01][C][template.binary_sensor:028]:   Device Class: 'connectivity'
> [13:19:01][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'button1'
> [13:19:01][C][gpio.binary_sensor:016]:   Pin: GPIO4
> [13:19:01][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'button2'
> [13:19:01][C][gpio.binary_sensor:016]:   Pin: GPIO0
> [13:19:02][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'button3'
> [13:19:02][C][gpio.binary_sensor:016]:   Pin: GPIO15
> [13:19:02][C][light:103]: Light 'Status LED'
> [13:19:02][C][light:103]: Light 'LEDs'
> [13:19:02][C][light:105]:   Default Transition Length: 1.0s
> [13:19:02][C][light:106]:   Gamma Correct: 2.80
> [13:19:02][C][light:103]: Light 'Luces 1 Galeria Patio'
> [13:19:02][C][light:103]: Light 'Luces 2 Galeria Patio'
> [13:19:02][C][light:103]: Light 'Luz Puerta Galeria Patio'
> [13:19:02][C][output.switch:068]: Output Switch 'Luces 1 Galeria Patio'
> [13:19:02][C][output.switch:091]:   Restore Mode: always OFF
> [13:19:02][C][output.switch:068]: Output Switch 'Luces 2 Galeria Patio'
[... 62 lines cut ...]

This other one is me pressing the button physically:

> [13:20:15][D][binary_sensor:036]: 'button3': Sending state OFF
> [13:20:16][D][binary_sensor:036]: 'button3': Sending state ON
> [13:20:16][D][light:036]: 'Luz Puerta Galeria Patio' Setting:
> [13:20:16][D][light:047]:   State: OFF
> [13:20:16][D][binary_sensor:036]: 'button3': Sending state OFF
> [13:20:17][D][binary_sensor:036]: 'button3': Sending state ON
> [13:20:17][D][light:036]: 'Luz Puerta Galeria Patio' Setting:
> [13:20:17][D][light:047]:   State: ON
> [13:20:17][D][binary_sensor:036]: 'button3': Sending state OFF
> [13:20:18][D][binary_sensor:036]: 'button3': Sending state ON
> [13:20:18][D][light:036]: 'Luz Puerta Galeria Patio' Setting:
> [13:20:18][D][light:047]:   State: OFF
> [13:20:18][D][binary_sensor:036]: 'button3': Sending state OFF
> [13:20:19][D][binary_sensor:036]: 'button3': Sending state ON
> [13:20:19][D][light:036]: 'Luz Puerta Galeria Patio' Setting:
> [13:20:19][D][light:047]:   State: ON
> [13:20:19][D][binary_sensor:036]: 'button3': Sending state OFF`

No response from the relay.

Thanks!
Santiago

ssieb, 2023-11-29:
You could try using a `switch` instead of an `output`, so that you'll get a clear message that it's turning on.
This really looks like a hardware issue of some sort.

santibur06, 2023-11-30:
yeah it's weird, the thing is I purchased these M5 3gang sonoffs on different times and I'm pretty sure these are different production lots, all of them are causing trouble.

Here's my new config using switches as you suggested (and below the logs):

```
substitutions:
  brand_code: snff
  model_code: m5-3c
  platform_name: esp32
  board_type: esp32dev
  device_name: living-4-snff-m5-3c
  area_name: Galeria Patio
  entity1_name: Luces 1
  entity2_name: Luces 2
  entity3_name: Luz Puerta
  status_led_pin: GPIO05
  button1_pin: GPIO4
  button2_pin: GPIO0
  button3_pin: GPIO15
  relay1_pin: GPIO23
  relay2_pin: GPIO19
  relay3_pin: GPIO22
esphome:
  name: living-4-snff-m5-3c
  build_path: build/living-4-snff-m5-3c
  friendly_name: ''
  area: ''
  platformio_options: {}
  includes: []
  libraries: []
  name_add_mac_suffix: false
  min_version: 2023.11.2
esp32:
  board: esp32dev
  flash_size: 4MB
  framework:
    version: 2.0.5
    source: ~3.20005.0
    platform_version: platformio/espressif32@5.4.0
    type: arduino
  variant: ESP32
wifi:
  domain: .lan.turkonet.duckdns.org
  power_save_mode: NONE
  ap:
    password: '!qaz2wsx323.turzi!'
    ap_timeout: 1min
  reboot_timeout: 15min
  fast_connect: false
  passive_scan: false
  enable_on_boot: true
  networks:
  - ssid: turkonet
    password: oaccbb845782
    priority: 0.0
  use_address: living-4-snff-m5-3c.lan.turkonet.duckdns.org
ota:
  safe_mode: true
  port: 3232
  reboot_timeout: 5min
  num_attempts: 10
api:
  port: 6053
  password: ''
  reboot_timeout: 15min
logger:
  level: DEBUG
  baud_rate: 115200
  tx_buffer_size: 512
  deassert_rts_dtr: false
[... 306 lines cut ...]
> [21:41:49][C][mdns:116]:   Hostname: living-4-snff-m5-3c
> [21:41:49][C][ota:097]: Over-The-Air Updates:
> [21:41:49][C][ota:098]:   Address: living-4-snff-m5-3c.lan.turkonet.duckdns.org:3232
> [21:41:49][C][api:139]: API Server:
> [21:41:49][C][api:140]:   Address: living-4-snff-m5-3c.lan.turkonet.duckdns.org:6053
> [21:41:49][C][api:144]:   Using noise encryption: NO
> [21:41:49][C][wifi_info:009]: WifiInfo IPAddress 'IP Address'
> [21:41:49][C][wifi_signal.sensor:009]: WiFi Signal 'RSSI'
> [21:41:49][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
> [21:41:49][C][wifi_signal.sensor:009]:   State Class: 'measurement'
> [21:41:49][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
> [21:41:49][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
> [21:42:01][D][sensor:094]: 'RSSI': Sending state -36.00000 dBm with 0 decimals of accuracy
> [21:42:09][D][binary_sensor:036]: 'button3': Sending state ON
> [21:42:09][D][switch:020]: 'Luz Puerta Galeria Patio' Toggling ON.
> [21:42:09][D][switch:055]: 'Luz Puerta Galeria Patio': Sending state ON
> [21:42:09][D][binary_sensor:036]: 'button3': Sending state OFF
> [21:42:10][D][binary_sensor:036]: 'button3': Sending state ON
> [21:42:10][D][switch:020]: 'Luz Puerta Galeria Patio' Toggling OFF.
> [21:42:10][D][switch:055]: 'Luz Puerta Galeria Patio': Sending state OFF
> [21:42:10][D][binary_sensor:036]: 'button3': Sending state OFF
> [21:42:44][D][switch:012]: 'Luz Puerta Galeria Patio' Turning ON.
> [21:42:44][D][switch:055]: 'Luz Puerta Galeria Patio': Sending state ON
> [21:42:48][D][switch:016]: 'Luz Puerta Galeria Patio' Turning OFF.
> [21:42:48][D][switch:055]: 'Luz Puerta Galeria Patio': Sending state OFF

nkonidaris, 2023-12-03:
Good day, @santibur06 

I am experiencing the same problem with my 3-gang Switch (US version), I think it might be a hardware problem. My third relay doesnt even switch after reboot or any changes. It stopped working a few months ago with no warning.

santibur06, 2023-12-03:
I found something else... So I have noticed that some of my Sonoff M5 **2**-Gang won't turn on the LED for button 2 after a while (hours/days). I just remembered that those LEDs are not directly connected to Relay 2 on the 2Gang sonoff, they are controlled independently and guess what, it's GPIO22, the same pin that controls relay3 on the 3gang. So GPIO22 is definitely the problem

santibur06, 2023-12-03:
Another piece of info that may be helpful. It seems that when these sonoffs get into this zombie mode, they also stop reporting their host name as the device name and they kind of default to esp32-macaddress, see screenshot below

[image]

nkonidaris, 2023-12-04:
Hey @santibur06 

You were correct! After a power cycle it worked again, I had tried power cycling it before but I did reflash it a few time after so not sure what broke it. I have also noted the light switch hostname changing to esp32-macaddress, I thought it was unrelated but perhaps this is some software bug.

cr0ss3r, 2023-12-04:
Hi,
I have the same issue with M5, not flashed - with original firmware. It is EU version and it stops working after a day or two. It reports switching on and off, but does not really do the job. Switching off and on electric fuse (literally powering it off and on again) fixes the problem for another day or two.

miwojcik, 2024-01-13:
Did anyone find a solution? I have a similar button LED problems with M5-2C running HAA firmware and additionally they are randomly crashing as I mentioned in the https://github.com/RavenSystem/esp-homekit-devices/issues/2190.

virustex, 2024-01-14:
Installed 15 Sonoff M5s with a mix of 1/2/3 Gangs, Have been facing the same problem. The switches go into zombie state, stop responding and then restart after 5 to 10 minutes. Also the 3rd relay hangs up after 24-48 hours of being turned on. Facing severe backlash from wife, if this does not sort out soon enough I'm afraid I'll loose my investment.

santibur06, 2024-01-14:
@virustex running esphome or stock firmware?

miwojcik, 2024-01-15:
I compared one of the broken and one of the working switches I have. They probably came from different sellers as they are different HW revisions. The biggest difference is the ESP chip itself! The affected one uses ESP32-DOWD R2-V3 while the one that is working uses ESP32-DOWD V3 (no R2 here!). From what I understand they differ in the PSRAM size.

Pictures:
Broken one:
[image]
[image]

Working one:
[image]
[image]

miwojcik, 2024-01-16:
In HAA discussion https://github.com/RavenSystem/esp-homekit-devices/discussions/2191#discussioncomment-8136443 it was suggested to declare GPIO16 as input and enable pull-up. I applied that to all affected Sonoffs and they have been stable for the last 9h (my APs decided to do update at night). Usually I'd have at least one crash in that time so I think it's worth a try!

virustex, 2024-01-17:
> @virustex running esphome or stock firmware?

Running esphome

miwojcik, 2024-01-17:
After two days I can confirm that setting GPIO16 as input pull-up solved both crashing and GPIO22 unresponsiveness.

santibur06, 2024-01-18:
@miwojcik that is a great find. Just to confirm, this what you did?

```
binary_sensor:
  - platform: gpio
    id: gpio16
    pin:
      number: GPIO16
      mode:
        input: true
        pullup: true

```

santibur06, 2024-01-18:
@miwojcik no worries. I flashed that change to 60+ devices last night. I'll post an update tomorrow after 24hs. Thanks! @virustex maybe you wanna give it a shot too?

ahsanimran96, 2024-01-18:
@santibur06 yep flashed half of the devices yesterday with the following addtion under binary sensor:

```
- platform: gpio
    id: fixgpio
    pin: 
      number: 16
      mode: 
        input: True
        pullup: True
`````
This did the trick, devices have been up for more than 24 hours now. Thanks alot guys!

andrewraj, 2024-02-27:
> In HAA discussion [RavenSystem/esp-homekit-devices#2191 (comment)](https://github.com/RavenSystem/esp-homekit-devices/discussions/2191#discussioncomment-8136443) it was suggested to declare GPIO16 as input and enable pull-up. I applied that to all affected Sonoffs and they have been stable for the last 9h (my APs decided to do update at night). Usually I'd have at least one crash in that time so I think it's worth a try!

I have the same issue mentioned above where the switch stops working and then upon a power cycle it continues working. Just to add, if I leave the switch as is, it will eventually work again without a power cycle. I have M5-1C, M5-2C and M5-3C but it looks like only M5-2C seems to be having this issue. So, I'm wondering if setting up GPIO16 as input pullup apply to all 3 models of switches or only the affected ones?

santibur06, 2024-02-27:
> I have the same issue mentioned above where the switch stops working and then upon a power cycle it continues working. Just to add, if I leave the switch as is, it will eventually work again without [...]

Hi. Yes, go ahead and try that, it worked for me on m5 2c and 3c

santibur06, 2025-08-26:
> sorry to necro this thread as I have a lot of EU version of 1C, 2C, and 3C but it seems one 2C will crash and restart itself (sometimes in bootloop for hours), but eventually it will come back and stays for few hours before crashing again. I've tried taking it apart and power the ESP32 directly from 5v and it seems stable, only crashing when it's connected to the back of the SONOFF, I was thinking it must be with hardware issue however I tried flashing it with TASMOTA and it's stable so this might be something wrong with ESPHome, but the weird thing is I have a lot of 2C with the same config and they are stable, only this one have issues.
> 
> May I know [@santibur06](https://github.com/santibur06) if your 2C and 3C is still stable after applying the pin16 fix? I was wondering if I should also apply the pin16 fix for all my 1C, 2C, and 3C that are stable.

Hey. I would try swapping the back part with another 2C and see what happens. Even maybe try moving that sonoff to another location to rule out any wifi related issues that may be causing your device to reboot. Also, these devices are not grounded so they are susceptible to some noise in the line.

My 3C devices have been stable since I applied that pin16 patch.