esphome/issues #5087: esphome/esphome#5616 broke bluetooth_proxy

cyberhuman, 2023-11-08:
### The problem

After esphome/esphome#5616, BLE proxy can't connect to clients anymore.

### Which version of ESPHome has the issue?

dev

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
esp32_ble_tracker:
  scan_parameters:
    # We currently use the defaults to ensure Bluetooth
    # can co-exist with WiFi In the future we may be able to
    # enable the built-in coexistence logic in ESP-IDF
    active: true

bluetooth_proxy:
  active: true
```
```

### Anything in the logs that might be useful for us?

```txt
[21:57:11][C][esp32_ble:027]: Setting up BLE...
[21:57:11][E][esp32_ble_client:026]: gattc app register failed. app_id=1 code=259
[21:57:11][E][component:113]: Component bluetooth_proxy was marked as failed.
[21:57:11][E][esp32_ble_client:026]: gattc app register failed. app_id=2 code=259
[21:57:11][E][component:113]: Component bluetooth_proxy was marked as failed.
[21:57:11][E][esp32_ble_client:026]: gattc app register failed. app_id=3 code=259
[21:57:11][E][component:113]: Component bluetooth_proxy was marked as failed.
```
```

### Additional information

esphome/esphome#5616 moves initialization of BLE from `ESP32::setup()` to `ESP32::ble_setup_()`, which is called from the first invocation of `ESP32::loop()`. It means when `BLEClientBase/BluetoothConnection::setup()` is called, BLE is not initialized yet and the bluetooth connection component fails as a result.

Rapsssito, 2023-11-08:
@cyberhuman, sorry, indeed my PR caused this issue. I did not catch the bluetooth_proxy dependency. Could you check if changing `esp32_ble_client/ble_client_base.cpp` `setup()` and `loop()` to this fixes your issues?

```cpp
void BLEClientBase::setup() {
  static uint8_t connection_index = 0;
  this->connection_index_ = connection_index++;
}

void BLEClientBase::loop() {
  if (!global_ble_server->is_running()) {
    this->set_state(espbt::ClientState::INIT);
    return;
  }
  if (this->state_ == espbt::ClientState::INIT) {
    auto ret = esp_ble_gattc_app_register(this->app_id);
    if (ret) {
      ESP_LOGE(TAG, "gattc app register failed. app_id=%d code=%d", this->app_id, ret);
      this->mark_failed();
    }
    this->set_state(espbt::ClientState::IDLE);
  }
  // READY_TO_CONNECT means we have discovered the device
  // and the scanner has been stopped by the tracker.
  if (this->state_ == espbt::ClientState::READY_TO_CONNECT) {
    this->connect();
  }
}
```

I would test it myself, but sadly, I am not able to test `bluetooth_proxy` with my current setup.

jesserockz, 2023-11-08:
@Rapsssito Can you open a PR with the changes?

Rapsssito, 2023-11-08:
@jesserockz, I have created the PR: https://github.com/esphome/esphome/pull/5704. However, I am not able to test it with my setup, so further testing is needed to check if the fix actually fixes the issue.

closed 2023-11-08 by jesserockz