esphome/issues #5089: Display ST7789v reset pin set as Required when it should be set as Optional but when not there tries to set the pin to 41.

NonaSuomy, 2023-11-08:
### The problem

```yaml
# TTGO-Camera Plus 001
# https://www.lilygo.cc/en-ca/products/t-camera-plus
# https://github.com/Xinyuan-LilyGO/esp32-camera-screen/blob/master/schematic/OV2640_V2.0.pdf

substitutions:  
  # SPI
  spi_clk: GPIO21
  spi_mosi: GPIO19
  spi_miso: GPIO22
  
  # Display Physical Connector
  #1: BLK = Backlight
  #2: D/C = Data/Command
  #3: RES = Reset
  #4: SDA = Serial Data or SPI MOSI
  #5: SCL = Serial Clock or SPI SCK
  #6: VCC (3.3V)
  #7: Ground
  dis_cs: GPIO12 # Chip Select
  dis_dc: GPIO15 # Data/Command
  dis_bk: GPIO2 # Backlight
  
display:
  - platform: st7789v
    id: st7789_display
    model: Custom
    height: 240
    width: 240
    offset_width: 0
    offset_height: 0
    backlight_pin: $dis_bk
    cs_pin: $dis_cs
    dc_pin: $dis_dc
    #reset_pin: 0
```

I first was getting this error when I had it set to model: Adafruit_Funhouse 240x240

```
display.st7789v: [source /config/ttgo-camera-plus.yaml:631]
  
  Invalid pin number: 41 (must be 0-39).
```

Jesse told me to use model: Custom and set the params for it as that model above is for an ESP32-S3.

After doing so I got this message:
```
"both dc_pin and reset_pin must be specified"
```

Jesse looked at it and showed this
```c++
    if CONF_DC_PIN not in config or CONF_RESET_PIN not in config:
        raise cv.Invalid(f"both {CONF_DC_PIN} and {CONF_RESET_PIN} must be specified")
```

Not sure about the GPIO41 but Jesse said it only exists on an ESP32-S3 not the ESP32-DOWDQ6 on this module.

My board is set to esp32dev so I'm not sure why it is trying to set gpio41 to it when you don't specify the reset pin

```yaml
esp32:
  board: esp32dev
  framework:
    #type: arduino
    type: esp-idf
```

The documentation should show it as optional as well.

The display does work though when I set the reset pin to an arbitrary GPIO number like 0 above.

Thank you.

### Which version of ESPHome has the issue?

2023.10.6

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

Issue with ESPHome not HA

### What platform are you using?

ESP32

### Board

TTGO-Camera Plus

### Component causing the issue

Display ST7789v

kiklhorn, 2023-11-10:
+1
Allow set reset_pin to -1, null, none or something similar. On multiple boards reset pin  not connected to any GPIO.

clydebarrow, 2023-11-23:
Fixed in PR #5406 - use the ili9xxx driver and bring in the PR.

```
external_components:
  - source: github://pr#5406
    components: [ ili9xxx ]

display:
  - platform: ili9xxx
    model: st7789v
    id: st7789v
    dimensions:
      height: 240
      width: 240
      offset_width: 0
      offset_height: 0
    cs_pin: $dis_cs
    dc_pin: $dis_dc
```

NonaSuomy, 2023-11-23:
Tested! Had to add invert_colors: true but it is working so far now!

Thank you

closed 2023-11-28 by jesserockz