esphome/issues #5091: Constant connection issues

TB-SE, 2023-11-08:
### The problem

My ESPHome devices constantly disconnect and turn unavailable for a couple of seconds (anything between 5-15 seconds). Here are some examples of the errors found in the system log:

```
Logger: aioesphomeapi.connection
Source: runner.py:188
First occurred: 00:25:50 (1 occurrences)
Last logged: 00:25:50

ep-lite-office @ 192.168.50.91: Connection error occurred: ep-lite-office @ 192.168.50.91: EOF received
```

```
Logger: aioesphomeapi.connection
Source: runner.py:188
First occurred: November 4, 2023 at 23:30:52 (232 occurrences)
Last logged: 01:04:19

ep1-guestroom @ 192.168.50.90: Connection error occurred: ep1-guestroom @ 192.168.50.90: EOF received
esp32-bt-proxy-hallway @ 192.168.50.193: Connection error occurred: [Errno 104] Connection reset by peer
ep-lite-office @ 192.168.50.91: Connection error occurred: ep-lite-office @ 192.168.50.91: EOF received
ep-lite-kitchen @ 192.168.50.92: Connection error occurred: ep-lite-kitchen @ 192.168.50.92: EOF received
ep1-hallway @ 192.168.50.89: Connection error occurred: [Errno 104] Connection reset by peer
```

Any idea what's happening? I'm actively using 16 ESPHome devices, five are BT proxies and two are voice assistants (M5 Stack Echo).

### Which version of ESPHome has the issue?

2023.10.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.1

### What platform are you using?

ESP32

### Board

Mostly generic ESP32 NODEMCU

LukasJerabek, 2024-03-08:
My tip for start is to add signal strength and uptime measures in wifi section (see doc) and make sure its not below -60.

hugh-martin, 2024-09-29:
Sounds similar to [#5754](https://github.com/esphome/issues/issues/5754) where adding `fast_connect: true` to the `wifi` section resolves (or works around) the issue.