esphome/issues #5092: Custom component won't compile on new HA build

greenbeast999, 2023-11-09:
### The problem

I've got a custom component for a split AC unit i have. I was able to compile in windows a few months back but had trouble adding it to HA so dropped the project for a bit, i had to get a RPI4 to have enough memory to compile directly from HA but now i have 2-3 errors compiling. Suspect something has changed in the newer version of ESPHome but i'm not good enough to figure out what.

Many thanks

### Which version of ESPHome has the issue?

2023.10.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.5

### What platform are you using?

ESP8266

### Board

OEM split AC board

### Example YAML snippet

```yaml
esphome:
  name: electriq-12000-ac
  includes:
    - electriq_ac.h

esp8266:
  board: esp01_1m

# Enable logging (and disable DEBUG)
logger:
  level: INFO

# Enable Home Assistant API
api:

ota:
  password: "SET_BY_HOME_ASSISTANT"

wifi:
  ssid: Nick
  password: MYWIFI7855

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Electriq-12000-Ac-1"
    password: "SET_BY_HOME_ASSISTANT"

captive_portal:

uart:
  id: uart_bus
  rx_pin: GPIO13
  tx_pin: GPIO15
  baud_rate: 9600

climate:
  - platform: custom
    lambda: |-
      auto electriqac = new ElectriqAC(id(uart_bus));
      App.register_component(electriqac);
      return {electriqac};
[... 215 lines cut ...]
      |              ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/pstl/glue_algorithm_defs.h:13,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/algorithm:74,
                 from src/esphome/core/optional.h:19,
                 from src/esphome/components/socket/socket.h:5,
                 from src/esphome/components/api/api_frame_helper.h:14,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/functional:785:5: note: candidate: 'template<class _Func, class ... _BoundArgs> typename std::_Bind_helper<std::__is_socketlike<_Func>::value, _Func, _BoundArgs ...>::type std::bind(_Func&&, _BoundArgs&& ...)'
  785 |     bind(_Func&& __f, _BoundArgs&&... __args)
      |     ^~~~
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/functional:785:5: note:   substitution of deduced template arguments resulted in errors seen above
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/functional:809:5: note: candidate: 'template<class _Result, class _Func, class ... _BoundArgs> typename std::_Bindres_helper<_Result, _Func, _BoundArgs>::type std::bind(_Func&&, _BoundArgs&& ...)'
  809 |     bind(_Func&& __f, _BoundArgs&&... __args)
      |     ^~~~
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/functional:809:5: note:   template argument deduction/substitution failed:
In file included from src/esphome.h:55,
                 from src/main.cpp:3:
src/esphome/core/base_automation.h:135:23: note:   couldn't deduce template parameter '_Result'
  135 |     auto f = std::bind(&DelayAction<Ts...>::play_next_, this, x...);
      |              ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pioenvs/electriq-12000-ac/src/main.cpp.o] Error 1
========================= [FAILED] Took 206.61 seconds =========================
```

ssieb, 2023-11-09:
Duplicate of https://github.com/esphome/issues/issues/5034

ssieb, 2023-11-09:
But as mentioned in the other issue, you shouldn't have long delays in triggers anyway.  Move it to a script.

closed 2023-11-09 by ssieb

greenbeast999, 2023-11-09:
> Move it to a script.

Unfortunately this doesn't mean much to me, i mean i know what a script is, but i don't know enough about HA/ESPHome to do anything with that statement. Could you possibly elaborate for me?

ssieb, 2023-11-09:
https://esphome.io/guides/automations#script-component
If you need more help, then come ask on the esphome discord server.

greenbeast999, 2023-11-09:
Thanks i will have a read