esphome/issues #5093: Error uploading binary using esphome CLI. AttributeError: 'Namespace' object has no attribute 'username'

dylan09, 2023-11-09:
### The problem

Stayed a long time on version 2022.10.1. After updating to 2023.10.4 I am unable to upload firmware to device using command line `esphome upload CONFIGURATION`. Error: `AttributeError: 'Namespace' object has no attribute 'username'`
Complete Traceback:
```
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 1015, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 1002, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 404, in command_upload
    exit_code = upload_program(config, args, port)
  File "/esphome/esphome/__main__.py", line 316, in upload_program
    config, args.username, args.password, args.client_id
AttributeError: 'Namespace' object has no attribute 'username'
```

I tried different esphome versions and found the first version where I get this error is 2023.6.1. 
Tried to find something in the release notes. The only entry I found was PR #3887. But thats marked as non breaking change

Maybe someone has a hint what I have to change in my configuration to get OTA upload working with versions past 2023.6.1.

 

### Which version of ESPHome has the issue?

2023.6.1 and newer

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.10.5

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

esphome CLI

### Example YAML snippet

```yaml
substitutions:
  node_name: wz-aq-vindriktning
  id_prefix: wz_aq_vindriktning
  name_prefix: wz_aq_vindriktning
  psm: HIGH
  node_comment: "IKEA Vindriktning Air Quality Sensor."

wifi:
  ssid: "H3N1"
  password: !secret h3n1_passwd
  domain: .home.umartiensen.de
  power_save_mode: ${psm}

  ap:
    ssid: ${node_name} fallback
    password: !secret fallback_ap_passwd

mdns:
  disabled: true

web_server:
  port: 80
  version: 1
  auth:
    username: XXXXXXXX
    password: !secret web_passwd

mqtt:
  broker: XXXX.XXXX.XXXXXX.de
  username: XXXXXX
  password: !secret mqtt_passwd
  discovery: true
  topic_prefix: esphome/${id_prefix}
  reboot_timeout: 60 min

time:
  - platform: sntp
    id: sntp_time
    servers: XXXX.XXXX.XXXXXX.de
    timezone: Europe/Berlin

ota:
  password: !secret ota_passwd

captive_portal:

switch:
  - platform: safe_mode
    id: ${id_prefix}_safe_restart
    name: ${name_prefix} Safe Restart
    
  - platform: restart
    id: ${id_prefix}_restart
    name: ${name_prefix} Restart

esphome:
  name: ${node_name}
  comment: ${node_comment}

esp8266:
  board: esp01_1m
  
logger:
  level: INFO
  baud_rate: 0

uart:
  rx_pin: GPIO3
  baud_rate: 9600

sensor:
  - platform: pm1006
    pm_2_5:
      name: pm25
      accuracy_decimals: 2
      filters:
        - or:
          - delta: 3
          - heartbeat: 10s
        - sliding_window_moving_average:
            window_size: 20
            send_every: 10
            send_first_at: 10
```

### Anything in the logs that might be useful for us?

```txt
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 1015, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 1002, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 404, in command_upload
    exit_code = upload_program(config, args, port)
  File "/esphome/esphome/__main__.py", line 316, in upload_program
    config, args.username, args.password, args.client_id
AttributeError: 'Namespace' object has no attribute 'username'
```

dylan09, 2023-11-13:
Anyone also using the esphome command line to build and manage esphome projects.
I don't know how to fix the error when uploading the binary to my esp device. And I am not familiar enough to test or debug the python code.

dylan09, 2023-11-13:
After a few tests, I found there was a bug with the changes to mqtt #3887  in esphome 2023.6.1.

The `run` command works as expected. But `upload` failed. Maybe `upload` did not collect all necessary parameter for the MQTT connection from yaml.

dylan09, 2023-11-13:
A bug has crept in with esphome 2023.6.1. The parameter parser for the upload command should also get the mqtt options.

ssieb, 2023-11-13:
The problem is that esphome has no way of finding the IP address of your device.  I think there was something added to try to get it from mqtt, but maybe the current esphome version on your device doesn't provide the right information.  Or maybe there's a bug in the code to find the address.  Very few people use this, so it's hard to test.
Yes, https://esphome.io/changelog/2023.6.0#all-changes, see `allow to use MQTT for discovery of IPs if mDNS is no option`.  I have no idea what it would have tried to do in any versions before that.

ssieb, 2023-11-14:
Also, it looks like you need at least that version for it to work anyway because the device won't know how to respond.

dylan09, 2023-11-14:
Yes. You are right. To get the changes introduced with esphome 2023.6.x working out of the box, you need the new MQTT discover command and response from the module. That's the same If you didn't have a working DNS. And the module name could not be resolved.

For this there was the parameter `--device IP'`  to connect the module for OTA update. But this parameter is also not working since esphome 2023.6.x

So in my opinion there are two bugs since 2023.6.x:
1. The parameter/configuration for the `run` command was extended by the necessary MQTT user and password params. This change was not applied to the `upload` command. This fix is easy. Even for me as a "no python" programmer. 😀
2. The parameter `--device IP` isn't working as it was before esphome 2023.6.x. If you add a `--device IP` to the esphome command it's ignored. Even with the parameter it is trying to discover the IP via MQTT. Don't know how to fix this in code 😖

The way to update a module with esphome prior to 2023.6.x was to compile and upload using the command `esphome run CONFIG --device IP-ADDR` (or upload after compile `esphome upload CONFIG --device IP-ADDR'). 
After the module is once updated the MQTT discover should work and the `--device IP` parameter is not needed anymore.

peterhoeg, 2024-01-10:
Did you by chance find a workaround? I'm hitting the same thing here.

dylan09, 2024-01-17:
I have changed the upload command parser. Have a look at my PR above. Only one or two lines of code.
And if I remember correct, the run command works. 

After I switches to the new esphome version I changed my devices from MQTT to use the API. There are some weird changes a few months ago regarding the entity naming scheme. 

And a last way was too use two different esphome versions (I use venv with different versions on my mac). Compile the code with the new esphome version. And than use an old version to run the upload command.

peterhoeg, 2024-01-25:
My issue is that I don't use mqtt discovery, but instead `--device $IP` (well, host name, but still) and the mentioned PR doesn't address that.

closed 2024-04-28 by jesserockz