esphome/issues #5094: ESP Cam can't share I2C bus with other devices

NonaSuomy, 2023-11-09:
### The problem

When you enable the camera on the I2C bus the scan will detect all the I2C devices yet only the Camera seems to optimizes the bus and no other devices respond.

ssieb pointed to this pull request that fixes the issue
https://github.com/espressif/esp32-camera/pull/413

### Which version of ESPHome has the issue?

2023.10.6

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

Issue with ESPHome not HA

### What platform are you using?

ESP32-IDF

### Board

TTGO-Camera Plus

### Component causing the issue

OV2640

### Example YAML snippet

```yaml
# Camera
  cam_d0: GPIO34   # Camera Data pin 0
  cam_d1: GPIO13   # Camera Data pin 1
  cam_d2: GPIO26   # Camera Data pin 2
  cam_d3: GPIO35   # Camera Data pin 3
  cam_d4: GPIO39   # Camera Data pin 4
  cam_d5: GPIO38   # Camera Data pin 5
  cam_d6: GPIO37   # Camera Data pin 6
  cam_d7: GPIO36   # Camera Data pin 7
  cam_vsync: GPIO5 # Camera VSYNC
  cam_href: GPIO27 # Camera HREF
  cam_pclk: GPIO25 # Camera Pixel Clock
  cam_xclk: GPIO4  # Camera External Clock
  cam_sda: GPIO18  # Camera SDA
  cam_scl: GPIO23  # Camera SCL

  # I2C Bus: 0x30 Camera OV2640 2MP, 0x75 IP5306 Battery Management, 0x68 MPU6050 Accelerometer/Gyroscope Sensor
  i2c_sda: GPIO18 # I2C Serial Data
  i2c_scl: GPIO23 # I2C Serial Clock
external_components:
  # IP5306 Battery Management
  - source:
      type: git
      url: https://github.com/ssieb/custom_components
    components: [ ip5306 ]

# TTGO Camera Plus
# OV2640 2Megapixel
esp32_camera:
  name: TTGOCameraPlus001
  external_clock:
    pin: $cam_xclk
    frequency: 20MHz
  i2c_pins:
    sda: $cam_sda
    scl: $cam_scl
  data_pins: [$cam_d0, $cam_d1, $cam_d2, $cam_d3, $cam_d4, $cam_d5, $cam_d6, $cam_d7]
  vsync_pin: $cam_vsync
  href_pin: $cam_href
  pixel_clock_pin: $cam_pclk
  vertical_flip: false
  horizontal_mirror: false
# Camera Web Server
esp32_camera_web_server:
#  - port: 8080
#    mode: stream
  - port: 8081
    mode: snapshot

# I2C Bus Addresses
# 0x30 Camera OV2640 2Megapixel
# 0x68 MPU6050 Accelerometer/Gyroscope Sensor
# 0x75 IP5306 Battery Management
i2c:
  sda: $i2c_sda
  scl: $i2c_scl
  scan: true
  id: bus_a

# IP5306 Battery Management
# Charging Current: 1A
# Battery: 3.7V lithium battery
ip5306:
  i2c_id: bus_a
  address: 0x75
  battery_level:   # sensor
    name: Battery Level
  charger_connected:  # binary_sensor
    id: connected
    name: Battery Charger Connected
    on_press:
      then:
        - lambda: ESP_LOGD("TEST", "charging");
    on_release:
      then:
        - lambda: ESP_LOGD("TEST", "not charging");
  charge_full:  # binary_sensor
    id: full
    name: Battery Charge Full
    on_press:
      then:
        - lambda: ESP_LOGD("TEST", "fully charged");
    on_release:
      then:
        - lambda: ESP_LOGD("TEST", "still charging");

# MPU6050 Accelerometer/Gyroscope Sensor
sensor:
  - platform: mpu6050
    i2c_id: bus_a
    address: 0x68
    update_interval: 1s
    accel_x:
      name: "MPU6050 Accel X"
    accel_y:
      name: "MPU6050 Accel Y"
    accel_z:
      name: "MPU6050 Accel z"
    gyro_x:
      name: "MPU6050 Gyro X"
    gyro_y:
      name: "MPU6050 Gyro Y"
    gyro_z:
      name: "MPU6050 Gyro z"
    temperature:
      name: "MPU6050 Temperature"
```

### Anything in the logs that might be useful for us?

```txt
The I2C scan does show all 3 devices on the bus

0x30 Cam
0x68 IP5306
0x75 MPU6050
```

NonaSuomy, 2023-11-10:
ssieb did this test with me and it worked for Arduino but they said the IDF camera library needs an update before this will work with IDF
```yaml
  - source: github://ssieb/esphome@cam_i2c
    components: [ esp32_camera ]
    refresh: 1min
 ```
 Have to remove the I2C pins from the camera pinout defined:
 ```yaml
   i2c_pins:
    sda: $cam_sda
    scl: $cam_scl
 ```
 
 Just use the regular i2c: defined in your YAML
 ```yaml
i2c:
  sda: $i2c_sda
  scl: $i2c_scl
  scan: true
  id: bus_a
```
 Note that trying to use a display at the same time caused this error message
 `[19:38:06][W][esp32_camera:190]: Got invalid frame from camera!`
 
 Without the display everything worked smoothly tested the I2C other devices down to refresh_interval: 100ms with no errors like above.
 
Thanks ssieb

NonaSuomy, 2025-03-16:
Did you try the ssieb build above with arduino or you are using idf?

ssieb, 2025-03-17:
It's not IDF that needs updating for this, it's the esp-cam library.  And that hasn't been updated yet.

xiasi0, 2025-03-17:
I am using the GC0308 (the same camera as the M5Stack CoreS3), which utilizes a shared I2C bus, but I have had no luck getting it to work.
Perhaps this camera is simply not supported; I've noticed that the community predominantly uses the OV2640 and OV5640.

NonaSuomy, 2025-03-18:
Seems like it is https://github.com/espressif/esp32-camera?tab=readme-ov-file#supported-sensor
Though from what ssieb says, maybe it was supported after they already added the code base to esphome.

Did you test with arduino framework set in esphome and using these settings?

Also remove any other i2c devices you may have put in the yaml. Only setup up the camera, make a simple yaml for it and see how it goes.

```yaml
esp32:
  board: esp32dev
  framework:
    type: arduino
external_components:
# ESP32 Camera 
  - source: github://ssieb/esphome@cam_i2c
    components: [ esp32_camera ]
```

xiasi0, 2025-03-19:
```
esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 16MB
  framework:
    type: arduino
captive_portal:
i2c:
  - id: i2c_bus
    sda: 1
    scl: 2
    scan: true
esp32_camera:
  name: My Camera
  external_clock:
    pin: GPIO5
    frequency: 20MHz
  i2c_id: i2c_bus
  data_pins: [GPIO16, GPIO18, GPIO8, GPIO17, GPIO15, GPIO6, GPIO4, GPIO9]
  vsync_pin: GPIO3
  href_pin: GPIO46
  pixel_clock_pin: GPIO7
  resolution: 640x480
```

[image]

[image]
The camera still isn't working. I see that the HA entity status is idle, but there's no image. From the logs, I see: Setup Failed: ESP_ERR_NOT_FOUND. The component esp32_camera is marked as FAILED.

NonaSuomy, 2025-03-19:
Hmm in the material it also says this is sda and sdl for the cam

[image]