esphome/issues #5095: Web server doesn't support IPv6 when using the Arduino framework

dokutan, 2023-11-10:
### The problem

The web server doesn't listen on IPv6 when using the arduino framework, esp-idf does work. I confirmed that the port is closed using nmap on the global and link local address.

### Which version of ESPHome has the issue?

2023.10.6

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2023.10.1

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

web_server

### Example YAML snippet

```yaml
esphome:
  name: test

esp32:
  board: esp32dev
  framework:
    type: arduino

network:
  enable_ipv6: true

web_server:
  port: 80
  local: true

sensor:
  - platform: uptime
    name: Uptime Sensor
```

HeMan, 2024-03-04:
Could you test with the dev branch?

HeMan, 2024-03-21:
Resolved in 2024.3.0

closed 2024-03-21 by HeMan