esphome/issues #5098: http_request response body truncated

gituser-rk, 2023-11-11:
### The problem

when requesting a status json from a gen1 shelly with the http_request component I get a truncated json body back.

i.e. http://192.168.178.58/status gives back this json (which is 1145 byte):
{"wifi_sta":{"connected":true,"ssid":"Tosse_T","ip":"192.168.178.58","rssi":-79},"cloud":{"enabled":true,"connected":true},"mqtt":{"connected":false},"time":"22:39","unixtime":1699738764,"serial":24707,"has_update":true,"mac":"E8DB84D7S828","cfg_changed_cnt":17,"actions_stats":{"skipped":0},"relays":[{"ison":true,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"source":"cloud"}],"meters":[{"power":0.00,"overpower":0.00,"is_valid":true,"timestamp":1699742364,"counters":[0.000, 0.000, 0.000],"total":83231597}],"inputs":[{"input":0,"event":"","event_cnt":0}],"temperature":22.24,"overtemperature":false,"tmp":{"tC":22.24,"tF":72.03, "is_valid":true},"temperature_status":"Normal","ext_sensors":{"temperature_unit":"C"},"ext_temperature":{"0":{"hwID":"28f8ca75d0034cdd","tC":5.06,"tF":41.11}},"ext_humidity":{},"update":{"status":"pending","has_update":true,"new_version":"20230913-113709/v1.14.0-gcb84623","old_version":"20220209-094317/v1.11.8-g8c7bb8d","beta_version":"20231107-164450/v1.14.1-rc1-g0617c15"},"ram_total":50456,"ram_free":37260,"fs_size":233681,"fs_free":148843,"uptime":52964766}

when logging the response body I saw only this fragment:
{"wifi_sta":{"connected":true,"ssid":"Tosse_T","ip":"192.168.178.58","rssi":-79},"cloud":{"enabled":true,"connected":true},"mqtt":{"connected":false},"time":"22:39","unixtime":1699738764,"serial":24707,"has_update":true,"mac":"E8DB84D7S828","cfg_changed_cnt":17,"actions_stats":{"skipped":0},"relays":[{"ison":true,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"source":"cloud"}],"meters":[{"power":0.00,"overpower":0.00,"is

you see, it's truncated after 942 characters. 

### Which version of ESPHome has the issue?

2023.10.6

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

lolin32

### Component causing the issue

http_request

### Example YAML snippet

```yaml
esphome:
  name: wallclock-yvonne
  friendly_name: wallclock-yvonne

time:
  - platform: sntp
    id: ntp
    timezone: Europe/Berlin
    servers: 172.16.1.1
    on_time:
      - seconds: 0
        minutes: /15
        then:
          - http_request.get: 
              url: http://192.168.178.58:80/status
              verify_ssl: false
              on_response:
                then:
                  - logger.log:
                      format: 'Response status: %d, Duration: %u ms'
                      args:
                        - status_code
                        - duration_ms
                  - lambda: |-
                      if (status_code == 200) {
                        const char* httpResponse = id(http_request_data).get_string();
                        std::string httpResponseStr = httpResponse;
                        //ESP_LOGD("http_response", "Response %s", httpResponseStr.c_str());
                        
                        json::parse_json(httpResponseStr, [](JsonObject root) {

                          id(temperature).publish_state(root["ext_temperature][0]["tC"]);
                        });                          
                      } else {
                        //ESP_LOGD("http_response", "Failed connection to remote sensor. Status code: %d", status_code);
                        id(temperature).publish_state(99);
                      }              

esp32:
  board: lolin32
  framework:
    type: arduino

# Enable logging
logger:

api:
  reboot_timeout: 0s

ota:
  password: "dh7shdas7dsdhsd"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "Wallclock"
    password: "GDgfd3dghsfmsfsaksKff"

captive_portal:

spi:
  clk_pin: 18
  mosi_pin: 23

sensor:
  - platform: template
    name: "temperature"
    id: temperature
  - platform: adc
    pin: 33
    name: "LDR"
    id: lumiosity
    update_interval: 0.2s
    filters:
    - multiply: 12

http_request:
  useragent: esphome/device
  timeout: 2s
  id: http_request_data
  

font:
  - file: "userdata_fonts/Arial-Narrow.ttf"
    id: digit_font1
    #size: 9
    size: 11
    glyphs:
       [':', '°', '0', '1', '2', '3', '4', '5', '6', '7', '8', '9', 'C', '.', ',', 'a', 'n']
display:
  - platform: max7219digit
    id: display_max
    setup_priority: -100
    cs_pin: 15
    num_chips: 6
    intensity: 5
    scroll_speed: 20ms
    scroll_delay: 5s
    scroll_dwell: 1s
    scroll_mode: CONTINUOUS
    lambda: |-
       it.intensity((int)id(lumiosity).state);
       it.strftime(24, -3, id(digit_font1), "%H:%M", id(ntp).now());
        if (id(temperature).state <= -10.0) {
          //too long, save some space
          it.printf(0, -1, id(digit_font1), "%.1f°", id(temperature).state);                        
        } else if (id(temperature).state < 10.0 && id(temperature).state >= 0.0) {
          it.printf(0, -1, id(digit_font1), "   %.1f°", id(temperature).state);     
        } else if (id(temperature).state < 0.0 && id(temperature).state > -10.0) {
          it.printf(0, -1, id(digit_font1), "  %.1f°", id(temperature).state);
        } else {
          it.printf(0, -1, id(digit_font1), " %.1f°", id(temperature).state);
        }
```

### Additional information

when changing the URL to a openweathermap API call, I'm able to retrieve the temperature within the jsaon with this code. but the response body (json) size is only below 500 byte ...

creepystefan, 2024-03-04:
https://forum.arduino.cc/t/shelly-pro-3em-esp8266-json/1152558/3

Shelly 3EM Pro      Generation 2

["em:0"]["total_act_power"]

http_request:
  useragent: esphome/device
  timeout: 5s
  id: getrequest

sensor:
  - platform: template
    name: testrequest
    id: testrequest

interval:
  - interval: 1s
    then:
      - http_request.get:
          url: http://IP-ADRESS-FROM-SHELLY/rpc/Shelly.GetStatus
          headers:
            Content-Type: application/json
          on_response:
            then:
              - lambda: |-
                  json::parse_json(id(getrequest).get_string(), [](JsonObject root) {
                      id(testrequest).publish_state(root["em:0"]["total_act_power"]);
                  });

creepystefan, 2024-03-04:
ohh sorry is not for your Problem, maybe it helpes

ssieb, 2024-03-04:
The log output has a limited length per line.

atus42, 2024-11-02:
I have a similar case (2024.8.3). A multiline html page gets truncated way before 942 char. My html is only 437 byte long:

`<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Transitional//EN" "http://www.w3.org/TR/xhtml1/DTD/xhtml1-transitional.dtd">`
`<html xmlns="http://www.w3.org/1999/xhtml">`
`<head>`
`<meta http-equiv="Content-Type" content="text/html; charset=UTF-8" />`
`<title>Welcome page</title>`
`</head>`
`<body>`
`almafa kortebokor fuszal`
`1234567890`
`1234567890`
`1234567890`
`1234567890`
`1234567890`
`1234567890`
`1234567890`
`12<!_HERE_!>34567890`
`1234567890`
`1234567890`
`</body>`
`</html>`

And it were truncated at the 8th line of 1234567890 marked with <!_HERE_!>, what count only 402 byte.

Moreover if I try to download a 400 byte xml from an inverter, it gets -1 byte data and crashes the whole board with a forever loop saying some pointer is gone.

palsbo, 2025-02-05:
This will only return 1000 bytes in body

http_request:
  id: http_request_component  
  buffer_size_rx: 8192  # Prøv større buffer
  buffer_size_tx: 512
  timeout: 20s

time:
  - platform: sntp
    id: my_time
    timezone: "Europe/Copenhagen"
    on_time:
      - seconds: 0,20,40
        then:
          - switch.turn_on: led2
          - delay: "00:00:02"
          - switch.turn_off: led2
          - http_request.get:
              url: "http://192.168.1.48"
              capture_response: true
              on_response:
                then:
                  - if:
                      condition:
                          lambda: return response->status_code == 200;
                      then:
                          - logger.log:
                              format: "Response receied status: %d, Length: %d, message \n%s"
                              args: [ 'response->status_code', 'body.length()', 'body.c_str()' ]
                      else:
                          - logger.log:
                              format: "Error: Response status: %d, message %s"
                              args: [ 'response->status_code', 'body.c_str()' ]

log:
[12:37:42][D][main:081]: Response receied status: 200, Length: 1000, message 

{"total":24,"filters":"{\"PriceArea\":\"DK2\"}","sort":"HourDK ASC","dataset":"Elspotprices","records":[{"HourUTC":"2024-05-26T22:00:00","HourDK":"2024-05-27T00:00:00","PriceArea":"DK2","SpotPriceDKK":584.469971,"SpotPriceEUR":78.339996},{"HourUTC":"2024-05-26T23:00:00","HourDK":"2024-05-27T01:00:00","PriceArea":"DK2","SpotPriceDKK":546.570007,"SpotPriceEUR":73.260002},{"HourUTC":"2024-05-27T00:00:00","HourDK":"2024-05-27T02:00:00"
[12:37:42][W][component:237]: Component time took a long time for an operation (309 ms).
[12:37:42][W][component:238]: Components should block for at most 30 ms.