esphome/issues #5099: bad mdns name use when name_add_mac_suffix: true

jflefebvre06, 2023-11-12:
### The problem

When `name_add_mac_suffix` is set to true, during wireless installation, esphome try to connect `${name}.local` instead of `${name}-MAC.local`

### Which version of ESPHome has the issue?

2023.10.6 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

ALL

### What platform are you using?

ESP32

### Board

m5stack-atom-lite

### Component causing the issue

wifi

### Example YAML snippet

```yaml
substitutions:
  name: atom-bluetooth-proxy

esphome:
  name: ${name}
  name_add_mac_suffix: true
  project:
    name: esphome.bluetooth-proxy
    version: "1.0"

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

api:
logger:
ota:
improv_serial:
# captive_portal:

dashboard_import:
  package_import_url: github://esphome/firmware/bluetooth-proxy/m5stack-atom-lite.yaml@main

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true

button:
- platform: safe_mode
  name: Safe Mode Boot
  entity_category: diagnostic

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: ${name}-0d350c.local
  #domain: .local

binary_sensor:
  # statut
  - platform: status
    name: "Statut"

sensor:
  # Qualité du signal
  - platform: wifi_signal
    name: "Signal Wifi"
    update_interval: 60s
  # Temps de fonctionnement
  - platform: uptime
    name: "Allumé depuis (s)"
    id: uptime_sec

switch:
  # Bouton de redémarrage
  - platform: restart
    name: "Redémarrage"

# Transformation des secondes en jours
text_sensor:
  - platform: template
    name: "Allumé depuis (j)"
    lambda: |-
      int seconds = (id(uptime_sec).state);
      int days = seconds / (24 * 3600);
      seconds = seconds % (24 * 3600); 
      int hours = seconds / 3600;
      seconds = seconds % 3600;
      int minutes = seconds /  60;
      seconds = seconds % 60;
      return { (to_string(days) +"d " + to_string(hours) +"h " + to_string(minutes) +"m "+ to_string(seconds) +"s").c_str() };
    icon: mdi:clock-start
    update_interval: 60s

  - platform: wifi_info
    ip_address:
      name: ESP IP Address
    ssid:
      name: ESP Connected SSID
    bssid:
      name: ESP Connected BSSID
    mac_address:
      name: ESP Mac Wifi Address
    scan_results:
      name: ESP Latest Scan Results
```

### Anything in the logs that might be useful for us?

```txt
Failed to conenct to atom-bluetooth-proxy.local
```

ssieb, 2023-11-12:
Are you a "creator"?  That option and the dashboard import are not intended for end users and there is a warning in the docs (that really should be clearer).  The dashboard does not know the mac address of the device, so it has no way of knowing the correct name to use.

closed 2023-11-12 by ssieb

jflefebvre06, 2023-11-13:
I m  not creator, do you mean I muste remove this from my template ?

```
dashboard_import:
  package_import_url: github://esphome/firmware/bluetooth-proxy/m5stack-atom-lite.yaml@main
```