esphome/issues #5102: Logger is not thread safe

rguca, 2023-11-13:
### The problem

If a task is created using `xTaskCreate()` and the task creates log entries with `ESP_LOG`, not all will be visible in the log.

Reason:
The logger omits the entry, if another task is creating an entry at the same time.
`recursion_guard_`
https://github.com/esphome/esphome/blob/45276cc24452b7ea1a7a0c6f88ed11ca38a4a1e6/esphome/components/logger/logger.cpp#L67

Possible fix:
Replace `recursion_guard_` with a mutex created by `xSemaphoreCreateMutex` and claim/release it with `xSemaphoreTake` and `xSemaphoreGive`.

Working PoC:
```
static SemaphoreHandle_t log_guard = xSemaphoreCreateMutex();

void HOT Logger::log_vprintf_(int level, const char *tag, int line, const char *format, va_list args) {  // NOLINT
  if (level > this->level_for(tag))
    return;
  xSemaphoreTake(log_guard, portMAX_DELAY);
  this->reset_buffer_();
  this->write_header_(level, tag, line);
  this->vprintf_to_buffer_(format, args);
  this->write_footer_();
  this->log_message_(level, tag);
  xSemaphoreGive(log_guard);
}
```

### Which version of ESPHome has the issue?

2023.10.6

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

logger

bdraco, 2023-11-14:
maybe related PR https://github.com/esphome/esphome/pull/5373

bdraco, 2025-05-09:
I see disconnects when logging is runner because the protobuf messages get corrupted
```
2025-05-09 13:42:46.565 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/opt/homebrew/Cellar/python@3.13/3.13.3/Frameworks/Python.framework/Versions/3.13/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "aioesphomeapi/_frame_helper/plain_text.py", line 48, in aioesphomeapi._frame_helper.plain_text.APIPlaintextFrameHelper.data_received
  File "aioesphomeapi/_frame_helper/plain_text.py", line 75, in aioesphomeapi._frame_helper.plain_text.APIPlaintextFrameHelper.data_received
  File "aioesphomeapi/connection.py", line 916, in aioesphomeapi.connection.APIConnection.process_packet
  File "aioesphomeapi/connection.py", line 892, in aioesphomeapi.connection.APIConnection.process_packet
google.protobuf.message.DecodeError: Error parsing message with type 'SubscribeLogsResponse'
2025-05-09 13:42:46.567 ERROR (MainThread) [homeassistant.components.inkbird.coordinator] 49:22:08:15:00:BB: Bluetooth error whilst polling: Invalid protobuf message: type=SubscribeLogsResponse data=b'\n\x08\xe1\xd4\x96\xfc\xc6\xa4/\x10Q"\r\x02\x01\x06\t\xffHQ364656]\x1b[1;31m[BTU_TASK]\x1b[0;36m: \x1b[0;33mW (876448) BT_APPL: gattc_conn_cb: if=3 st=0 id=259 rsn=0x16\x1b[0m\n\x1b[0m': Error parsing message with type 'SubscribeLogsResponse'
```

bdraco, 2025-05-09:
something like this
```
diff --git a/esphome/components/logger/logger.cpp b/esphome/components/logger/logger.cpp
index 57f0ba9f9..7dfe97836 100644
--- a/esphome/components/logger/logger.cpp
+++ b/esphome/components/logger/logger.cpp
@@ -59,24 +59,49 @@ void Logger::write_header_(int level, const char *tag, int line) {
 }
 
 void HOT Logger::log_vprintf_(int level, const char *tag, int line, const char *format, va_list args) {  // NOLINT
-  if (level > this->level_for(tag) || recursion_guard_)
+  if (level > this->level_for(tag))
     return;
 
+#if defined(USE_ESP32) || defined(USE_ESP8266)
+  // Use mutex for thread safety on ESP platforms
+  xSemaphoreTake(this->log_mutex_, portMAX_DELAY);
+#else
+  // Use recursion guard on other platforms
+  if (recursion_guard_)
+    return;
   recursion_guard_ = true;
+#endif
+
+  // Common logging code
   this->reset_buffer_();
   this->write_header_(level, tag, line);
   this->vprintf_to_buffer_(format, args);
   this->write_footer_();
   this->log_message_(level, tag);
+
+#if defined(USE_ESP32) || defined(USE_ESP8266)
+  xSemaphoreGive(this->log_mutex_);
+#else
   recursion_guard_ = false;
+#endif
 }
 #ifdef USE_STORE_LOG_STR_IN_FLASH
 void Logger::log_vprintf_(int level, const char *tag, int line, const __FlashStringHelper *format,
                           va_list args) {  // NOLINT
-  if (level > this->level_for(tag) || recursion_guard_)
+  if (level > this->level_for(tag))
     return;
 
+#if defined(USE_ESP32) || defined(USE_ESP8266)
+  // Use mutex for thread safety on ESP platforms
+  xSemaphoreTake(this->log_mutex_, portMAX_DELAY);
+#else
+  // Use recursion guard on other platforms
+  if (recursion_guard_)
+    return;
   recursion_guard_ = true;
+#endif
+
+  // Common logging code
   this->reset_buffer_();
   // copy format string
   auto *format_pgm_p = reinterpret_cast<const uint8_t *>(format);
@@ -86,8 +111,14 @@ void Logger::log_vprintf_(int level, const char *tag, int line, const __FlashStr
     this->tx_buffer_[this->tx_buffer_at_++] = ch = (char) progmem_read_byte(format_pgm_p++);
   }
   // Buffer full form copying format
-  if (this->is_buffer_full_())
+  if (this->is_buffer_full_()) {
+#if defined(USE_ESP32) || defined(USE_ESP8266)
+    xSemaphoreGive(this->log_mutex_);
+#else
+    recursion_guard_ = false;
+#endif
     return;
+  }
 
   // length of format string, includes null terminator
   uint32_t offset = this->tx_buffer_at_;
@@ -97,7 +128,12 @@ void Logger::log_vprintf_(int level, const char *tag, int line, const __FlashStr
   this->vprintf_to_buffer_(this->tx_buffer_, args);
   this->write_footer_();
   this->log_message_(level, tag, offset);
+
+#if defined(USE_ESP32) || defined(USE_ESP8266)
+  xSemaphoreGive(this->log_mutex_);
+#else
   recursion_guard_ = false;
+#endif
 }
 #endif
 
@@ -140,6 +176,10 @@ Logger::Logger(uint32_t baud_rate, size_t tx_buffer_size) : baud_rate_(baud_rate
 #if defined(USE_ESP32) || defined(USE_LIBRETINY)
   this->main_task_ = xTaskGetCurrentTaskHandle();
 #endif
+#if defined(USE_ESP32) || defined(USE_ESP8266)
+  // Create mutex for thread-safe logging
+  this->log_mutex_ = xSemaphoreCreateMutex();
+#endif
 }
 
 #ifdef USE_LOGGER_USB_CDC
diff --git a/esphome/components/logger/logger.h b/esphome/components/logger/logger.h
index c4c873e02..e2fb30b0a 100644
--- a/esphome/components/logger/logger.h
+++ b/esphome/components/logger/logger.h
@@ -169,8 +169,12 @@ class Logger : public Component {
   std::map<std::string, int> log_levels_{};
   CallbackManager<void(int, const char *, const char *)> log_callback_{};
   int current_level_{ESPHOME_LOG_LEVEL_VERY_VERBOSE};
-  /// Prevents recursive log calls, if true a log message is already being processed.
+#if defined(USE_ESP32) || defined(USE_ESP8266)
+  SemaphoreHandle_t log_mutex_{nullptr};  // Mutex for thread-safe logging
+#else
+  /// Prevents recursive log calls within same thread (not thread-safe across multiple threads)
   bool recursion_guard_ = false;
+#endif
   void *main_task_ = nullptr;
   CallbackManager<void(int)> level_callback_{};
 };
```

bdraco, 2025-05-09:
Locking is generally not great with logging though as with high level of logging it can eat 10-20% of the cpu time ... but thats hopefully a rare case that we would log that often

bdraco, 2025-05-12:
```
void* currentValue = pvTaskGetThreadLocalStoragePointer(NULL, tls_index);
if (currentValue != nullptr) {
    // The current task is already logging
    return;
}
vTaskSetThreadLocalStoragePointer(NULL, tls_index, currentTask);
...logging code...
vTaskSetThreadLocalStoragePointer(NULL, tls_index, nullptr);
```

That sound be far lower overhead and only prevent recursion in the current task.

bdraco, 2025-05-12:
Should be able to make those changes now that the other issue is solved.

closed 2025-05-15 by jesserockz