esphome/issues #5104: E1.31 not responding after update

mhentschke, 2023-11-14:
### The problem

When using E1.31 on WLAN connected ESP8266 for addressable led strip with Hyperion NG. Updating ESPHome to V2023.6.0 or later breaks the E1.31 integration, probably by https://github.com/esphome/esphome/pull/4832#issue-1710788306. 

Issue https://github.com/esphome/issues/issues/4645 has been closed as fixing the issue but it seems the fix only works for ESP32 and not ESP8266.

The problem persists in 2023.10.6 and also in both beta and dev branches

### Which version of ESPHome has the issue?

2023.10.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.2

### What platform are you using?

ESP8266

### Board

ESP-01

### Component causing the issue

E1.31

### Example YAML snippet

```yaml
esphome:
  name: backlight
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "SSID"
  password: "pass"
  use_address: XXX.XXX.XXX.XXX

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "SSID"
    password: "pass"

captive_portal:

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "key"

ota:
  password: "pass"

mqtt:
  broker: XXX.XXX.XXX.XXX
  username: "username"
  password: "pass"
  
wled:
e131:
  method: unicast

light:
  - platform: neopixelbus
    name: Ultrawide_backlight
    pin: 2
    num_leds: 49
    variant: WS2812
    effects:
      - addressable_color_wipe:
      - addressable_twinkle:
      - addressable_fireworks:
      - addressable_flicker:
      - flicker:
          name: Flicker Effect With Custom Values
          alpha: 95%
          intensity: 1.5%
      - random:
          name: "Slow Random"
          transition_length: 30s
          update_interval: 30s
      - random:
          name: "Fast Random"
          transition_length: 2s
          update_interval: 2s
      - addressable_rainbow:
      - wled:
          port: 21324
      - e131:
          name: "Bias"
          universe: 1
          channels: RGB
      - e131:
          name: "Sound"
          universe: 2
          channels: RGB
```

### Anything in the logs that might be useful for us?

```txt
Even with very vebose logs, I see no mention of the E131 component nor any errors related to it
```

daandav, 2023-11-15:
Can confirm that the issue is **not** fixed on ESP8266.

mhentschke, 2023-11-16:
I did see you changed the library but I don't think there is such an option in ESP8266 right? changing to FastLED might be an option but it will force me to downgrade the Arduino framework which is not ideal and I'm still trying to get it to work.

hcoohb, 2023-11-16:
Same issue here on esp8266...

RobertCoop, 2023-12-06:
If you are using HomeAssistant's ESPHome integration and are running into this error, you can use this addon to install legacy versions of ESPHome (in parallel with the current version) and update the chip OTA with a working version.  I installed a duplicate legacy of 2023.4 and switched to unicast and was able to solve this issue.

https://github.com/khenderick/esphome-legacy-addons

Cosmostrator, 2023-12-24:
Can confirm this is still an issue on esp8266.  For reference I'm running a Kauf RGBWW smart bulb and using the partition light platform.  It was working before the update with multicast. I can post the yaml if it helps.

d-rez, 2024-02-24:
I can also confirm, got a couple ESP8266-based strips and it choosing the E1.31 Effect just doesn't do anything, tried multiple apps to control it.

Is there any known workaround without spinning up an older version of ESPHome container?

DrBlokmeister, 2024-04-25:
Commenting here that I'm also suffering from this issue and would like to see this fixed. ESP32 responding just fine, two separate ESP8266s are not.

mikosoft83, 2024-09-24:
Just to bump this, I have the same issue.
I stumbled upon a temporary solution when researching this, there's an addon that can bring up older ESPHome versions. With 2023.5 I was able to get it working again. Link to the repo:
https://github.com/khenderick/esphome-legacy-addons

SparkyDan555, 2024-10-25:
Also got the same issue here. Working with 2023.5 but not 2024.10. It would be great to see this feature fixed.

MattWestb, 2024-12-13:
I think its "little" ignorant from the maintainers flagging the original issue fixed without testing it on all platform and then not responding on the original closed issue and over one year on this.
For me ESP8266 / D1 Mini is great for this application but i think i must getting some C/S3 Minis but it shall normally not being needed if all is / was working well.

CC @balloob

ItsOktawian, 2025-01-03:
issue still not fixed @ ESP8266

daandav, 2025-01-03:
@jesserockz old issue was incorrectly closed. People are already waiting for a fix for 1.5 years.

maciejla, 2025-01-04:
The problem still exists. E1.31 works fine with xLights on ESPHome 2023.4.4, however I cannot use that version as multiple PCA9685 drivers refuse to work on it. Willing to put a bounty on the issue if that's possible. Otherwise, if the issue is unlikely to be resolved soon - please say so. I will switch my controllers to ESP32.

maciejla, 2025-01-04:
Okay, I was able to use something else as a workaround. Thanks to Brian from @KaufHA it's possible to use a custom component that makes the lights available via DDP. Works great on ESPHome 2024.12.2 and xLights 2024.20.1 https://github.com/KaufHA/common/blob/main/components/README.md#ddp

bkaufx, 2025-12-24:
This should be tagged as an issue with the socket component.  The socket component does not work for esp8266 when used as is being done in the e131 component.  I've tried to get socket to work with my custom ddp component and I run into the same issue.  The socket component just doesn't receive anything ever on esp8266.