esphome/issues #5106: Smart Meter Interface - SML has issues decoding obis_code "1-0:96.1.0"

stoffelll, 2023-11-15:
### The problem

I installed yesterday my new IR reading head with an ESP01s running esphome for my smart meter. This new reading head with wifi is an replacement for my existing one, which was connected previously via USB to my PI 4 which is running home assistant.

Unfortunately I experienced regular "crashes" (values in HA Dashboard were available/shown for some miliseconds and afterwards disappearing again). This behaviour started when I added a text sensor for the obis_code: 1-0:96.1.0" (metering ID/serial number). When I removed it again, I got stable values in the HA Dashboard.

Using the "old" IR reading head (with USB), it worked well and got the correct metering ID.

This is my config:

```
  uart:
  id: uart_bus
  tx_pin: 1
  rx_pin: 3
  baud_rate: 9600
  debug:

sml:
  id: mysml
  uart_id: uart_bus
  
text_sensor:  
  - platform: sml
    name: "Metering ID"
    sml_id: mysml
    #server_id: "0123456789abcdef"
    obis_code: "1-0:96.1.0"
```

And this is the value I get via debug:
```
[22:48:38][D][sml:066]: OBIS info:
[22:48:38][D][sml:072]:   (0a0149534b000479f6f9) 1-0:96.50.1 [0x49534b]
[22:48:38][D][sml:072]:   (0a0149534b000479f6f9) 1-0:96.1.0 [0x0a0149534b000479f6f9]
[22:48:38][D][sml:072]:   (0a0149534b000479f6f9) 1-0:1.8.0 [removed]
[22:48:38][D][sml:072]:   (0a0149534b000479f6f9) 1-0:2.8.0 [removed]
[22:48:38][D][sml:072]:   (0a0149534b000479f6f9) 1-0:16.7.0 [removed]
```

and this was causes the "crash":
`[22:48:38][D][sml:072]:   (0a0149534b000479f6f9) 1-0:96.1.0 [0x0a0149534b000479f6f9]`

### Which version of ESPHome has the issue?

2023.10.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.2

### What platform are you using?

ESP8266

### Board

ESP01S

### Component causing the issue

SML

hmichels, 2024-02-13:
Hello!
I have similar the same problem, the OBIS identifier "1-0:96.1.0" cannot be interpreted as text (roughly the HEX code as above).

Does anyone have a solution for this?

Abdull, 2024-08-18:
@stoffelll :

> ```
> [22:48:38][D][sml:072]:   (0a0149534b000479f6f9) 1-0:96.1.0 [0x0a0149534b000479f6f9]
> ```

May it be you are/were reporting data from an iskra smart meter? Because `0x0a0149534b000479f6f9` decodes as follows:

* Line Feed (LF) character (0a)
* Start of Heading (SOH) control character (01)
* ASCII representation of "ISK"
* Null character (00)
* End of Transmission (EOT) control character (04)
* ASCII representation of "y"
* The last two bytes (f6 and f9) are outside the standard ASCII range and could represent extended ASCII characters, part of a multi-byte encoding, or some kind of checksum or special data

I have the same symptoms as described by you, caused by an Apator smart meter. For OBIS identifier "1-0:96.1.0", ESPHome reports a similar telegram with LF, SOH, and EOT control characters, just as the middle three-letter representation, it gives "APA" (as in "Apator").

I have the feeling that some software component within the ESPHome / SML stack is incorrectly unwrapping payload data.

Regarding "*values in HA Dashboard were available/shown for some miliseconds and afterwards disappearing again*": in my system the root cause seems to be an exception happening on the *Home Assistant* side:

``` journalctl
# ...
# the following log entries repeat again and again
# ...
Aug 17 18:25:57 my-ha-server hass[996]: 2024-08-17 18:25:57.858 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successfully connected to my-esp32-device @ 192.168.11.22 in 0.045s
Aug 17 18:25:58 my-ha-server hass[996]: 2024-08-17 18:25:58.155 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successful handshake with my-esp32-device @ 192.168.11.22 in 0.298s
Aug 17 18:26:01 my-ha-server hass[996]: 2024-08-17 18:26:01.433 ERROR (MainThread) [aioesphomeapi.connection] my-esp32-device @ 192.168.11.22: Invalid protobuf message: type=TextSensorStateResponse data=b'\r\xbf`\xd6!\x12\n\n\x01APA\x01\x04\xe6\x00\xcb': Error parsing message
Aug 17 18:26:01 my-ha-server hass[996]: Traceback (most recent call last):
Aug 17 18:26:01 my-ha-server hass[996]:   File "aioesphomeapi/connection.py", line 893, in aioesphomeapi.connection.APIConnection.process_packet
Aug 17 18:26:01 my-ha-server hass[996]: google.protobuf.message.DecodeError: Error parsing message
Aug 17 18:26:01 my-ha-server hass[996]: 2024-08-17 18:26:01.434 WARNING (MainThread) [aioesphomeapi.connection] my-esp32-device @ 192.168.11.22: Connection error occurred: Invalid proto
buf message: type=TextSensorStateResponse data=b'\r\xbf`\xd6!\x12\n\n\x01APA\x01\x04\xe6\x00\xcb': Error parsing message
# ...
# above log entries repeat again and again and again
# ...
```

My assumption:
* Smart meter produces new "telegram" once per second, including the 1-0:96.1.0 data.
* ESPHome stack incorrectly processes 1-0:96.1.0 value, creates **unvalidated** protobuf message from this garbled value, and sends it to Home Assistant API.
* Home Assistant receives message, **parses/validates it**, notices it stinks and crashes.
* Repeat once per second with new smart meter telegram; and/or ESPHome retries delivering the same garbled data. Latter would explain why the Home Assistant dashboard flickers between showing "Unknown" and any previous correct data *even when removing the IR sensor from the smart meter sender.

stoffelll, 2024-08-18:
Correct, it’s an iskra smart meter, but as you have the same issue it seems to be a generic issue.

CarstenGrohmann, 2024-12-07:
I see the same issue with my configuration:

```
text_sensor:
  - platform: sml
    name: "Manufacturer"
    sml_id: sml_instance
    obis_code: "1-0:96.50.1"

  - platform: sml
    obis_code: "1-0:96.1.0"
    name: "Serial Number"
```

esphome send OBIS values (but not the code) to HA

```
[21:50:15][D][sml:078]: OBIS info:
[21:50:15][D][sml:085]:   (0a014546522003b292fe) 1-0:96.50.1 [0x454652] type: 0
[21:50:15][D][sml:085]:   (0a014546522003b292fe) 1-0:96.1.0 [0x0a014546522003b292fe] type: 0

[21:50:15][V][text_sensor:013]: 'Manufacturer': Received new state EFR
[21:50:15][D][text_sensor:064]: 'Manufacturer': Sending state 'EFR'
[21:50:15][V][mqtt:496]: Publish(topic='electricmeter/sensor/manufacturer/state' payload='EFR' retain=1 qos=0)
[21:50:15][V][text_sensor:013]: 'Serial Number': Received new state 0x0a014546522003b292fe
[21:50:15][D][text_sensor:064]: 'Serial Number': Sending state '0x0a014546522003b292fe'
[21:50:16][V][mqtt:496]: Publish(topic='electricmeter/sensor/serial_number/state' payload='0x0a014546522003b292fe' retain=1 qos=0)
```

and HA failed to process:

```
2024-12-07 08:39:33.059 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'\n\x01EFR \x03\xb2\x92\xfe' on electricmeter/sensor/serial_number/state with encoding utf-8 (for <Job None HassJobType.Callback functools.partial(<bound method MqttEntity._message_callback of <entity sensor.electricmeter_serial_number=unknown>>, <bound method MqttSensor._state_message_received of <entity sensor.electricmeter_serial_number=unknown>>, {'_attr_native_value', '_expired', '_attr_last_reset'})>)
```

Who is responsible for processing this data? HA or ESPHome?

I assume ESPHome, as the OBIS code is not shared with HA and a decoding is not possible without that code.

Morg42, 2025-01-13:
FYI, I don't use HA, but stumbled on this OBIS code also... incidentally, this is possible on Iskra, ZPA and EMH meters...

my EMH:
0a01454d480000e66f1d
->
0a (unknown)
01 = (int) 1
45 4d 48 = (str) EMH
00 = (int) 0
00 = (int) 0
e66f1d = (dec) 15101725 

The smartmeter number as printed on the front is
1 EMH00 1510 1725

Too much to be incidental, I think.

For your serial, I'm not sure, as counter numbers usually are 8 digits -

0a0149534b000479f6f9
->
0a (unknown)
01 = (int) 1
49 53 4b = (str) ISK
00 = (int) 0
04 = (int) 4
79f6f9 = (dec) 07993081

So your smartmeter number might be
1 ISK04 0799 3081

Maybe so, maybe wrong. But if you cross check, we might find a possible decoding rule...

stoffelll, 2025-01-13:
Hi Morg42,

Unfortunately it is not fitting...

My meter id starts with: 
1 ISK00 7510 xxxx

Best
SL

Morg42, 2025-01-13:
Okay, so...

```
def conv_id(id: str) -> str:
    if len(id) != 20:
        print(f'wrong len: {len(id)}')
        return ''

    sparte = int(id[2:4])
    mfg = bytearray.fromhex(id[4:10]).decode()
    fab = int(id[10:12])
    ser = int(id[12:20], 16)

    return f'{sparte}{mfg}{fab:02d}{ser:08d}'
```

Seems to work for all serials/ids I could get a hold on (and was able to verify...)

elemakil, 2025-04-09:
Based on @Morg42 python code, I have added the following to my esphome yaml configuration to get this to work properly:

````
text_sensor:
  # Workaround for https://github.com/esphome/issues/issues/5106
  - platform: sml
    id: serial_number_raw
    internal: true
    sml_id: smlbus
    obis_code: "1-0:96.1.0"
    format: hex

  - platform: template
    name: "Serial Number"
    icon: mdi:numeric
    lambda: |-
      const auto value = id(serial_number_raw).state;
      if (value.length() != 22 || !std::equal(value.begin(), value.begin() + 4, "0x0a")) {
          return "Unknown: " + value;
      }

      int sparte = std::stoi(value.substr(4, 2));
      std::string mfg;
      for (int i = 6; i < 12; i+=2) {
          mfg += static_cast<char>(std::stoi(value.substr(i, 2), nullptr, 16));
      }
      int fab = std::stoi(value.substr(12, 2));
      int ser = std::stoi(value.substr(14, 8), nullptr, 16);

      char buffer[20];
      int len = sprintf(buffer, "%d%s%02d%08d", sparte, mfg.c_str(), fab, ser);
      return std::string(buffer, len);
````

Eagle3386, 2026-04-23:
> I don't understand what you're trying to achieve. If you want to convert a byte into text, you can use `sprintf`.

No. I want to work with the raw data, properly converted/decoded, without any `*printf()` method as we're dealing with a value which shall only be transformed - the printing is done way later, either inside the `webserver` component or HA.

> [@skleeschulte](https://github.com/skleeschulte) has posted a full blown "obis2text"-decoder in it's lambda: [#5106 (comment)](https://github.com/esphome/issues/issues/5106#issuecomment-4285755732) What info are you missing beyond that?

That's exactly what's wrong with it to begin with!
1. It's acting inside `on_value` instead of as a `filters` lambda.
2. Due to that, it has to basically "re-invent the wheel" by, e.g., doing the publishing (again) - that code is already there, but inside the `sml` component.
3. It doesn't iterate over the hex byte sequence, but selectively extracts each byte (see `manufacturer_id_*` lines) - 20 years ago, back at university, one of the first things they taught me, was this:
> If you do something very similar more than once, either write a method for it or at least use a loop, because chances are that you gonna need it even more times in the future.

What - IMHO - should be done & what I'm trying to develop:
1. Extract the first hex byte, compare it against the sequence's `size()` length minus 1 instead of hard coded comparison against `0x0a`, hence universally valid.
2. Iterate over the sequence's remainder, convert each read hex byte into ASCII, add it to the "result string" - except for the type-length byte, this already works locally & is basically just a bit shifting loop (AFAIK way faster than any `printf`, let alone printing is semantically wrong here anyway).
3. Return said string (instead of the original hex byte one) to the remaining pipeline, so that `text_sensor` with `format: text` returns a plaintext ASCII string - just like the OBIS code for manufacturer already does.

Of course, "error out" on each of the above upon failure with proper error message.

Eagle3386, 2026-04-23:
> You have obviously not read or understood how the bytes are encoded.

A hex value usually consists of 2 chars, hence 10 hex values would occupy 20 char bytes. 
However, this hex byte sequence is only 10 bytes in length, so there's obviously a different encoding applied.

> Anyway, good luck in "not reinventing the wheel". When you're done, you will finally find out that [@skleeschulte](https://github.com/skleeschulte) was faster than you.

I didn't know it's about who's first with code. I thought it's about concise, fast executable code. Rather back- (= inside `sml` component) than frontend (= YAML / HA, etc.).

But thus far you've managed to never actually help, but always only offend me - so I basically don't give a damn towards anything from you.
As I already told you elsewhere: "Nice work you did, you're gonna go far, kid".

skleeschulte, 2026-04-24:
> [@skleeschulte](https://github.com/skleeschulte): Great work!

Thank you, and thanks as well for the review and the comments!

> The local variable `char device_id[15]` shadows the ESPHome sensor ID of the same name, which works by accident because `id(device_id)` is resolved by the ESPHome macro expander before the C++ compiler sees the shadowing. Renaming `char device_id[15];` to e.g. `char formatted_id[15];` prevents this situation.

My understanding was that nothing would actually be shadowed since - as you explained - `id(...)` would be resolved during macro expansion, so this wasn't really accidental. I've renamed the variable to avoid any ambiguity nevertheless.

> What do you think about adding a check for 0 for sector id also?

Good point - I've added a check for the full sector ID byte.

In addition, after reading that "[...]ESPHome uses a single-threaded loop for all components[...]" in the [ESPHome Developer Documentation](https://developers.esphome.io/contributing/code/#c), I've changed `const std::string v = id(device_id_raw).state;` (pass by value) to `const std::string &v = id(device_id_raw).state;` (pass by reference), assuming that this is safe. Any thoughts on that?

I also polished the comments and the log output a bit.

skleeschulte, 2026-04-24:
@Eagle3386 

>     1. It's acting inside `on_value` instead of as a `filters` lambda.

Thank you for pointing out filter lambdas, that eliminates the need for a second sensor and makes the code more compact. I updated my configuration.

Additionally, I'd like to clarify a few potential misunderstandings:

- The raw value received in the SML message is a sequence of bytes.
- One way of storing a sequence of bytes is in a character sequence (string), where each character has a size of one byte. These characters do not necessarily represent encoded text (e.g. ASCII), but can simply be seen as 8-bit values. This is what happens in the SML component, [where internally the type "text" is named `SML_OCTET`](https://github.com/esphome/esphome/blob/dev/esphome/components/sml/text_sensor/__init__.py#L12).
- A hexadecimal representation is a good way to give a human-readable representation of an arbitrary byte sequence, as each byte can be represented by two hexadecimal digits. However, a string containing hexadecimal digits representing bytes requires about twice the memory compared to a string containing bytes. In addition, when you need to work with the raw bits and bytes, like converting them to other data types, a string containing hexadecimal digits is much more complicated and less efficient to work with.
- See my [comment above](https://github.com/esphome/issues/issues/5106#issuecomment-4292393934) for the actual encoding of the device ID:
  - The first byte is not simply the length of the message, but encodes both the length and the type of the remaining bytes. Thus comparing the numerical value of the first byte to the byte length of the message is not a feasible universal validation method.
  - The remaining bytes are only partially ASCII encoded. Thus converting each byte into an ASCII character will not give the desired result of the decoded device ID.
- Even if "printing" may not be the ideal term, the function `snprintf` does exactly what is needed here: it formats values of different types into a single string.

Eagle3386, 2026-04-26:
> Why? That SML record is a compound type in itself and a text sensor is something transporting "text". On the layers ESPhome is providing you should either treat it as a compound type of unknown semantics and forward it upwards as binary data or take it apart locally and generate an appropriate number of text sensors containing the individual data fields.

Nope. It's 1 OBIS message with an absolute precisely specified payload.
If the user requests to get the specific reply in plaintext as s/he wants it to be published via `text_sensor` instead of a "normal" one that could deal with "binary data", it's the component's job to do exactly that!
Otherwise s/he would've asked for the hex representation, but s/he explicitly chose `text` - it's not on you, to put the burden onto the user, to get what s/he wants, but to comply to what s/he told you, perfectly clear, regarding the expected outcome.
Code exists to simplify people's life, not require them to learn coding, so they can deal with your inability to provide them with easy to use results.

> Creating an unnecessary additional issue is redundantly unnecessary.

Yes, but the component fails to fulfill basic expectations: it publishes a message _with_ its type/length value while all it was asked to, was to deliver a darn meter ID.

You're clinging onto an incomplete, yet overly complicated technical output - that even breaks requested rules (`format: text ` makes perfectly clear that no encoded stuff shall be returned, but its _plain_ text). Now, that's what I'd like to call "unnecessary".

> This entire OBIS message is a mistake and certainly against the idea that led to the definition of the protocol. You can easily guess what kind of "engineer" invented that thing instead of doing it right and defining a group of OBIS codes. It's not like they were running out of numbers.

Yes, absolutely! 👍🏻

It's beyond stupidity, to include a manufacturer's abbreviation in a meter's ID, i.e., its unique number (if not, for the sake of simplicity, just its serial number - and I do mean _number_ literally here!). Let alone that bullshit with its type (electricity) or "manufacturing code" (easily distinguishable by serial numbers).

Yet, here we are: that OBIS code is all we got for reading serial numbers & the format engineers, in their almighty wisdom, chose to force all manufacturers to base their meter ID onto `<type><mfg. abbr.><mfg. code><serial no.>`.

Now, we can mourn that bullshit, rage against it & afterwards found ourselves in the very same spot as before.
Or we deal with it, centralize the decoding into the component & give them the biggest FU middle finger that there is in this space. 😎

I'm with the party that loves to ~fuck~ mess with such "engineers", so that I steal them their satisfying moment as hard as possible. 🥳

skleeschulte, 2026-04-27:
> Put simply, checking for `0` as the upper 4 bits's value & `a` for the lower 4 bits pretty much nails it down.

Good to see we agree on that point.

> Though, I'd probably go for checking only the lower one against `a` & then the first "payload byte" for being `1` (= electricity).
> 
> That way, I can be pretty sure it's the OBIS message for meter ID & ensure it's the correct meter type, too.

This approach is not universally applicable.

For example, consider a value of the type octet string with a payload length of 159 bytes. In this case, type and length are encoded across multiple bytes:

First byte: `1000 1010 = 0x8A`
Second byte: `0000 0001 = 0x01`

Or, consider a list with 10 elements, where the first element is an octet string of length 1:

First byte: `0111 1010 = 0x7A`
Second byte: `0000 0001 = 0x01`

> All other OBIS messages would go through as before, i.e., untouched by my code.

Which values to decode (and how) should be determined primarily by the OBIS code, not by the value itself.

> Though, TBH, neither this issue nor others came across an OBIS message in production scenarios that would fulfill this theoretic scenario we're currently talking about, right? 😉

I understood that your goal was a more general solution than just comparing the first byte to `0x0A`, so I focused my feedback accordingly.