esphome/issues #5107: Component not found: opentherm after 2023.11 update

lorenc88, 2023-11-15:
### The problem

Hello, I was using the opentherm component on my esp8266 flawlessly until the 2023.11 update of esphome. Now I am unable to recompile it without experiencing an error: **Component not found: opentherm**

I am using external repos:

`external_components:
  source: github://ov1d1u/esphome-opentherm@main`

or

`external_components:
  source: github://arthurrump/esphome-opentherm@main`

### Which version of ESPHome has the issue?

2023.11

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.2

### What platform are you using?

ESP8266

### Board

Wemos

### Component causing the issue

opentherm

### Example YAML snippet

```yaml
external_components:
  #source: github://arthurrump/esphome-opentherm@main
  source: github://ov1d1u/esphome-opentherm@main

opentherm:
  in_pin: 4
  out_pin: 5
  ch_enable: true
  dhw_enable: true
  cooling_enable: false
  otc_active: true
  ch2_active: false
 

sensor:
  - platform: opentherm
    rel_mod_level:
      name: "Boiler Relative modulation level"
    ch_pressure:
      name: "Boiler Water pressure in CH circuit"
    dhw_flow_rate:
      name: "Boiler Water flow rate in DHW circuit"
    t_boiler:
      name: "Boiler water temperature"
      filters:
      - median:
          window_size: 10
          send_every: 10
          send_first_at: 10
      - skip_initial: 3
```

randybb, 2023-11-15:
Don't see any breaking change in esphome, so maybe this component is just having issues with python 3.11 - such issue is described here https://github.com/arthurrump/esphome-opentherm#component-not-found-opentherm
Since the issue is only with this component, then this is the right place: https://github.com/arthurrump/esphome-opentherm/issues/10

ssieb, 2023-11-15:
If you do a compile, you should get more information about why it's failing.  But as mentioned, this isn't the right place for issues with external components.

closed 2023-11-15 by ssieb

lorenc88, 2023-11-16:
> @lorenc88 can you give a try to `github://ov1d1u/esphome-opentherm@dev`?

Still cannot recompile getting:
`Platform not found: 'switch.opentherm'. `