esphome/issues #5108: ESPHome doesn't update the device's version when updating with another HA

dolodobendan, 2023-11-15:
### The problem

When I update an ESP32 using HA 1, then HA 2 doesn't know that an update has been installed.

My regular HA installation is on a Raspberry Pi 3B+ (HA 2). It has not enough memory to compile code for some ESP32s. That's why I have another, virtual HA installation (HA 1) that I use for updating the ESP32s.

After having updated my ESP32s, HA 2 still shows that updates are available, while in fact they are already installed.

### Which version of ESPHome has the issue?

2023.11

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.2

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
n/a
```

### Anything in the logs that might be useful for us?

```txt
n/a
```

### Additional information

n/a

jsuanet, 2023-11-15:
This is not a bug. The "update available" text is shown based on the information in a local json file which is generated as soon as you start the compile. So, on your HA 2 device this file is not updated. The easiest way to get rid of the update info: just click the update button on your HA 2 device, and stop the action as soon as the message "Info compiling app" shows up.   Or move all your compile activities to the HA 1 device and remove the ESPHome add-on from your HA 2 device. Now you have your YAML's (for the ESP32s) on two different places and for sure they will be out of sync somewhere in the future 😃

bdraco, 2023-11-15:
While its working as designed, we could likely make it better since we can compare the version with what we see in the mdns text record.  The dashboard still needs a lot of pre-work to be able to get to that point

dolodobendan, 2023-11-22:
> This is not a bug. The "update available" text is shown based on the information in a local json file which is generated as soon as you start the compile. So, on your HA 2 device this file is not [...]

😵‍💫

> Or move all your compile activities to the HA 1 device and remove the ESPHome add-on from your HA 2 device. Now you have your YAML's (for the ESP32s) on two different places and for sure they will be out of sync somewhere in the future 😃

Since I exclusively use HA 1 for updating the ESPs, I wouldn't be informed about new updates in HA 2. I could just check once a month, of course.

dolodobendan, 2023-11-22:
> While its working as designed, we could likely make it better since we can compare the version with what we see in the mdns text record. The dashboard still needs a lot of pre-work to be able to get to that point

That would be awesome, thank you.

dolodobendan, 2024-03-27:
> While its working as designed, we could likely make it better since we can compare the version with what we see in the mdns text record. The dashboard still needs a lot of pre-work to be able to get to that point

It seems you guys did just that. Thank you!

closed 2024-03-27 by dolodobendan