esphome/issues #5109: Error compiling code, which includes "font" component

AdmiralStipe, 2023-11-15:
### The problem

After udpating to ESPHome 2023.11.0, I get an error when trying to update devices, where I use a different fonts for displaying values on attached display. The issue is not present on version 2023.10.6, it only occured with the latest version. 

### Which version of ESPHome has the issue?

2023.11.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.2

### What platform are you using?

ESP8266

### Board

wemos D1 mini

### Component causing the issue

font

### Example YAML snippet

```yaml
esphome:
  name: wemos-max7219
  platform: ESP8266
  board: d1_mini
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
captive_portal:
logger:
api:
font:
  - file: "pixelmix.ttf"
    id: mojfont
    size: 8
    glyphs: [' ', '!', '?', '"', "'", "’", '*', '%', (, ), +, ',', '-', ., /, '&', '#', '0', '1', '2', '3', '4', '5', '6', '7', '8', '9', ':', A, B, C, Č, Ć, D, Đ, E, F, G, H, I, J, K, L, M, N, O, P, Q, R, S, Š, T, U, V, W, X, Y, Z, Ž, _, a, b, c, č, ć, d, đ, e, f, g, h, i, j, k, l, m, n, o, p, q, r, s, š, t, u, v, w, x, y, z, ž, °]
spi:
  clk_pin: D5
  mosi_pin: D7

text_sensor:
  - platform: homeassistant
    id: balkonspred
    entity_id: sensor.aqara_temperatura
  - platform: homeassistant
    id: balkonzad
    entity_id: sensor.aqara_temperatura_2
  - platform: homeassistant
    id: dnevna
    entity_id: sensor.temperatura_dnevna
  - platform: homeassistant
    id: napis1
    entity_id: var.napis1
  - platform: homeassistant
    id: napis2
    entity_id: var.napis2
  - platform: homeassistant
    id: napis3
    entity_id: var.napis3
  - platform: homeassistant
    id: napis4
    entity_id: var.napis4
  - platform: homeassistant
    id: casovnik
    entity_id: timer.display
  - platform: homeassistant
    id: casovnik1
    entity_id: timer.casovnik1
  - platform: homeassistant
    id: casovnik2
    entity_id: timer.casovnik2
  - platform: homeassistant
    id: casovnik3
    entity_id: timer.casovnik3
  - platform: homeassistant
    id: casovnik4
    entity_id: timer.casovnik4
display:
  - platform: max7219digit
    cs_pin: D8
    num_chips: 4
    intensity: 10
    scroll_speed: 20ms
    scroll_delay: 0s
    lambda: |-
          if ((id(casovnik).state == "active") and (id(casovnik1).state == "active") and (id(casovnik2).state == "active")) {
               std::string str1=id(balkonzad).state;
               std::string str2=id(balkonspred).state;
               std::string str3=id(dnevna).state;
               std::string izpis0="  "+id(balkonspred).state+"°  "+id(balkonzad).state+"°  "+id(dnevna).state+"°C  ";
               std::string izpis1="  "+id(napis1).state+"  ";
               std::string izpis2="  "+id(napis2).state+"  ";
               std::string izpis=izpis0+izpis1+izpis2;
               it.print(0, 0, id(mojfont), izpis.c_str());
                 }          
          else if ((id(casovnik).state == "active") and (id(casovnik1).state == "active")) {
               std::string str1=id(balkonzad).state;
               std::string str2=id(balkonspred).state;
               std::string str3=id(dnevna).state;
               std::string izpis0="  "+id(balkonspred).state+"°  "+id(balkonzad).state+"°  "+id(dnevna).state+"°C  ";
               std::string izpis1="  "+id(napis1).state+"  ";
               std::string izpis2="";
               std::string izpis=izpis0+izpis1+izpis2;
               it.print(0, 0, id(mojfont), izpis.c_str());
                 }          
          else if ((id(casovnik).state == "active") and  (id(casovnik2).state == "active")) {
               std::string str1=id(balkonzad).state;
               std::string str2=id(balkonspred).state;
               std::string str3=id(dnevna).state;
               std::string izpis0="  "+id(balkonspred).state+"°  "+id(balkonzad).state+"°  "+id(dnevna).state+"°C  ";
               std::string izpis1="";
               std::string izpis2="  "+id(napis2).state+"  ";
               std::string izpis=izpis0+izpis1+izpis2;
               it.print(0, 0, id(mojfont), izpis.c_str());
                 }          
          else if ((id(casovnik1).state == "active") and (id(casovnik2).state == "active")) {
               std::string izpis0="";
               std::string izpis1="  "+id(napis1).state+"  ";
               std::string izpis2="  "+id(napis2).state+"  ";
               std::string izpis=izpis0+izpis1+izpis2;
               it.print(0, 0, id(mojfont), izpis.c_str());
                 }                    
          else if (id(casovnik).state == "active") {
               std::string str1=id(balkonzad).state;
               std::string str2=id(balkonspred).state;
               std::string str3=id(dnevna).state;
               std::string izpis0="  "+id(balkonspred).state+"°  "+id(balkonzad).state+"°  "+id(dnevna).state+"°C  ";
               it.print(0, 0, id(mojfont), izpis0.c_str());
                 }
          else if (id(casovnik1).state == "active") {
               std::string izpis1="  "+id(napis1).state+"  ";
               it.print(0, 0, id(mojfont), izpis1.c_str());
                 }
          else if (id(casovnik2).state == "active") {
               std::string izpis2="  "+id(napis2).state+"  ";
               it.print(0, 0, id(mojfont), izpis2.c_str());
                 }
          else {
                 }
  - platform: max7219digit
    cs_pin: D1
    num_chips: 4
    intensity: 11
    scroll_speed: 20ms
    scroll_delay: 0s
    rotate_chip: 180
    reverse_enable: true
    lambda: |-
          if ((id(casovnik3).state == "active") and (id(casovnik4).state == "active")) {
               std::string izpis5="  "+id(napis4).state+"  "+id(napis3).state+"  ";
               it.print(0, 0, id(mojfont), izpis5.c_str());
                 }
          else if (id(casovnik3).state == "active") {
               std::string izpis3="  "+id(napis3).state+"  ";
               it.print(0, 0, id(mojfont), izpis3.c_str());
                 }
          else if (id(casovnik4).state == "active") {
               std::string izpis4="  "+id(napis4).state+"  ";
               it.print(0, 0, id(mojfont), izpis4.c_str());
                 }
          else {
                 }

button:
  - platform: restart
    name: "Wemos MAX7219 Restart"
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.11.0
INFO Reading configuration /config/esphome/wemos-max7219.yaml...
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1040, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1027, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 406, in command_compile
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 192, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 204, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 674, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 184, in wrapped
    await coro(conf)
  File "/esphome/esphome/components/font/__init__.py", line 356, in to_code
    font = load_ttf_font(path, config[CONF_SIZE])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/components/font/__init__.py", line 340, in load_ttf_font
    from PIL import ImageFont
  File "/usr/local/lib/python3.11/dist-packages/PIL/ImageFont.py", line 35, in <module>
    from . import Image
  File "/usr/local/lib/python3.11/dist-packages/PIL/Image.py", line 82, in <module>
    from . import _imaging as core
ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory
```

fordovo, 2023-11-16:
the same issue here.

poundy, 2023-11-17:
My experience was slightly different, I had no issues witn 2023.11.0 and 2023.11.1, and only hit this after 2023.11.2. I literally had a display device I hadn't powered on in a few months that I switched on this morning, it worked straight on power up, but then I updated it with 2023.11.1 as that's what I was running at the time. After that had finished, all was good, noticed 2023.11.2 so applied it and then went to update the same device and couldn't due to above issue; attempted to update some of my other devices, all ones with displays failed (I have only a small number of devices without displays)

greblosdier, 2023-11-17:
I can confirm that I'm encountering the issue with 2023.11.1 (and previously 2023.11.0) using the ESPHome add-in on a fully up-to-date HASS OS stack (Raspberry Pi, 32-bit.)  It was compiling and updating successfully on 2023.10.6 and prior.  My device YAML file uses gfonts.  I'm building for a Wemos D1 mini/ESP8266.

Initially it was set up like this:
```
font:
  
  - file: "gfonts://Roboto Mono"
    id: mono_lg
    size: 22
```

I changed it to this while troubleshooting, in case that syntax had become deprecated:
```
font:
  
  - file:
      type: gfonts
      family: "Roboto Mono"
    id: mono_lg
    size: 22
```

Both result in the same compilation errors, as seen below.  If I remove the fonts and display section from the YAML, it will compile and install/update correctly, but obviously without the display functionality on the device.

```
Install playhouse-thermostat.yaml
INFO ESPHome 2023.11.1
INFO Reading configuration /config/esphome/playhouse-thermostat.yaml...
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1040, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1027, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 457, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 192, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 204, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 674, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 184, in wrapped
    await coro(conf)
  File "/esphome/esphome/components/font/__init__.py", line 359, in to_code
    font = load_ttf_font(path, config[CONF_SIZE])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/components/font/__init__.py", line 340, in load_ttf_font
    from PIL import ImageFont
  File "/usr/local/lib/python3.11/dist-packages/PIL/ImageFont.py", line 35, in <module>
    from . import Image
  File "/usr/local/lib/python3.11/dist-packages/PIL/Image.py", line 82, in <module>
    from . import _imaging as core
ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory
```

glaaze, 2023-11-18:
Looks like something is missing in the container. I'm using `apt-get install libopenjp2-7 libtiff6` as a workaround until it's fixed

AdmiralStipe, 2023-11-18:
> Looks like something is missing in the container. I'm using `apt-get install libopenjp2-7 libtiff6` as a workaround until it's fixed

I tried it too, but without success - I got an error "Unable to  locate package"... could you describe all the needed steps in details?

glaaze, 2023-11-18:
Maybe you just forgot to `apt update` before install. BTW, get into the container with a `docker exec -it addon_xxx_esphome /bin/bash` then `apt update && apt install libopenjp2-7 libtiff6`

And 2023.11.2 is still broken... :(

Svenpscript, 2023-11-18:
Same error. I run the home assistant os. Apparently it is not possible to use bash or apt-get. I'll just have to wait for an update.

cvandesande, 2023-11-18:
It is possible on home assistant os, there are some extra steps.

If you install the "advanced ssh & web terminal" and run it with "protection mode" disabled, you can access all the docker containers on the host.  

Then you can access the esphome container
```
docker exec -it addon_5c53de3b_esphome /bin/bash
```
And then you can run
```
apt-get update
apt-get install install libopenjp2-7 libtiff6
```

danieltidona, 2023-11-18:
Same issue from 2023.11.0, i use Roboto-Medium.ttf. I have tried to update the library with the "pip install libopenjp2" command but it gives me an error "No matching distribution found for ...". What I do? Should I try reinstalling everything or wait for an new esphome update? What do you recommend? I am a newbie, this is the first time I have integrated esphome into my home assistant. Thanks!

greblosdier, 2023-11-19:
> ```
> apt-get install install libopenjp2-7 libtiff6
> ```

instead:
`apt-get install libopenjp2-7 libtiff6`

It was still broken for me in ESPHome 2023.11.2.  I followed cvandesande's instructions and omitted the extra "install".  It installed those two packages along with four dependencies.  My YAML with fonts now compiles and installs successfully again.

To avoid a potential conflict, I uninstalled the "Terminal & SSH" add-on before installing the "Advanced SSH & Web Terminal" from the community add-on repository, disabling Protection Mode on the add-on's Info tab.

Home Assistant OS (Raspberry Pi, 32-bit)
Core 2023.11.2
Supervisor 2023.11.3
Operating System 11.1
ESPHome 2023.11.2

cvandesande, 2023-11-19:
Nice one @greblosdier ! I edited my original comment

AdmiralStipe, 2023-11-19:
> Maybe you just forgot to `apt update` before install. 

Yep, that was it, got it working now, thx :).

Svenpscript, 2023-11-19:
This worked perfectly!
I first reverted back to 2023.10.6 to make the changes I wanted to do. With this workaround I was able to update to the latest version without any problems.
Thanx again!

tastyllama, 2023-11-19:
This worked perfectly for my inkplate 10.  Thank you @cvandesande and @greblosdier !

gordoasan, 2023-11-20:
> It is possible on home assistant os, there are some extra steps.
> 
> If you install the "advanced ssh & web terminal" addon and run it with "protection mode" disabled, you can access all the docker containers on the host.
> 
> Then you can access the esphome container
> 
> ```
> docker exec -it addon_5c53de3b_esphome /bin/bash
> ```
> 
> And then you can run
> 
> ```
> apt-get update
> apt-get install libopenjp2-7 libtiff6
> ```
> 
> *Edit removed duplicate word "install"

This worked great for me Thank you so much

AdmiralStipe, 2023-11-20:
Is any of you guys capable of creating a PR to fix this? Unfortunately I have no idea, what has to be done, but I believe a PR would make solving this issue much faster than just waiting for creators of the code to attend to this.... I'm afraid it will quicly become annoying to apply this workaround everytime ESPHome updates...

AdmiralStipe, 2023-11-20:
I just found out, that the workaround works, but for me only for a short time.
After a few hours the ESPHome docker image seems to "reset" itself and installed libopen is missing again - all ESPHhome devices show "Update available" and the same error is present again, if I try to update devices.

Is any of you guys capable of creating a PR to fix this? Unfortunately I have no idea, what has to be done, but I believe a PR would make solving this issue much faster than just waiting for creators of the code to attend to this.... I'm afraid it will quickly become annoying to apply this workaround so often...

glaaze, 2023-11-20:
Well, the thing is that if you look at  PR [5074](https://github.com/esphome/esphome/pull/5074) the guy that submitted the PR said he was going to fix the dockerfile by adding the dependencies for armv7. I'm not sure if he was going to re-add python-pil3 as a dependency or just libopenjp2-7 and libtiff6 and let pip3 manage pillow. Besides, I'm not even sure why pip with a "--break-system-packages" switch wouldn't work even if the pil3 deb was installed 

On top of that, if that was already broken last July, I'm wondering why everything worked here until a couple of days ago....
I'll have a better look tomorrow btw

danieltidona, 2023-11-20:
I am very sorry for my inexperience, but I have not been able to make this workaround work. My HA does not run on docker and I have not been able to apply this solution even temporarily. I configured my terminal without the protection mode and I still cannot do an apt-get, I have tried it with pip install and it gives me an error that it cannot find this reference in the repository. I will continue waiting for an update perhaps on ESPHome

jesserockz, 2023-11-21:
Fixed with https://github.com/esphome/esphome/pull/5808 and will be in ESPHome 2023.11.3

closed 2023-11-21 by jesserockz

Atarinside, 2023-12-21:
Hi,
i had the exact same issue with HA OS :
Core 2023.12.3
Supervisor 2023.12.0
Operating System 11.2

and after upgrading from ESPhome 2023.11.6 to ESPhome 2023.12.0

The solution above solved the issue (docker then apt upgrade..)

Atarinside, 2023-12-21:
Hi,
same problem again (with ESPhome 2023.12.1).
FYI, the problem occurs on ESP32, (i dont' have any ESP8266).

cvandesande, 2023-12-21:
The version of libtiff6 that was being installed in the esphome Docker container was replaced in Debian with a security update. That's why it's missing again. I submitted  https://github.com/esphome/esphome/pull/5985 with the new version