esphome/issues #5110: ESPhome not starting with update 2023.11.0

alflexRH, 2023-11-15:
### The problem

[13:49:27] INFO: Starting ESPHome dashboard...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 10, in <module>
    from importlib.metadata import distribution
ImportError: cannot import name 'distribution' from 'importlib.metadata' (/usr/lib/python3.11/importlib/metadata/__init__.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 13, in <module>
    from importlib_metadata import distribution
ModuleNotFoundError: No module named 'importlib_metadata'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 15, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 27, in <module>
    import functools
  File "/usr/lib/python3.11/functools.py", line 18, in <module>
  File "/usr/lib/python3.11/collections/__init__.py", line 35, in <module>
    from keyword import iskeyword as _iskeyword
ImportError: cannot import name 'iskeyword' from 'keyword' (/usr/lib/python3.11/keyword.py)
[13:49:27] INFO: Service ESPHome dashboard exited with code 1 (by signal 0)

### Which version of ESPHome has the issue?

23023.11.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

alflexRH, 2023-11-20:
I have deleted and reinstalled the addon, and updated to 2023.11.2 and the problem is still there.

oyvindwe, 2024-01-10:
I have a similar problem. Dashboard works on 2023.10.6, but fails on 2023.11.0 and later.

Command: `docker run --rm --name esphome -v /home/omw/esphome/:/config/ -v /etc/localtime:/etc/localtime:ro --net=host ghcr.io/esphome/esphome:2023.11.0 dashboard --username omw --password '<redacted>' --address 127.0.0.1 /config`

Log:
```
2024-01-10 18:50:54,718 INFO Starting dashboard web server on http://127.0.0.1:6052 and configuration dir /config...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1040, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1003, in run_esphome
    return PRE_CONFIG_ACTIONS[args.command](args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 517, in command_dashboard
    return dashboard.start_web_server(args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/dashboard/dashboard.py", line 1509, in start_web_server
    status_thread.start()
  File "/usr/lib/python3.11/threading.py", line 957, in start
    _start_new_thread(self._bootstrap, ())
RuntimeError: can't start new thread
```

ssieb, 2024-01-10:
@oyvindwe that is https://github.com/esphome/issues/issues/5142.