esphome/issues #5111: Compiling error introduced in ESPHome 2023.11.0 - "sh: 1: xtensa-esp32-elf-g++: not found"

ricarva, 2023-11-15:
### The problem

Compiling process fails with "sh: 1: xtensa-esp32-elf-g++: not found", Error 127.

Seems to have been introduced with ESPHome 2023.11.0; previous version compiled and uploaded fine.

### Which version of ESPHome has the issue?

2023.11.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.2

### What platform are you using?

ESP32

### Board

ESP32-DevKitC-VE

### Anything in the logs that might be useful for us?

```txt
Updating /config/esphome/senshubproto.yaml
------------------------------------------------------------

INFO ESPHome 2023.11.0
INFO Reading configuration /config/esphome/senshubproto.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing senshubproto (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- SPI @ 2.0.0
|-- BSEC Software Library @ 1.6.1480
Compiling .pioenvs/senshubproto/src/esphome/components/adc/adc_sensor.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
*** [.pioenvs/senshubproto/src/esphome/components/adc/adc_sensor.cpp.o] Error 127
========================== [FAILED] Took 8.23 seconds ==========================
======== [ERROR] /config/esphome/senshubproto.yaml ========

======================== [SUMMARY] ========================
  - /config/esphome/senshubproto.yaml: FAILED
```

ssieb, 2023-11-15:
Duplicate of https://github.com/esphome/issues/issues/3092

ricarva, 2023-11-15:
> Duplicate of #3092

Is it a duplicate, really?

More users of the ESPHome HA addon are reporting this issue since the latest 2023.11.0 update (was previously fine in 2023.10.6, and a user who has rolled back reports everything OK again).

Also, at least in my case, I'm running 32 bit everything (OS, kernel).

ssieb, 2023-11-15:
Can you come to discord to do some checking?  Regardless, it's still a platformio issue anyway, and nothing that esphome can do anything about.

ricarva, 2023-11-16:
> Can you come to discord to do some checking? Regardless, it's still a platformio issue anyway, and nothing that esphome can do anything about.

Sure, I'll come by Discord. What timezone are you on? Thanks for the help :-)

ssieb, 2023-11-17:
This is related to a workaround in the 32-bit arm container needed to let the toolchain run.  The container was updated, so the workaround needs to be updated as well.

closed 2023-11-17 by jesserockz

leppa, 2023-11-17:
Thank you, @ssieb!

For those who want to apply the fix without waiting for the next release, you need to do the following (ESPHome container must be running):
```bash
docker exec -it addon_5c53de3b_esphome /bin/bash # Adjust addon_5c53de3b_esphome to the name of your ESPHome container
ln -s /lib/arm-linux-gnueabihf/ld-linux-armhf.so.3 /lib/ld-linux.so.3
```

vespadj, 2023-11-17:
On mine RPi, 
resolved thank to
[this chat on Discord](https://discord.com/channels/429907082951524364/1174699586535821352)
docker exec -it esphome /bin/bash

uname -m # armv7l
apt update
apt install file
# ...
cd .esphome #from config
file ./platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++
file -L /lib/ld-linux-armhf.so.3
ln -s ld-linux-armhf.so.3 /lib/ld-linux.so.3