esphome/issues #5114: Can not "<<: !include ..." and then extend "display" and "touchscreen"

NdK73, 2023-11-15:
### The problem

I'm trying to separate board definitions from device logic by using
<<: !include boards/boardname.yaml
from device's yaml.

The file boards/esp32-2432s028.yaml is reported below.

But when I try to use:
```
display:
- id: !extend disp
  pages:
    - ....
```
or
```
touchscreen:
- id: !extend touch
  calibration_x_min: 3840
  calibration_x_max: 290
  calibration_y_min: 220
  calibration_y_max: 3730
```
I get errors stating "Source for extension of ID disp was not found." (and the same for "touch").

Am I doing something wrong or I just hit an unsupported case?

Tks.

### Which version of ESPHome has the issue?

2023.11.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.2

### What platform are you using?

ESP32

### Board

ESP32-2432S028

### Component causing the issue

touchscreen and display

### Example YAML snippet

```yaml
The file boards/esp32-2432s028.yaml contains

esp32:
  board: esp32dev
  framework:
    type: arduino # REQUIRED for integrated webserver

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  #ap:
  #  ssid: "fallback"
  #  password: "*********"
  #
#captive_portal:

#esp32_improv:
#  authorizer: binary_sensor_id

spi:
  # Order of SPI busses is important: if spi_touch is defined first, touch ctrl is not recognized
  - id: spi_disp
    clk_pin: GPIO14
    mosi_pin: GPIO13
    miso_pin: GPIO12
  - id: spi_touch
    clk_pin: GPIO25
    mosi_pin: GPIO32
    miso_pin: GPIO39

display:
-  id: disp
  platform: ili9xxx
  model: TFT 2.4
  spi_id: spi_disp
  rotation: 270
  cs_pin: GPIO15
  dc_pin: GPIO2
  #led_pin: GPIO21
  #reset_pin: 2

touchscreen:
-  id: touch
  platform: xpt2046
  cs_pin: GPIO33
  #display: fake_disp
  spi_id: spi_touch
  interrupt_pin: GPIO36
  update_interval: 50ms
  report_interval: 1s
  threshold: 400
  #swap_x_y: true
  on_touch:
  - lambda: |-
      ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%0d",
        id(touch).x,
        id(touch).y,
        id(touch).x_raw,
        id(touch).y_raw
        );
```
```

ssieb, 2023-11-15:
I don't seen any includes there, but to use `!extend`, you need to use packages, not a plain include.

NdK73, 2023-11-16:
In device's yaml code I have a
```
<<: !include boards/esp32-2432s028.yaml
```

If !extend can only be used with packages, then the docs at https://esphome.io/guides/configuration-types.html#extend are at least misleading. But it seems to imply the reverse, since examples for packages do not use !extend ...

ssieb, 2023-11-16:
The extend section was added recently and the example might not be quite correct like that.  Try using packages instead and see what happens.

NdK73, 2023-11-17:
Thanks, seems that with packages it's now "working" (does not complain anymore about sources of extensions not found).
But it's now giving errors with nearly every use of id(key):
```
In file included from src/esphome/components/sensor/sensor.h:3,
                 from src/esphome/components/adc/adc_sensor.h:6,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
/config/esphome/boards/esp32-2432s028.yaml: In lambda function:
/config/esphome/boards/esp32-2432s028.yaml:55:14: error: base operand of '->' has non-pointer type 'esphome::touchscreen::TouchPoint'
         id(touch).x,
              ^~
```
Quite puzzling, since I'm using '.' and not '->' ...
The lambdas were working OK in the monolithic configuration. Do packages require code changes?

ssieb, 2023-11-17:
It has nothing to do with packages or includes.  In that lambda, there's a variable called `touch` which is not an `id` and is different than the component with the id `touch`.

NdK73, 2023-11-17:
🤔 ... 🤦‍♂️ 
You're right: 'touch' is the name of the variable passed to lambda...
Tks.

closed 2023-11-17 by NdK73