esphome/issues #5115: LD2420 component on ESP8266 triggers a boot loop

afarago, 2023-11-15:
### The problem

Enabling LD2420 on ESP8266 triggers a boot loop

### Which version of ESPHome has the issue?

2023.11.0

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

d1_mini

### Component causing the issue

ld2420

### Example YAML snippet

```yaml
substitutions:
  systemName: "garazs-temp"
  device_name: "${systemName}"

esphome:
  name: $systemName

esp8266:
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  password: !secret ota_password
  safe_mode: yes

logger:
  level: info

ld2420:

uart:
  id: ld2420_uart
  tx_pin: D1
  rx_pin: D2
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

text_sensor:
  - platform: ld2420
    fw_version:
      name: LD2420 Firmware

sensor:
  - platform: ld2420
    moving_distance:
      name : Moving Distance

binary_sensor:
  - platform: ld2420
[... 106 lines cut ...]
[23:10:41]3fffffb0:  3fffdad0 00000000 3fff0bd8 4022b368  
WARNING Decoded 0x4022b368: loop_wrapper() at core_esp8266_main.cpp
[23:10:41]3fffffc0:  feefeffe feefeffe 3ffe8640 401006d5
WARNING Decoded 0x401006d5: cont_wrapper
[23:10:41]<<<stack<<<
[23:10:41]
[23:10:41]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[23:10:41]
[23:10:41] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[23:10:41]
[23:10:41]load 0x4010f000, len 3460, room 16
[23:10:41]tail 4
[23:10:41]chksum 0xcc
[23:10:41]load 0x3fff20b8, len 40, room 4
[23:10:41]tail 4
[23:10:41]chksum 0xc9
[23:10:41]csum 0xc9
[23:10:41]v00077550
[23:10:41]~ld
[23:10:41][I][logger:326]: Log initialized
[23:10:41][I][app:029]: Running through setup()...
[23:10:41]
[23:10:41]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[23:10:41]
```

ssieb, 2023-11-15:
I don't know why it's crashing, unless it has run out of memory, but the software serial can't handle that baud rate anyway.

afarago, 2023-11-16:
> I don't know why it's crashing, unless it has run out of memory, but the software serial can't handle that baud rate anyway.

Good point, yet I have also tried with RX, TX pins for hardware serial first, this is an nth iteration.

I assume that LD2420 is working with ESP8266 and would try to find that approach. Otherwise it should be ESP32 only.

ssieb, 2023-11-16:
Does it still crash if you use the hardware uart?  You need to disable serial logging if you do that which makes it more difficult to debug.

vinulac, 2023-11-18:
Hi, I have a similar problem. The esp8266 spits out as soon as I add the operating mode.

```
select:
  - platform: ld2420
    operating_mode:
      name: Operating Mode
```

realdfender, 2023-11-19:
New poster, sorry for bad formatting.

After some weeks i think i have finally some solution to this.

Please bear in mind the change in TX port and baud rate on different firmware.
ref esphome documentation. Mine is the later fw 1.5.8.

The sensor is really unhappy on software uart, need HW uart uneless there is so much noice.
I am running 3pcs, 2 x wemos d1 mini, 1 x bare module.

First install software using the HW uart GPIO1 and GPIO3 in UART Bus config for the sensor (And i also set log component to UART1, unsure if any effect).

Then connect the sensor.

ESP cannot boot if connected to computer after so use usb power supply.

Subsequent updates over wifi.

example my config:
`esphome:
  name: presence-2
  friendly_name: Presence 2

esp8266:
  board: nodemcuv2

# Enable logging
logger:
  hardware_uart: UART1

# Enable Home Assistant API
api:
  encryption:
    key: "hidden"

ota:
  password: "hidden"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Presence-2 Fallback Hotspot"
    password: "hidden"

captive_portal:
    
web_server:
  port: 80

uart:
  id: ld2420_uart
  tx_pin: GPIO14
  rx_pin: GPIO12
  baud_rate: 115200
  parity: NONE
  stop_bits: 1

ld2420:

text_sensor:
  - platform: ld2420
    fw_version:
      name: LD2420 Firmware

sensor:
  - platform: ld2420
    moving_distance:
      name : Moving Distance

binary_sensor:
  - platform: ld2420
    has_target:
      name: Presence

select:
  - platform: ld2420
    operating_mode:
      name: Operating Mode

number:
  - platform: ld2420
    presence_timeout:
      name: Detection Presence Timeout
    min_gate_distance:
      name: Detection Gate Minimum
    max_gate_distance:
      name: Detection Gate Maximum

    gate_select:
      name: Select Gate to Set
    still_threshold:
      name: Set Still Threshold Value
    move_threshold:
      name: Set Move Threshold Value

    gate_move_sensitivity:
      name: Move Calibration Sensitivity Factor
    gate_still_sensitivity:
      name: Still Calibration Sensitivity Factor

    gate_0:
      move_threshold:
        name: Gate 0 Move Threshold
      still_threshold:
        name: Gate 0 Still Threshold
    gate_1:
      move_threshold:
        name: Gate 1 Move Threshold
      still_threshold:
        name: Gate 1 Still Threshold

button:
  - platform: ld2420
    apply_config:
      name: Apply Config
    factory_reset:
      name: Factory Reset
    restart_module:
      name: Restart Module
    revert_config:
      name: Undo Edits`

ride8055, 2023-11-26:
I appear to have the same problem, I have tried with ESP8266 & ESP32 using version 2023.11.4, once the LD2420 code is added and flashed, boot loop.

simonyel, 2023-11-29:
Same here on ESP32. Tried:
- three different ESP32dev boards
- two known working LD2420s (tested on PC over serial tool)
- both hardware UARTs 
- barebone build with just the radar doesn't work either

What does work, however, is code from https://github.com/esphome/esphome/pull/4847. I believe something didn't get merged correctly. @descipher

With published configuration and the kernel crashes as soon as ld2420 module starts with error:
`Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.`

descipher, 2023-11-29:
The only difference with PR#4847 and the current esphome code is an PR#5727 (Missed Ifdefs)
An important thing to note when using some boards is the power requirement of using the LD2420 and supplying the 3.3 volts. Many boards like the D1 Mini cannot support both the LD2420 and the ESP at the same time. It will brown out due to the WiFi power surge and or load variations of the WiFi radio. I just complied the current esphome code for an ESP32 and have no issues running it on my https://www.gelidus.ca/product/gelidus-research-ac-powered-mmwave-presence-sensor/ targets. This board has ample current to handle both the LD2420 and the ESP32. I could not even flash the same board using low end USB to serial adapters using the same 3.3v LDO's used on the D1 Mini.

descipher, 2023-11-29:
@simonyel Odd that you have the esp32dev working on PR#4847 and not the current repo. Please post your YAML and I will check on a compile run with it.

simonyel, 2023-11-29:
```
substitutions:
  device_name: radar-test
  entity_prefix: ld2420
  friendly_name: radar-test
  device_type: ESP32
  device_comment: LD2420-test

esphome:
  name:  ${device_name}
  comment: "${device_type}: ${device_comment}"

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "**redacted**"

web_server:
  port: 80

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: .**redacted**.local

ota:
  password: !secret ota_password

uart:
  id: ld2420_uart
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 115200
  parity: NONE
  stop_bits: 1  

ld2420:

text_sensor:
  - platform: version
    name: ${entity_prefix} Version
    id: esphome_version
    hide_timestamp: True
  - platform: wifi_info
    ip_address:
      id: ip_address
      name: ${entity_prefix} IP Address
    mac_address:
      name: ${entity_prefix} MAC
      id: mac_address
  - platform: ld2420
    fw_version:
      name: ${entity_prefix} LD2420 Firmware

sensor:
  - platform: ld2420
    moving_distance:
      name : ${entity_prefix} Moving Distance  
  - platform: uptime
    name: ${entity_prefix} Node Uptime
    update_interval: 15min
  - platform: wifi_signal
    name: ${entity_prefix} Wifi Signal
    update_interval: 1min
    filters:
      - filter_out: nan
      # exclude spurious readings
      - lambda: |- 
          if (x < -20) return x;
          else return {};
      - or:
        - throttle: 30min
        - delta: 5

binary_sensor:
  - platform: status
    name: ${entity_prefix} Node Status
  - platform: ld2420
    has_target:
      name: ${entity_prefix} Presence
      
button:
  - platform: ld2420
    apply_config:
      name: ${entity_prefix} Apply Config
    factory_reset:
      name: ${entity_prefix} Factory Reset
    restart_module:
      name: ${entity_prefix} Restart Module
    revert_config:
      name: ${entity_prefix} Undo Edits        
  - platform: restart
    name: ${entity_prefix} Node Restart
    entity_category: "diagnostic"

select:
  - platform: ld2420
    operating_mode:
      name: ${entity_prefix} Operating Mode

number:
  - platform: ld2420
    presence_timeout:
      name: ${entity_prefix} Presence Timeout
    min_gate_distance:
      name: ${entity_prefix} Gate Minimum
    max_gate_distance:
      name: ${entity_prefix} Gate Maximum
    # See "Number" section below for detail
    gate_select:
      name: ${entity_prefix} Select Gate to Set
    still_threshold:
      name: ${entity_prefix} Set Still Threshold Value
    move_threshold:
      name: ${entity_prefix} Set Move Threshold Value

```
This is what the logs look like through espweb over serial:
[image]

descipher, 2023-11-29:
[image]

Compiles and boots up Ok here.
Based on the log it happens on the first command send of config enable.
load the exception into here and lets see what call it dies on. 
https://maximeborges.github.io/esp-stacktrace-decoder/

descipher, 2023-11-30:
While this should not matter I thought I would note it.
I have ESP32 Tx on GPIO16 and ESP32 Rx on GPIO17 Since it is a hardware UART port it may have some strange effect with IO pin mappings.

simonyel, 2023-11-30:
I'm still trying to figure it out, but during today's testing I was getting mixed results. At some point I had prod version working and PR not. I think it's something to do with way I'm naming the entities.

Sent from mobile. Please pardon any typos

descipher, 2023-11-30:
One thing I experienced during the development of this component was the poor quality of certain  esp32dev modules coming from aliexpress. I have three of them. Only one is still functioning.

simonyel, 2023-11-30:
> ![image](https://private-user-images.githubusercontent.com/10734527/286767489-2893045e-0589-44db-babb-60c2b3e53f48.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

How do I get the .elf binary? And which the portion of the log is the "stack trace" 

- Thanks for helping out!

ride8055, 2023-11-30:
Hey @descipher If you boot up the esp32 without the LD2420 connected, does the esp still boot and run properly? I am having the same problems but I also don't know if my LD2420 is good or defective and curious if that could be the problem?

Thanks

simonyel, 2023-11-30:
> How do I get the .elf binary? And which the portion of the log is the "stack trace"
>
> * Thanks for helping out!

Does this tell you anything? I finally got to the .elf

```
Output:
0x40176b78: esphome::uart::ESP32ArduinoUARTComponent::write_array(unsigned char const*, unsigned int) at /config/.esphome/build/presence-1/src/esphome/components/uart/uart_component_esp32_arduino.cpp:138
0x400db5b1: esphome::uart::UARTComponent::write_byte(unsigned char) at /config/.esphome/build/presence-1/src/esphome/components/uart/uart_component.h:35
0x400db5b1: esphome::uart::UARTDevice::write_byte(unsigned char) at /config/.esphome/build/presence-1/src/esphome/components/uart/uart.h:19
0x400db5b1: esphome::ld2420::LD2420Component::send_cmd_from_array(esphome::ld2420::LD2420Component::CmdFrameT) at /config/.esphome/build/presence-1/src/esphome/components/ld2420/ld2420.cpp:527
0x400db6ce: esphome::ld2420::LD2420Component::set_config_mode(bool) at /config/.esphome/build/presence-1/src/esphome/components/ld2420/ld2420.cpp:568
0x400dbe70: esphome::ld2420::LD2420Component::setup() at /config/.esphome/build/presence-1/src/esphome/components/ld2420/ld2420.cpp:115
0x40176f65: esphome::Component::call_setup() at /config/.esphome/build/presence-1/src/esphome/core/component.cpp:77
0x40177015: esphome::Component::call() at /config/.esphome/build/presence-1/src/esphome/core/component.cpp:98
0x400e6a10: esphome::Application::setup() at /config/.esphome/build/presence-1/src/esphome/core/application.cpp:38
0x400e9bf6: setup() at /config/_config/core.yaml:479
0x400f4ff6: loopTask(void*) at /config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```

descipher, 2023-11-30:
@simonyel Yes, it confirms that it died on the uart tx call sending the first command to the ld2420. Have you done a "clean build files". Sometimes an old lib results in these issues and that would clean up old lib downloads.

descipher, 2023-11-30:
> Hey @descipher If you boot up the esp32 without the LD2420 connected, does the esp still boot and run properly? I am having the same problems but I also don't know if my LD2420 is good or defective and curious if that could be the problem?
> 
> Thanks

It will boot and report a failure to communicate with the LD2420 in the logs.

ride8055, 2023-11-30:
For what it's worth, I connected my LD2420 to a usb to serial adapter and using OT1 as Tx at 115200 I get a stream of

`Range 106
ON
Range 106
ON
Range 106
ON
Range 106
ON
Range 106
ON
Range 105`

So it seems my module is doing something.
Do you know a way or command to check its version?

descipher, 2023-11-30:
Ok, The issue is now identified, with the additional feature of numbers etc. we had to check what the LD2420 module version is before configuring the components. If the LD2420 is not connected correctly, missing or defective we are comparing an un-initialized version string and that exceptions. With the version string initialized it will boot up without a functional LD2420. I will do a PR for this.

ride8055, 2023-11-30:
@descipher Those are the pins I've been using ( my uart is the same as the example above that worked for you ).
I'm also using a separate 3.3v supply for the LD2420 to rule out a brown out condition (with everything grounded).
I will double check all my wiring and do another sanity check.
thanks

afarago, 2023-11-30:
If it might help my device runs on firmware version 1.5.4.
Just checked it with the creepy exe tool.

ride8055, 2023-12-01:
@descipher It works!!
For whatever reason. I had to change the UART from
'
uart:
  id: ld2420_uart
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 115200
'
To
'
uart:
  id: ld2420_uart
  tx_pin: GPIO21
  rx_pin: GPIO19
  baud_rate: 115200
'
Weird but it now works.

yousaf465, 2023-12-01:
> While this should not matter I thought I would note it. I have ESP32 Tx on GPIO16 and ESP32 Rx on GPIO17 Since it is a hardware UART port it may have some strange effect with IO pin mappings.

esp32 RX GPIO16 and TX GPIO17 boot looping 

```yaml

uart:
  tx_pin: 17
  rx_pin: 16
  baud_rate: 256000
# Status Led
light:
  - platform: status_led
    name: "Switch state"
    pin: GPIO2
    id: ld2410statusled  
# Example configuration entry
ld2420:

text_sensor:
  - platform: ld2420
    fw_version:
      name: LD2420 Firmware

sensor:
  - platform: ld2420
    moving_distance:
      name : Moving Distance

binary_sensor:
  - platform: ld2420
    has_target:
      name: Presence

select:
  - platform: ld2420
    operating_mode:
      name: Operating Mode

number:
  - platform: ld2420
    presence_timeout:
      name: Detection Presence Timeout
    min_gate_distance:
      name: Detection Gate Minimum
    max_gate_distance:
      name: Detection Gate Maximum
    # See "Number" section below for detail
    gate_select:
      name: Select Gate to Set
    still_threshold:
      name: Set Still Threshold Value
    move_threshold:
      name: Set Move Threshold Value

button:
  - platform: ld2420
    apply_config:
      name: Apply Config
    factory_reset:
      name: Factory Reset
    restart_module:
      name: Restart Module
    revert_config:
      name: Undo Edits

```

descipher, 2023-12-01:
@yousaf465 That was ruled, you can switch them around without issue provided they match the ld2420 actuals.
add the following to test the fix for the boot loop:

~~~
external_components:
  - source: github://pr#5865
    components: [ ld2420 ]
    refresh: 0s
~~~

descipher, 2023-12-01:
@yousaf465 Do you have a pre v1.5.3 firmware?

yousaf465, 2023-12-01:
> uart:
>   id: ld2420_uart
>   tx_pin: GPIO21
>   rx_pin: GPIO19
>   baud_rate: 115200

how do I check the firmware?

yousaf465, 2023-12-01:
> You would have to use the HLK tool. But it more likely the baud is 115200 for them unless you purchased it very early.
> 
> ```
> status_led:
>   pin: GPIO2
> ```
> 
> No light on the LD2420 ..

Purchased on 11.11 sale,  status led is for the esp32 board. How to connect to HLK tool? for 2410B and 2410C I use the android app, how to connect this one to the HLK tool?

descipher, 2023-12-01:
November 11/23 is well after the v1.5.2 firmware.
Use the following, and you really only need status_led: by its self to see the LD2420 component state. 
~~~
external_components:
  - source: github://pr#5865
    components: [ ld2420 ]
    refresh: 0s

uart:
  id: ld2420_uart
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 115200
  parity: NONE
  stop_bits: 1

# The LD2420 has 16 sense gates 0-15 and each gate detects 0.7 meters 15th gate = 9m
ld2420:

text_sensor:
  - platform: ld2420
    fw_version:
      name: LD2420 Firmware

sensor:
  - platform: ld2420
    moving_distance:
      name : Moving Distance

binary_sensor:
  - platform: ld2420
    has_target:
      name: Presence

switch:
  - platform: restart
    name: Restart ESPHome Node

select:
  - platform: ld2420
    operating_mode:
      name: (s) Operating Mode

number:
  - platform: ld2420
    presence_timeout:
      name: Detection Presence Timeout
    min_gate_distance:
      name: Detection Gate Minimum
    max_gate_distance:
      name: Detection Gate Maximum
    gate_select:
      name: Select Gate to Set
    still_threshold:
      name: Set Still Threshold Value
    move_threshold:
      name: Set Move Threshold Value
    gate_move_sensitivity:
      name: Calibration Move Sensitivity Factor
    gate_still_sensitivity:
      name: Calibration Still Sensitivity Factor
 
button:
  - platform: ld2420
    apply_config:
      name: (b) Apply Config
    factory_reset:
      name: (b) Factory Reset
    restart_module:
      name: (b) Restart Module
    revert_config:
      name: (b) Abort Edits

status_led:
  pin: GPIO2
~~~

yousaf465, 2023-12-01:
> November 11/23 is well after the v1.5.2 firmware. Use the following, and you really only need status_led: by its self to see the LD2420 component state.
>
> ```
> external_components:
>   - source: [...]

bootloop 

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6608
load:0x40078000,len:15060
ho 0 tail 12 room 4
load:0x40080400,len:3816
entry 0x40080698
```

yousaf465, 2023-12-01:
> Which esp32 module are you using and how is it powered? See [#5115 (comment)](https://github.com/esphome/issues/issues/5115#issuecomment-1831124158)

https://www.daraz.pk/products/i300585360-s1971183177.html?urlFlag=true&mp=1

ESP32 CH340 Type-C. connected to the carrier board and powered by USB C 

ld2410C works fine with it

descipher, 2023-12-01:
That's the module I developed the code on, have you run a clean and what's the version of ESPHome?
Post the rest of the config YAML, (redacted form)

yousaf465, 2023-12-01:
```yaml
esphome:
  name: esp32-typec
  friendly_name: ESP32-TypeC

esp32:
  board: esp32dev
  framework:
    type: Esp-IDF

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
  use_address: 192.168.18.167  

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Typec Fallback Hotspot"
    password: "xxxxxxxxxxxx"

captive_portal:

web_server:
  port: 80

#Bluetooth
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: True

uart:
  tx_pin: 17
  rx_pin: 16
  baud_rate: 256000
# Status Led
light:
  - platform: status_led
    name: "Switch state"
    pin: GPIO2
    id: ld2410statusled  
#LD2410 component
ld2410:
#ld2410 binary sensor actual sensors requried  
binary_sensor:
  - platform: ld2410
    has_target:
      name: Presence
      
    has_moving_target:
      name: Moving Target
      on_press:
        then:
        - light.turn_on: ld2410statusled
      on_release:
        then:
        - light.turn_off: ld2410statusled
    has_still_target:
      name: Still Target
    out_pin_presence_status:
      name: out pin presence status
  - platform: gpio
    pin: 33
    name: "PIR Sensor"
    device_class: motion      

#LD2410 text sensors
text_sensor:
  - platform: ld2410
    version:
      name: "firmware version"
    mac_address:
      name: "mac address"

#LD210 cinfig. sensors
sensor:
  - platform: ld2410
    light:
      name: light
    moving_distance:
      name : Moving Distance
    still_distance:
      name: Still Distance
    moving_energy:
      name: Move Energy
    still_energy:
      name: Still Energy
    detection_distance:
      name: Detection Distance
    g0:
      move_energy:
        name: g0 move energy
      still_energy:
        name: g0 still energy
    g1:
      move_energy:
        name: g1 move energy
      still_energy:
        name: g1 still energy
    g2:
      move_energy:
        name: g2 move energy
      still_energy:
        name: g2 still energy
    g3:
      move_energy:
        name: g3 move energy
      still_energy:
        name: g3 still energy
    g4:
      move_energy:
        name: g4 move energy
      still_energy:
        name: g4 still energy
    g5:
      move_energy:
        name: g5 move energy
      still_energy:
        name: g5 still energy
    g6:
      move_energy:
        name: g6 move energy
      still_energy:
        name: g6 still energy
    g7:
      move_energy:
        name: g7 move energy
      still_energy:
        name: g7 still energy
    g8:
      move_energy:
        name: g8 move energy
      still_energy:
        name: g8 still energy

  
number:
  - platform: ld2410
    timeout:
      name: timeout
    light_threshold:
      name: light threshold
    max_move_distance_gate:
      name: max move distance gate
    max_still_distance_gate:
      name: max still distance gate
    g0:
      move_threshold:
        name: g0 move threshold
      still_threshold:
        name: g0 still threshold
    g1:
      move_threshold:
        name: g1 move threshold
      still_threshold:
        name: g1 still threshold
    g2:
      move_threshold:
        name: g2 move threshold
      still_threshold:
        name: g2 still threshold
    g3:
      move_threshold:
        name: g3 move threshold
      still_threshold:
        name: g3 still threshold
    g4:
      move_threshold:
        name: g4 move threshold
      still_threshold:
        name: g4 still threshold
    g5:
      move_threshold:
        name: g5 move threshold
      still_threshold:
        name: g5 still threshold
    g6:
      move_threshold:
        name: g6 move threshold
      still_threshold:
        name: g6 still threshold
    g7:
      move_threshold:
        name: g7 move threshold
      still_threshold:
        name: g7 still threshold
    g8:
      move_threshold:
        name: g8 move threshold
      still_threshold:
        name: g8 still threshold
select:
  - platform: ld2410
    distance_resolution:
      name: "distance resolution"
    baud_rate:
      name: "baud rate"
    light_function:
      name: light function
    out_pin_level:
      name: out pin level
switch:
  - platform: ld2410
    engineering_mode:
      name: "engineering mode"
    bluetooth:
      name: "control bluetooth"
# Restart and safe mode configuration entry
button:
  - platform: restart
    name: LD2410 restart
    icon: "mdi:emoticon-outline"
    on_press:
      - logger.log: "Button pressed"
  - platform: safe_mode
    name: "LD2410 Restart (Safe Mode)"
  #Shutdown configuration entry 
  - platform: shutdown
    name: "LD2410 Shutdown"    

```

descipher, 2023-12-01:
Just tested on type: esp-idf it works as well.
~~~
[C][uart.arduino_esp32:124]: UART Bus 1:
[C][uart.arduino_esp32:125]:   TX Pin: GPIO17
[C][uart.arduino_esp32:126]:   RX Pin: GPIO16
[C][uart.arduino_esp32:128]:   RX Buffer Size: 256
[C][uart.arduino_esp32:130]:   Baud Rate: 115200 baud
[C][uart.arduino_esp32:131]:   Data Bits: 8
[C][uart.arduino_esp32:132]:   Parity: NONE
[C][uart.arduino_esp32:133]:   Stop bits: 1
[C][ld2420:068]: LD2420:
[C][ld2420:069]:   Firmware Version :  v1.5.6
~~~

I don't know why you sent an LD2410 config, that not going to help you.

ride8055, 2023-12-01:
Hey @yousaf465 Try this 
```
uart:
  id: ld2420_uart
  tx_pin: GPIO21 #17
  rx_pin: GPIO19 #16
  baud_rate: 115200
  #parity: NONE
  #stop_bits: 1
```
Change you wires also, this worked for me.

yousaf465, 2023-12-01:
> Just tested on type: esp-idf it works as well.
> 
> ```
> [C][uart.arduino_esp32:124]: UART Bus 1:
> [C][uart.arduino_esp32:125]:   TX Pin: GPIO17
> [C][uart.arduino_esp32:126]:   RX Pin: GPIO16
> [C][uart.arduino_esp32:128]:   RX Buffer Size: 256
> [C][uart.arduino_esp32:130]:   Baud Rate: 115200 baud
> [C][uart.arduino_esp32:131]:   Data Bits: 8
> [C][uart.arduino_esp32:132]:   Parity: NONE
> [C][uart.arduino_esp32:133]:   Stop bits: 1
> [C][ld2420:068]: LD2420:
> [C][ld2420:069]:   Firmware Version :  v1.5.6
> ```
> 
> I don't know why you sent an LD2410 config, that's not going to help you.

nothing connected. bootloop 

```
[C][ld2420:083]: ld2420  Icon: 'mdi:restart'
[C][ld2420:084]: ld2420  Factory Reset: '(b) Factory Reset'
[C][ld2420:084]: ld2420  Icon: 'mdi:database'
[C][ld242
abort() was called at PC 0x401a670a on core 1

Backtrace: 0x40082bc6:0x3ffcf840 0x4009172d:0x3ffcf860 0x400976c2:0x3ffcf880 0x401a670a:0x3ffcf8f0 0x401a520f:0x3ffcf910 0x401a5def:0x3ffcf930 0x400dfc05:0x3ffcf950 0x400e038d:0x3ffcf9a0 0x401bd5f5:0x3ffcf9f0 0x400eaf59:0x3ffcfa10 0x400ed40a:0x3ffcfa40 0x400dc49e:0x3ffcfa60

ELF file SHA256: 377fec2899d3bb96

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6608
load:0x40078000,len:15060
ho 0 tail 12 room 4
load:0x40080400,len:3816
entry 0x40080698
I (29) boot: ESP-IDF 4.4.5 2nd stage bootloader
I (29) boot: compile time 03:16:46
```
```yaml

esphome:
  name: esp32-typec
  friendly_name: ESP32-TypeC

esp32:
  board: esp32dev
  framework:
    type: Esp-IDF

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "T+xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx="

ota:
  password: "xxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: xxxxxxxxxxxxxxxxxxxxxxx
  use_address: 192.168.18.167  

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Typec Fallback Hotspot"
    password: "xxxxxxxxxxxxxxxxxxxxxxx"

captive_portal:

web_server:
  port: 80

#Bluetooth
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: True

external_components:
  - source: github://pr#5865
    components: [ ld2420 ]
    refresh: 0s

uart:
  id: ld2420_uart
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 115200
  parity: NONE
  stop_bits: 1

# The LD2420 has 16 sense gates 0-15 and each gate detects 0.7 meters 15th gate = 9m
ld2420:

text_sensor:
  - platform: ld2420
    fw_version:
      name: LD2420 Firmware

sensor:
  - platform: ld2420
    moving_distance:
      name : Moving Distance

binary_sensor:
  - platform: ld2420
    has_target:
      name: Presence

switch:
  - platform: restart
    name: Restart ESPHome Node

select:
  - platform: ld2420
    operating_mode:
      name: (s) Operating Mode

number:
  - platform: ld2420
    presence_timeout:
      name: Detection Presence Timeout
    min_gate_distance:
      name: Detection Gate Minimum
    max_gate_distance:
      name: Detection Gate Maximum
    gate_select:
      name: Select Gate to Set
    still_threshold:
      name: Set Still Threshold Value
    move_threshold:
      name: Set Move Threshold Value
    gate_move_sensitivity:
      name: Calibration Move Sensitivity Factor
    gate_still_sensitivity:
      name: Calibration Still Sensitivity Factor
 
button:
  - platform: ld2420
    apply_config:
      name: (b) Apply Config
    factory_reset:
      name: (b) Factory Reset
    restart_module:
      name: (b) Restart Module
    revert_config:
      name: (b) Abort Edits

status_led:
  pin: GPIO2

```

yousaf465, 2023-12-01:
> Try removing the blue tooth.

```[C][ld2420:084]: ld2420  Factory Reset: '(b) Factory Reset'
[C][ld2420:084]: ld2420  Icon: 'mdi:database'
[C][ld2420:085]: ld2420  Restart Module: '(b) Restart Module'
[C][ld2420:085]: ld2420  Icon: 'mdi:database'
[C][ld2420:087]: LD2420 Select:
[C][ld2420:088]: ld2420  Operating Mode '(s) Operating Mode'

abort() was called at PC 0x4016441f on core 1

Backtrace:0x400837e1:0x3ffb25c00x4008b819:0x3ffb25e0 0x40090d99:0x3ffb2600 0x4016441f:0x3ffb2680 0x40164466:0x3ffb26a0 0x4016455f:0x3ffb26c0 0x40164f63:0x3ffb26e0 0x400db361:0x3ffb2700 0x400dbb08:0x3ffb2760 0x401793b9:0x3ffb27b0 0x400e63c1:0x3ffb27d0 0x400e88d6:0x3ffb2800 0x400f5d15:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:326]: Log initialized
[C][status_led:014]: Setting up Status LED...
[C][ota:473]: There have been 2 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[C][uart.arduino_esp32:077]: Setting up UART...
[C][ld2420:116]: Setting up LD2420...
[D][ld2420:569]: Sending set config enable command: FF
[E][ld2420:118]: LD2420 module has failed to respond, check baud rate and serial connections.
[E][component:113]: Component ld2420 was marked as failed
```
still boot looping

descipher, 2023-12-01:
Your wired wrong or the module is dead .. [E][component:113]: Component ld2420 was marked as failed.

yousaf465, 2023-12-01:
> Your wired wrong or the module is dead .. [E][component:113]: Component ld2420 was marked as failed.

it is not connected. testing stability only.

descipher, 2023-12-01:
I see, did you try using the arduino framework at any point?

yousaf465, 2023-12-01:
> I see, did you try using the arduino framework at any point?

current config, same issue

```yaml
esphome:
  name: esp32-typec
  friendly_name: ESP32-TypeC

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "T+xxxxxxxxxxxxxxxxxxxx="

ota:
  password: "xxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: xxxxxxxxxxxxxxxxxxxxxxx
  password: xxxxxxxxxxxxxxxxxx
  use_address: 192.168.18.167  

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Typec Fallback Hotspot"
    password: "xxxxxxxxxxxxxxxx"

captive_portal:

web_server:
  port: 80

external_components:
  - source: github://pr#5865
    components: [ ld2420 ]
    refresh: 0s

uart:
  id: ld2420_uart
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 115200
  parity: NONE
  stop_bits: 1

# The LD2420 has 16 sense gates 0-15 and each gate detects 0.7 meters 15th gate = 9m
ld2420:

text_sensor:
  - platform: ld2420
    fw_version:
      name: LD2420 Firmware

sensor:
  - platform: ld2420
    moving_distance:
      name : Moving Distance

binary_sensor:
  - platform: ld2420
    has_target:
      name: Presence

switch:
  - platform: restart
    name: Restart ESPHome Node

select:
  - platform: ld2420
    operating_mode:
      name: (s) Operating Mode

number:
  - platform: ld2420
    presence_timeout:
      name: Detection Presence Timeout
    min_gate_distance:
      name: Detection Gate Minimum
    max_gate_distance:
      name: Detection Gate Maximum
    gate_select:
      name: Select Gate to Set
    still_threshold:
      name: Set Still Threshold Value
    move_threshold:
      name: Set Move Threshold Value
    gate_move_sensitivity:
      name: Calibration Move Sensitivity Factor
    gate_still_sensitivity:
      name: Calibration Still Sensitivity Factor
 
button:
  - platform: ld2420
    apply_config:
      name: (b) Apply Config
    factory_reset:
      name: (b) Factory Reset
    restart_module:
      name: (b) Restart Module
    revert_config:
      name: (b) Abort Edits

status_led:
  pin: GPIO2

```

descipher, 2023-12-01:
I suspect it's a esp type issue, give the pin swap a try:
It could be related to the internal pin mappings.
~~~
uart:
  id: ld2420_uart
  tx_pin: GPIO21 #17
  rx_pin: GPIO19 #16
~~~

yousaf465, 2023-12-01:
> I suspect it's a esp type issue, give the pin swap a try: It could be related to the internal pin mappings.
> 
> ```
> uart:
>   id: ld2420_uart
>   tx_pin: GPIO21 #17
>   rx_pin: GPIO19 #16
> ```
same issue.
@ride8055 

```yaml

external_components:
  - source: github://pr#5865
    components: [ ld2420 ]
    refresh: 0s

uart:
  id: ld2420_uart
  tx_pin: GPIO21 #17
  rx_pin: GPIO19 #16
  baud_rate: 115200
  #parity: NONE
  #stop_bits: 1
# The LD2420 has 16 sense gates 0-15 and each gate detects 0.7 meters 15th gate = 9m
ld2420:

```

```yaml
[C][ld2420:087]: LD2420 Select:
[C][ld2420:088]: ld2420  Operating Mode '(s) Operating Mode'

abort() was called at PC 0x4016426f on core 1

Backtrace:0x400837cd:0x3ffb25c00x4008b805:0x3ffb25e0 0x40090d85:0x3ffb2600 0x4016426f:0x3ffb2680 0x401642b6:0x3ffb26a0 0x401643af:0x3ffb26c0 0x40164db3:0x3ffb26e0 0x400db341:0x3ffb2700 0x400dbae8:0x3ffb2760 0x40179209:0x3ffb27b0 0x400e6279:0x3ffb27d0 0x400e878e:0x3ffb2800 0x400f5b7d:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:326]: Log initialized
[C][ota:473]: There have been 3 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[C][uart.arduino_esp32:077]: Setting up UART...
[C][ld2420:116]: Setting up LD2420...
[D][ld2420:569]: Sending set config enable command: FF
[E][ld2420:118]: LD2420 module has failed to respond, check baud rate and serial connections.
[E][component:113]: Component ld2420 was marked as failed.
[C][wifi:038]: Setting up WiFi...
[C][wifi:051]: Starting WiFi...
```

descipher, 2023-12-01:
If you do the wireless log install take note of the following output:
~~~
INFO ESPHome 2023.11.6
INFO Reading configuration /config/esphome/presence-1.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5865/head <- this one should be here
INFO Generating C++ source...
INFO Compiling app...
Processing presence-1 (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
~~~

ride8055, 2023-12-01:
To be the same test, try the different pins without 
```
external_components:
  - source: github://pr#5865
    components: [ ld2420 ]
    refresh: 0s
```
I only say because I had the same problem as you.
If that doesn't work, wire your ld2420 up to a usb-uart to make sure it's alive.

descipher, 2023-12-02:
Building with the ble_tracker and idf is out, it crashes here. Not enough memory.

descipher, 2023-12-02:
I see random exceptions running the ESP-IDF framework on a esp32dev module. Seems more sensitive to brown outs from what I am observing. The issue does not occur on my AC powered design so it looks like the brown out sensitivity setting in the IDF framework is at a more sensitive threshold than the Arduino framework setting.

yousaf465, 2023-12-02:
```yaml

external_components:
  - source: github://pr#5865
    components: [ ld2420 ]
    refresh: 0s

uart:
  id: ld2420_uart
  tx_pin: GPIO21
  rx_pin: GPIO20
  baud_rate: 115200
  parity: NONE
  stop_bits: 1

# The LD2420 has 16 sense gates 0-15 and each gate detects 0.7 meters 15th gate = 9m
ld2420:

text_sensor:
  - platform: ld2420
    fw_version:
      name: LD2420 Firmware

switch:
  - platform: restart
    name: Restart ESPHome Node

select:
  - platform: ld2420
    operating_mode:
      name: (s) Operating Mode

number:
  - platform: ld2420
    presence_timeout:
      name: Detection Presence Timeout
    min_gate_distance:
      name: Detection Gate Minimum
    max_gate_distance:
      name: Detection Gate Maximum
    gate_select:
      name: Select Gate to Set
    still_threshold:
      name: Set Still Threshold Value
    move_threshold:
      name: Set Move Threshold Value
    gate_move_sensitivity:
      name: Calibration Move Sensitivity Factor
    gate_still_sensitivity:
      name: Calibration Still Sensitivity Factor
```
crashing on m5stack C3U too.

descipher, 2023-12-02:
@yousaf465 Please retry the bug fix PR now. Made a change to the version string init value that appears to make it work better on current releases. (2023.12.0-dev)

yousaf465, 2023-12-03:
got some improvement. Sensor not connected yet. 

[image]

yousaf465, 2023-12-03:
Karachi we are live!! 🥳 🎉 @descipher Thanks it is working fine it seems 
[image]

yousaf465, 2023-12-03:
```
yaml
Linking .pioenvs/ld2420-sample/firmware.elf
RAM:   [=         ]  13.2% (used 43404 bytes from 327680 bytes)
Flash: [=====     ]  53.0% (used 972669 bytes from 1835008 bytes)
Building .pioenvs/ld2420-sample/firmware.bin
```
still enough space for more components.

descipher, 2023-12-04:
@yousaf465 Awesome! Good to see. Thanks for reporting the issue.

yousaf465, 2023-12-04:
> @yousaf465 Awesome! Good to see. Thanks for reporting the issue.

it has got quite a range, way more than 2410B or 2410C.

descipher, 2023-12-04:
Yes, it’s very sensitive when tuned up. It can even detect a finger wiggle in a low noise environment.

descipher, 2023-12-04:
It does not have a light sensor, bluetooth is not critical, really there for config and local proximity control apps.

Soprano30, 2023-12-26:
@yousaf465 I have followed the thread from the beginning but I can't see the solution. I have the same problem with unknown sensor. My LD2420 is still in unknown status. I have it on an ESP32 and another on a Mini D1 8266 and neither of them work. Both unknown. I copied your configuration but it's still the same. I'm very new to this and I can't get it to work. Please could you help me? Thanks a lot.

Soprano30, 2023-12-26:
esphome:
  name: presencia-esp32
  friendly_name: presencia-esp32

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Presencia-Esp32 Fallback Hotspot"
    password: "xxxxxxxxxxxxxxxxxxx"
# Webserver
web_server:
  port: 80

captive_portal:

external_components:
  - source: github://pr#5865
    components: [ ld2420 ]
    refresh: 1min

uart:
  id: ld2420_uart
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 256000
  parity: NONE
  stop_bits: 1
  
ld2420: 

text_sensor:
  - platform: ld2420
    fw_version:
      name: LD2420 Firmware

sensor:
  - platform: ld2420
    moving_distance:
      name : Movimiento

binary_sensor:
  - platform: ld2420
    has_target:
      name: Presencia

switch:
  - platform: restart
    name: Restart ESPHome Node

select:
  - platform: ld2420
    operating_mode:
      name: Operating Mode

number:
  - platform: ld2420
    presence_timeout:
      name: Detection Presence Timeout
    min_gate_distance:
      name: Detection Gate Minimum
    max_gate_distance:
      name: Detection Gate Maximum

    gate_select:
      name: Select Gate to Set
    still_threshold:
      name: Set Still Threshold Value
    move_threshold:
      name: Set Move Threshold Value

    gate_move_sensitivity:
      name: Move Calibration Sensitivity Factor
    gate_still_sensitivity:
      name: Still Calibration Sensitivity Factor

    gate_0:
      move_threshold:
        name: Gate 0 Move Threshold
      still_threshold:
        name: Gate 0 Still Threshold
    gate_1:
      move_threshold:
        name: Gate 1 Move Threshold
      still_threshold:
        name: Gate 1 Still Threshold

button:
  - platform: ld2420
    apply_config:
      name: Apply Config
    factory_reset:
      name: Factory Reset
    restart_module:
      name: Restart Module
    revert_config:
      name: Undo Edits

status_led:
  pin: GPIO2

yousaf465, 2023-12-27:
@Soprano30 did you try using this with a serial to UART adapter and Hilink's desktop app.

[49 more replies]

closed 2024-03-18 by jesserockz