esphome/issues #5116: Wifi not connecting after update to 2023.11.0

grego1981, 2023-11-16:
### The problem

I have updated 2 wemos NodeMCU to the new version and after that the devices went offline, I have hooked them up to USB and get the following logs. No change has been made to my wifi network. Also although from the log the local AP is activated, it is not scanned by any of my devices. 

### Which version of ESPHome has the issue?

2023.11.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.2

### What platform are you using?

ESP8266

### Board

Wemos NodeMCU

### Component causing the issue

uart

### Example YAML snippet

```yaml
esphome:
  name: children-bedroom-air-sensor
  friendly_name: children_bedroom_air_sensor

esp8266:
  board: esp01_1m

# Enable logging
logger:
  level: DEBUG
  hardware_uart: UART0
  #baud_rate: 0
# Enable Home Assistant API
api:
  encryption:
    key: "Xn0Yye6o1Sf4hs5aZd9e2Lt0sSZKix6NmSL8nNVTwMo="

web_server:
  port: 80
  version: 2
  include_internal: true

ota:
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  #fast_connect: true
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Children-Bedroom-Air-Sensor"
    password: "42yUjEbMJ4DM"

captive_portal:

uart:
  rx_pin: GPIO4
  baud_rate: 9600

sensor:
  - platform: pm1006
    pm_2_5:
      name: "Particulate Matter 2.5µm Concentration"
```

### Anything in the logs that might be useful for us?

```txt
[C][ota:473]: There have been 1 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[C][uart.arduino_esp8266:059]: Setting up UART bus...
[C][wifi:038]: Setting up WiFi...
[C][wifi:051]: Starting WiFi...
[C][wifi:052]:   Local MAC: 2C:3A:E8:2F:7F:09
[D][wifi:441]: Starting scan...
[D][pm1006:091]: Got PM2.5 Concentration: 0 µg/m³
[D][sensor:093]: 'Particulate Matter 2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[D][pm1006:091]: Got PM2.5 Concentration: 0 µg/m³
[D][sensor:093]: 'Particulate Matter 2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[D][wifi:456]: Found networks:
[D][wifi:458]:   No network found!
[D][pm1006:091]: Got PM2.5 Concentration: 0 µg/m³
[D][sensor:093]: 'Particulate Matter 2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[D][pm1006:091]: Got PM2.5 Concentration: 0 µg/m³
[D][sensor:093]: 'Particulate Matter 2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[D][pm1006:091]: Got PM2.5 Concentration: 0 µg/m³
[D][sensor:093]: 'Particulate Matter 2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[D][wifi:441]: Starting scan...
[D][pm1006:091]: Got PM2.5 Concentration: 0 µg/m³
[D][sensor:093]: 'Particulate Matter 2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[D][pm1006:091]: Got PM2.5 Concentration: 0 µg/m³
[D][sensor:093]: 'Particulate Matter 2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[D][wifi:456]: Found networks:
[D][wifi:458]:   No network found!
[D][wifi:441]: Starting scan...
[D][wifi:456]: Found networks:
[D][wifi:458]:   No network found!
[D][wifi:441]: Starting scan...
[D][pm1006:091]: Got PM2.5 Concentration: 0 µg/m³
[D][sensor:093]: 'Particulate Matter 2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[D][pm1006:091]: Got PM2.5 Concentration: 0 µg/m³
[D][sensor:093]: 'Particulate Matter 2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[D][pm1006:091]: Got PM2.5 Concentration: 0 µg/m³
[D][sensor:093]: 'Particulate Matter 2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[D][pm1006:091]: Got PM2.5 Concentration: 0 µg/m³
[D][sensor:093]: 'Particulate Matter 2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[D][wifi:456]: Found networks:
[D][wifi:458]:   No network found!
[D][pm1006:091]: Got PM2.5 Concentration: 0 µg/m³
[D][sensor:093]: 'Particulate Matter 2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[D][pm1006:091]: Got PM2.5 Concentration: 0 µg/m³
[D][sensor:093]: 'Particulate Matter 2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[D][wifi:441]: Starting scan...
[D][pm1006:091]: Got PM2.5 Concentration: 0 µg/m³
[D][sensor:093]: 'Particulate Matter 2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[D][wifi:456]: Found networks:
[D][wifi:458]:   No network found!
[D][wifi:441]: Starting scan...
[I][wifi:165]: Starting fallback AP!
[C][wifi:239]: Setting up AP...
[C][wifi:241]:   AP SSID: 'Children-Bedroom-Air-Sensor'
[C][wifi:242]:   AP Password: '42yUjEbMJ4DM'
[C][wifi:251]:   IP Address: 192.168.4.1
```

clownjoy, 2023-11-16:
Same here on my devices. What could be interesting, when removing any sensor configuration just keeping bare wifi setup, it works. It seems like it  breaks after adding i2c or any GPIO manipulation / initial setup: 
i2c:
  sda: GPIO4
  scl: GPIO5
  scan: false
  id: bus_a
  frequency: 800kHz

grego1981, 2023-11-16:
Can verify that without sensor configuration it connects just fine.

grego1981, 2023-11-25:
Today I checked on different pins to no avail, if the sensor config is active WiFi does not work. Checked gpios 4 5 12 13 14. Is there a solution??

chillfactor17, 2024-01-03:
Same probleme with my Sonoff Basic:

Updated to the latest ESPhome version (2023.12.5) and using a I2C Sensor @GPIO14, no wifi connection possbile.
UART Log:
    [D][wifi:470]: Found networks:
    [D][wifi:472]:   No network found!

    
Compiled without the I2C sensor: working
Again compiled with I2C sensor: NOT working

Recompiled after downgrade of the ESPhome version to 2023.10.6 works perfectly with the sensor as desired

clownjoy, 2024-05-22:
I'm adding a comment after while, which is not very technical, but I believe it could help to identify the issue. Since release 2023.11.0 There were some releases that  bring ESP8266 / nodemcuv2 back to life again. It was at least the release 2024.3.0 and some 2024.2.* releases. 2024.3.0 was the last working one. Would be nice to see the code comparison between 2024.3.0 and 2024.3.1. What I could see, there are more boards based on ESP8266 which is affected by this. For ex. very popular Sonoff RF switch RFR2.