esphome/issues #5117: Can't collect logs from device

X4V1, 2023-11-16:
### The problem

I just updated to the version 2023.11.1 and I'm not not able to connect to show logs anymore.
I installed esphome with pip (manual install) on a windows 11 machine.
I was able to flash a running device but can't see logs.
After flashing the device it shows an error message
[image]
If I go the web interface, everything seems to be running as expected and logs are visible. It's just not possible to get logs with the `esphome logs whatever.yaml` command. Am I missing something ?

I also tried to get logs from another device running an older version of esphome and I'm getting the same error.

### Which version of ESPHome has the issue?

2023.11.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2023.11.2

### What platform are you using?

ESP32

### Board

generic esp32

marshn, 2023-11-16:
I am seeing exactly the same thing when Windows 10, pip and v2023.11.0. Was not happening with v2023.10.6

ssieb, 2023-11-16:
@bdraco maybe related to the logger changes and windows?

bdraco, 2023-11-16:
@X4V1 you can work around this by uninstalling `aioesphomeapi` and reinstalling it with `SKIP_CYTHON=1` in the env

bdraco, 2023-11-16:
@X4V1 
<https://github.com/esphome/aioesphomeapi/releases/tag/v18.5.2> cooking now <https://github.com/esphome/aioesphomeapi/actions/runs/6897892491>

Can you give it a try once its published to pypi?

closed 2023-11-17 by bdraco

X4V1, 2023-11-17:
@bdraco Thank you for your answer. It works fine with the updated version of aioesphomeapi (18.5.2).

bdraco, 2023-11-17:
Thanks for confirming