esphome/issues #5119: esp32_ble_tracker scan is not restarted after active connection

fightforlife, 2023-11-16:
### The problem

I have a ESP32 with Ethernet running as a bluetooth proxy for HomeAssistant.

I am using the recommended parameters on latest ESP-IDF:
esp32_ble_tracker:
  scan_parameters:
    active: false #to not drain the battery of passive BLE sensors
    interval: 1100ms
    window: 1100ms

bluetooth_proxy:
  active: true  #to access data on active ble actors
  cache_services: true

Once a day the it happens that the esp32_ble_tracker component does not restart the paused scan after a connection is made.
Then all BLE connected devices are reported as offline, since no scan is running.
This is fixed by rebooting the ESP32.
Note: I also have this problem using Wifi and/or using the arduino framework.

I have a 50mb logfile from Boot till the error happens. Below is an excerpt of the relevant parts.

### Which version of ESPHome has the issue?

ESPHome version 2023.11.0b3 compiled on Nov 12 2023, 09:36:50

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Board

ESP32 + WS5500 Ethernet

### Component causing the issue

esp32_ble_tracker

### Example YAML snippet

```yaml
substitutions:
  name: gateway2
  friendly_name: Gateway2

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

esp32:
  board: esp32doit-devkit-v1
  variant: esp32
  framework:
    type: esp-idf
    version: latest
    platform_version: 6.4.0

external_components:
  - source:
      type: local
      path: ./components
    components:
      - ethernet
      
esp32_ble_tracker:
  scan_parameters:
[... 442 lines cut ...]
[09:23:29][V][bluetooth_proxy:097]: [1] Free connection
[09:23:29][V][bluetooth_proxy:099]: [2] Used connection by [00:1A:22:0B:06:46]
[09:23:29][I][esp32_ble_client:068]: [0] [00:1A:22:0B:07:65] 0x00 Attempting BLE connection
[09:23:29][V][esp-idf:000]: W (4754756) BT_L2CAP: L2CAP - LE - cannot start new connection at conn st: 1

[09:23:29][I][bluetooth_proxy:277]: [1] [00:1A:22:0B:07:86] Connecting v3 with cache
[09:23:29][V][bluetooth_proxy:099]: [0] Used connection by [00:1A:22:0B:07:65]
[09:23:29][V][bluetooth_proxy:099]: [1] Used connection by [00:1A:22:0B:07:86]
[09:23:29][V][bluetooth_proxy:099]: [2] Used connection by [00:1A:22:0B:06:46]
[09:23:40][V][esp-idf:000]: E (4765815) BT_L2CAP: L2CAP got BLE scanner conn_comp in bad state: 0

[09:23:40][V][esp-idf:000]: E (4765880) BT_L2CAP: L2CAP - rcvd ACL for unknown handle:0 ls:0 cid:6 opcode:11 cur count:0

[09:23:49][I][esp32_ble_client:086]: [0] [00:1A:22:0B:07:65] Disconnecting.
[09:23:49][I][esp32_ble_client:068]: [1] [00:1A:22:0B:07:86] 0x00 Attempting BLE connection
[09:23:49][V][esp-idf:000]: W (4774758) BT_L2CAP: L2CAP - LE - cannot start new connection at conn st: 1

[09:23:49][I][esp32_ble_client:086]: [1] [00:1A:22:0B:07:86] Disconnecting.
[09:25:10][V][esp-idf:000]: W (4855915) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16

[13:35:53][V][web_server_idf:066]: Enter AsyncWebServer::request_handler. method=1, uri=/
[13:35:53][V][web_server_idf:066]: Enter AsyncWebServer::request_handler. method=1, uri=/events
[13:35:53][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[13:35:53][V][json:058]: Size after shrink 132 bytes
```

bdraco, 2023-11-20:
It looks like your esp is always trying to establish a connection or disconnecting so it won’t start the scan until the connect or disconnect finishes

There are some known issues with esp-idf taking a long time to disconnect which are fixed in 4.4.6, unfortunately ESPHome still uses 4.4.5

fightforlife, 2023-11-20:
That sounds promising! 
But currently I am using latest esp-idf which should result in v5.1. 
Do you have a link to the relevant changes on 4.4.6 maybe they were not ported to 5.1?

In the meantime I will try with esp-idf 5.1.2 and 4.4.6.

bdraco, 2023-11-20:
https://github.com/espressif/esp-idf/releases/tag/v4.4.6

Fixed BLE disconnection failure on ESP32 (https://github.com/espressif/esp-idf/commit/efdd0859f6c1e8dc524b7125ecaafd7e4da15411)

fightforlife, 2023-11-20:
I checked the code for the dfferent Branches, and it seems that the fix is already availablwe atleast in 5.2 5.1 5.0 4.4 4.3.
So I think this is another case. Is there a possibility to compile with esp-idf logs? Or should i just try with very verbose.

jesserockz, 2023-11-21:
Unfortunately 4.4.6 also has not made it onto the platformio registry which is kinda annoying
https://registry.platformio.org/tools/platformio/framework-espidf/versions

fightforlife, 2023-12-03:
Current esphome beta with esp-idf 5.1.1 seems to be much more stable.
Will wait 1-2 days before closing.

bdraco, 2023-12-05:
Lets leave it open until we bump ESP-IDF since we know what the solution is.

At this point we are waiting for platformio to release the new 4.4.6 idf

fightforlife, 2023-12-16:
It started happening again after 2 weeks with 5.5.1.
Need some time to get logs.

blind-oracle, 2023-12-22:
Same here for me also with Ethernet ESP32. Some random lithium BMS that's lying near the ESP is lost from Home Assistant after some hours of uptime. Then, after the ESP reboot it reappears instantly.

Gonna try setting `active: false` for proxy and tracker and see if that helps. Or switch to beta.

bdraco, 2024-01-01:
Looks like 4.4.6 is now available on platform IO

bdraco, 2024-01-01:
As is 5.1.2

fightforlife, 2024-01-05:
Had this happen again this night with 5.1.2. 
Will have to find a good way to log this.

bdraco, 2024-01-05:
Maybe you have a device stuck in a connecting state somehow.

Maybe add logging of the number of each of these 
https://github.com/esphome/esphome/blob/a2e152ad1252444a9d12e3a129270f62076d4c12/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp#L102

loca5790, 2024-01-22:
I'm having the same issue, nearly identical yaml.  Any success?

fightforlife, 2024-01-23:
Last week I had this happening 1-2 per day. 
Then I recompiled again with esp-idf 5.1.2 and the esphome beta branch. 
Now it is already stable for 3 days in a row.
I am pretty sure it will start happening again. Then I will add the logging mentioned above.

blind-oracle, 2024-01-23:
I've fallen back to `active: false` and it seems to be stable that way.

bdraco, 2024-01-24:
I have `active: false` on all my ESPs so that likely why I never saw this issue

bdraco, 2024-02-17:
I had one fail today with 4.4.6 and passive.  It took > 1000 hours of uptime to hit the failure case unfortunately.

bdraco, 2024-02-18:
[5.2](https://github.com/espressif/esp-idf/releases/tag/v5.2) is out with some more BLE fixes.

fightforlife, 2024-03-03:
I created an automation in HA whcih reboots my btproxy when at least one BT Devcie is not seen in 5 minutes.
it is just a bandage, but at least it heals itself. Uptime is quite a mixed bag.
Maximum being around 30h
[image]

bdraco, 2024-05-08:
4.4.7 has some more fixes
https://github.com/espressif/esp-idf/releases/tag/v4.4.7

Fixed scan event timeout on ESP32. ( https://github.com/espressif/esp-idf/commit/597a4d49261c1092dbe6e391df1b01d4cc2f021b)

fightforlife, 2024-11-25:
For me it still happens once a day that the ble scan seems to crash and just does not run anymore. The rest of the esphome works normally. reboot fixes this.

I will try to get very Verbose logs.

bdraco, 2024-11-25:
Did you try 4.4.8?

There were some more fixes in that version

bdraco, 2024-11-25:
Nevermind 5.3.1 has those fixes as well

tomaszduda23, 2024-12-14:
Mine gets stuck in connecting state. I have impression that there is a few problems though.
```
[19:02:03][V][esp32_ble:339]: (BLE) gap_event_handler - 3
[19:02:03][V][esp32_ble_tracker:411]: gap_scan_result - event 0
[19:02:03][V][bluetooth_proxy:059]: Proxying raw packet from yy, length 36. RSSI: -82 dB
[19:02:04][V][bluetooth_proxy:061]: Proxying 1 packets
[19:02:04][D][esp32_ble_client:110]: [3] [yy] Found device
[19:02:04][D][esp32_ble_tracker:703]: Found device yy RSSI=-82
[19:02:04][D][esp32_ble_tracker:724]:   Address Type: PUBLIC
[19:02:04][D][esp32_ble_tracker:726]:   Name: 'ATC_'
[19:02:04][D][esp32_ble_tracker:231]: Pausing scan to make connection...
[19:02:04][V][esp32_ble:339]: (BLE) gap_event_handler - 3
[19:02:04][V][esp32_ble_tracker:411]: gap_scan_result - event 0
[19:02:04][V][esp32_ble:339]: (BLE) gap_event_handler - 18
[19:02:04][V][esp32_ble_tracker:406]: gap_scan_stop_complete - status 0
[19:02:04][D][esp32_ble_tracker:116]: connecting: 0, discovered 1, searching 0, disconnecting 0
[19:02:04][V][bluetooth_proxy:059]: Proxying raw packet from yyyy, length 61. RSSI: -79 dB
[19:02:04][V][bluetooth_proxy:061]: Proxying 1 packets
[19:02:04][I][esp32_ble_client:067]: [3] p=[yy] 0x00 Attempting BLE connection
[19:02:04][D][esp32_ble_tracker:116]: connecting: 1, discovered 0, searching 0, disconnecting 0
[19:02:19][V][sensor:043]: 'esp-livingroom-sensors_uptime': Received new state 52.255001
[19:02:19][D][sensor:094]: 'esp-livingroom-sensors_uptime': Sending state 52.25500 s with 0 decimals of accuracy
[19:02:27][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[19:02:27][D][esp32.preferences:114]: Saving 1 preferences to flash...
[19:02:27][V][esp32.preferences:126]: sync: key: 233825507, len: 4
[19:02:27][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[19:03:19][V][sensor:043]: 'esp-livingroom-sensors_uptime': Received new state 112.254997
[19:03:19][D][sensor:094]: 'esp-livingroom-sensors_uptime': Sending state 112.25500 s with 0 decimals of accuracy
[19:04:19][V][sensor:043]: 'esp-livingroom-sensors_uptime': Received new state 172.261002
[19:04:19][D][sensor:094]: 'esp-livingroom-sensors_uptime': Sending state 172.26100 s with 0 decimals of accuracy
[19:05:19][V][sensor:043]: 'esp-livingroom-sensors_uptime': Received new state 232.307999
[19:05:19][D][sensor:094]: 'esp-livingroom-sensors_uptime': Sending state 232.30800 s with 0 decimals of accuracy
[19:05:19][D][time:051]: Synchronized time: 2024-12-14 19:05:19
```

tomaszduda23, 2024-12-25:
aah, I'm missing https://github.com/esphome/esphome/pull/7960 after merge.

tomaszduda23, 2024-12-31:
Another interesting case. Everything works as expected besides `ESP_GAP_SEARCH_INQ_RES_EVT`. Any idea how to handle that?

@bdraco would you be down to add `reboot_timeout` to esp32_ble_tracker. Which could restart esp if no results e.g. for 15 min

```
[05:09:34][I][app:100]: ESPHome version 2024.12.1 compiled on Dec 25 2024, 19:23:31
[05:09:34][C][bluetooth_proxy:091]: Bluetooth Proxy:
[05:09:34][C][bluetooth_proxy:092]:   Active: NO
[05:09:34][C][bluetooth_proxy:093]:   Connections: 0
[05:09:34][C][bluetooth_proxy:094]:   Raw advertisements: YES
[05:09:34][C][display.pvvx_mithermometer:011]: PVVX MiThermometer display:
[05:09:34][C][display.pvvx_mithermometer:012]:   MAC address           : 
[05:09:34][C][display.pvvx_mithermometer:013]:   Service UUID          : 00001F10-0000-1000-8000-00805F9B34FB
[05:09:34][C][display.pvvx_mithermometer:014]:   Characteristic UUID   : 00001F1F-0000-1000-8000-00805F9B34FB
[05:09:34][C][display.pvvx_mithermometer:015]:   Auto clear            : YES
[05:09:34][C][display.pvvx_mithermometer:017]:   Set time on connection: YES
[05:09:34][C][display.pvvx_mithermometer:019]:   Disconnect delay      : 30000ms
[05:09:34][C][display.pvvx_mithermometer:020]:   Update Interval: 86400.0s
[05:09:34][C][esp32_ble:399]: ESP32 BLE:
[05:09:34][C][esp32_ble:401]:   MAC address: 
[05:09:34][C][esp32_ble:402]:   IO Capability: keyboard_only
[05:09:34][C][esp32_ble_tracker:677]: BLE Tracker:
[05:09:34][C][esp32_ble_tracker:678]:   Scan Duration: 300 s
[05:09:34][C][esp32_ble_tracker:679]:   Scan Interval: 1100.0 ms
[05:09:34][C][esp32_ble_tracker:680]:   Scan Window: 1100.0 ms
[05:09:34][C][esp32_ble_tracker:681]:   Scan Type: ACTIVE
[05:09:34][C][esp32_ble_tracker:682]:   Continuous Scanning: YES
[05:09:34][C][esp32_ble_tracker:683]:   Scanner Idle: NO
[05:09:34][C][esp32_ble_tracker:684]:   Scan End: NO
[05:09:34][C][esp32_ble_tracker:686]:   Connecting: 0, discovered: 0, searching: 0, disconnecting: 0
[05:09:34][C][ble_client:027]: BLE Client:
[05:09:34][C][esp32_ble_client:048]:   Address: 
[05:09:34][C][esp32_ble_client:049]:   Auto-Connect: TRUE
[05:09:34][C][esp32_ble_client:083]:   State: IDLE
[05:12:15][V][esp32_ble:339]: (BLE) gap_event_handler - 3
[05:12:15][V][esp32_ble_tracker:411]: gap_scan_result - event 1
[05:12:15][D][esp32_ble_tracker:282]: Starting scan...
[05:12:15][V][esp32_ble:339]: (BLE) gap_event_handler - 2
[05:12:15][V][esp32_ble_tracker:384]: gap_scan_set_param_complete - status 0
[05:12:15][V][esp32_ble:339]: (BLE) gap_event_handler - 7
[05:12:15][V][esp32_ble_tracker:393]: gap_scan_start_complete - status 0
[05:17:15][V][esp32_ble:339]: (BLE) gap_event_handler - 3
[05:17:15][V][esp32_ble_tracker:411]: gap_scan_result - event 1
[05:17:15][D][esp32_ble_tracker:282]: Starting scan...
[05:17:15][V][esp32_ble:339]: (BLE) gap_event_handler - 2
[05:17:15][V][esp32_ble_tracker:384]: gap_scan_set_param_complete - status 0
[05:17:16][V][esp32_ble:339]: (BLE) gap_event_handler - 7
[05:17:16][V][esp32_ble_tracker:393]: gap_scan_start_complete - status 0
[05:22:16][V][esp32_ble:339]: (BLE) gap_event_handler - 3
[05:22:16][V][esp32_ble_tracker:411]: gap_scan_result - event 1
[05:22:16][D][esp32_ble_tracker:282]: Starting scan...
[05:22:16][V][esp32_ble:339]: (BLE) gap_event_handler - 2
[05:22:16][V][esp32_ble_tracker:384]: gap_scan_set_param_complete - status 0
[05:22:16][V][esp32_ble:339]: (BLE) gap_event_handler - 7
[05:22:16][V][esp32_ble_tracker:393]: gap_scan_start_complete - status 0

```

bdraco, 2024-12-31:
I think we could only do that if we know that we were previously getting results as there may be cases where there are no ble devices in range and we don't want to reboot

tomaszduda23, 2025-01-04:
It has happened again. Uptime is a day or two. After that there is no callback with scan results.

fightforlife, 2025-01-04:
Some things definitely are better with the latest PRs.
I had a 8 day uptime, which never happened before. 
Esp32 restarted on its own. (Reason not specified, might not be able related)

bdraco, 2025-02-26:
Possible fix https://github.com/esphome/esphome/pull/8328

bdraco, 2025-04-22:
I found a race in the scanner logic.

Can you give https://github.com/esphome/esphome/pull/8601 a try?