esphome/issues #5120: Unreliable SN75HC595N output with SPI

Azylog, 2023-11-16:
### The problem

Hello,
I'm trying to use a SN74HC595N using the SPI bus as clock and data source to switch a LED (that's only for testing).
It basically works ... a few times. After a few switches, it stops powering on or off my output LED and miss one or more switches, dephasing between the esphome switch state and real output state.

To check if the fault is at my breadboard, I tried same hardware but with the GPIO configuration of SN74HC595 component and everything's fine.
I can see in the sources that the SPI clock rate is forced at 4Mhz for the SN74HC595SPIComponent. As I'm testing on a breadboard, could it be the cause of this erratic behavior ?

Could we add an option to downscale the SPI bus frequency for this use-case ?

### Which version of ESPHome has the issue?

2023.11.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.1

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

SN74HC595

### Example YAML snippet

```yaml
esphome:
  name: sn74hc595-test
  friendly_name: SN74HC595-test

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

web_server:
  port: 80

spi:
  id: spi_bus0
  clk_pin: 33
  miso_pin: 25
  mosi_pin: 26

sn74hc595:
  - id: 'sn74hc595_hub'
    latch_pin: 32
    oe_pin: 14
    spi_id: spi_bus0
    sr_count: 1
#    data_rate: 200kHz

switch:
  - platform: gpio
    name: "SN74HC595 Pin #0"
    id: SN74HC595_0
    pin:
      sn74hc595: sn74hc595_hub
      number: 0
      inverted: true
```

Azylog, 2023-11-16:
I didn't thought of that at first, but I have plugged the HC595's output to a simple GPIO binary sensor.
As you can see in the following log, on the second switch to 'ON' the GPIO input didn't change and then when the HC595's output was turned OFF, the input detected a ON state :

```
[16:35:22][D][switch:016]: 'SN74HC595 Pin #0' Turning OFF.
[16:35:22][D][switch:055]: 'SN74HC595 Pin #0': Sending state OFF
[16:35:22][D][binary_sensor:036]: 'SN74HC595 output #0 state': Sending state OFF
[16:35:28][D][switch:012]: 'SN74HC595 Pin #0' Turning ON.
[16:35:28][D][switch:055]: 'SN74HC595 Pin #0': Sending state ON
[16:35:31][D][switch:016]: 'SN74HC595 Pin #0' Turning OFF.
[16:35:31][D][switch:055]: 'SN74HC595 Pin #0': Sending state OFF
[16:35:31][D][binary_sensor:036]: 'SN74HC595 output #0 state': Sending state ON
```

Azylog, 2023-11-21:
@angelnu Sorry to bother you but as you seems to be the main contributor for the SPI addition to 74HC595, perhaps have you an idea to understand what's wrong my my setup and the component ?
Thanks for your help

angelnu, 2023-11-24:
Hi @Azylog - I do not have this device myself ( @notsonominal does).

But let us see if we can debug the issue - do you see the same issue when running in non-spi mode?

Azylog, 2023-11-27:
Thanks for answering !
For reference, here's the schematic of my breadboard :
[image]

That's the simpler wiring AFAIK. Has you can see, I added a pull-up resistor for OE as stated in doc but not sure it's really useful.

First, I'm trying with non-spi mode (TL;DR : it works).
Here's the complete config : 
```
esphome:
  name: sn74hc595-test
  friendly_name: SN74HC595-test

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxx"

ota:
  password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

web_server:
  port: 80

sn74hc595:
  - id: 'sn74hc595_hub'
    clock_pin: 33
    data_pin: 25
    latch_pin: 32
    oe_pin: 14
    sr_count: 1

switch:
  - platform: gpio
    name: "SN74HC595 Pin #0"
    id: SN74HC595_0
    pin:
      sn74hc595: sn74hc595_hub
      number: 0
      inverted: true

binary_sensor:
  - platform: gpio
    pin: 21
    name: "GPIO21 input state"
```
The GPIO21 strictly follows the switch on first bit of HC595 : 
```
[22:26:26][D][switch:012]: 'SN74HC595 Pin #0' Turning ON.
[22:26:26][D][switch:055]: 'SN74HC595 Pin #0': Sending state ON
[22:26:26][D][binary_sensor:036]: 'GPIO21 input state': Sending state ON
[22:26:27][D][switch:016]: 'SN74HC595 Pin #0' Turning OFF.
[22:26:27][D][switch:055]: 'SN74HC595 Pin #0': Sending state OFF
[22:26:27][D][binary_sensor:036]: 'GPIO21 input state': Sending state OFF
[22:26:27][D][switch:012]: 'SN74HC595 Pin #0' Turning ON.
[22:26:27][D][switch:055]: 'SN74HC595 Pin #0': Sending state ON
[22:26:27][D][binary_sensor:036]: 'GPIO21 input state': Sending state ON
[22:26:28][D][switch:016]: 'SN74HC595 Pin #0' Turning OFF.
[22:26:28][D][switch:055]: 'SN74HC595 Pin #0': Sending state OFF
[22:26:28][D][binary_sensor:036]: 'GPIO21 input state': Sending state OFF
[22:26:29][D][switch:012]: 'SN74HC595 Pin #0' Turning ON.
[22:26:29][D][switch:055]: 'SN74HC595 Pin #0': Sending state ON
[22:26:29][D][binary_sensor:036]: 'GPIO21 input state': Sending state ON
[22:26:30][D][switch:016]: 'SN74HC595 Pin #0' Turning OFF.
[22:26:30][D][switch:055]: 'SN74HC595 Pin #0': Sending state OFF
[22:26:30][D][binary_sensor:036]: 'GPIO21 input state': Sending state OFF
```
... and so on.

Then, I'm moving to spi mode : 
```
esphome:
  name: sn74hc595-test
  friendly_name: SN74HC595-test

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxx"

ota:
  password: "xxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

web_server:
  port: 80

spi:
  id: spi_bus0
  clk_pin: 33
  miso_pin: 25

sn74hc595:
  - id: 'sn74hc595_hub'
    latch_pin: 32
    oe_pin: 14
    sr_count: 1
    spi_id: spi_bus0

switch:
  - platform: gpio
    name: "SN74HC595 Pin #0"
    id: SN74HC595_0
    pin:
      sn74hc595: sn74hc595_hub
      number: 0
      inverted: true

binary_sensor:
  - platform: gpio
    pin: 21
    name: "GPIO21 input state"
```
And the output becomes unpredictable : 
```
[15:16:50][D][switch:012]: 'SN74HC595 Pin #0' Turning ON.
[15:16:50][D][switch:055]: 'SN74HC595 Pin #0': Sending state ON
[15:16:50][D][binary_sensor:036]: 'GPIO21 input state': Sending state ON
[15:16:52][D][switch:016]: 'SN74HC595 Pin #0' Turning OFF.
[15:16:52][D][switch:055]: 'SN74HC595 Pin #0': Sending state OFF
[15:16:52][D][binary_sensor:036]: 'GPIO21 input state': Sending state OFF
[15:16:53][D][switch:012]: 'SN74HC595 Pin #0' Turning ON.
[15:16:53][D][switch:055]: 'SN74HC595 Pin #0': Sending state ON
[15:16:54][D][switch:016]: 'SN74HC595 Pin #0' Turning OFF.
[15:16:54][D][switch:055]: 'SN74HC595 Pin #0': Sending state OFF
[15:16:56][D][switch:012]: 'SN74HC595 Pin #0' Turning ON.
[15:16:56][D][switch:055]: 'SN74HC595 Pin #0': Sending state ON
```
From time to time, the input is flipped but it can even becomes ON when the output is supposed to be OFF and the inverse.

uli-rpi, 2024-02-26:
@PandemiK911

``` 
spi:
  id: spi_bus0
  clk_pin: 33
  miso_pin: 25
``` 

I think, you should use the **mosi_pin** (master out slave in), instead of the miso_pin ?

PandemiK911, 2024-02-26:
Perhaps, thanks for checking !
My breadboard is currently used for another project. I'll try as soon as possible (before end of week, probably). I'll let you know if that was my mistake.

PandemiK911, 2024-03-02:
@uli-rpi You are right : using GPIO19 as mosi solved my problem. Thanks for spotting my error.

I'm really wondering why it seemed to work with miso_pin. The ESP32 should set it to high from time to time and makes the FIRST output of the 595 switch from time to time. I've been able to use the slow_pwn integration with the 595's output and it works like a charm.

Adding a small example of SPI bus setup in the doc could probably help too-fast readers like me.

I also had some trouble with the oe pin of the SN74HC595 : it should absolutely NOT left floating.

So .... i think we can consider this one as solved, my apologies.

closed 2024-03-02 by PandemiK911

uli-rpi, 2024-03-02:
@PandemiK911 Connect OE always to GND, and everything will be fine.

alex-mextner, 2024-04-06:
> Adding a small example of SPI bus setup in the doc could probably help too-fast readers like me.

Totally agree, current documentation is totally incorrect:
https://esphome.io/components/sn74hc595#over-spi
```
# Example configuration entry
spi:
sn74hc595:
  - id: 'sn74hc595_hub'
    latch_pin: D7
    oe_pin: D6
    sr_count: 2

# Individual outputs
switch:
  - platform: gpio
    name: "SN74HC595 Pin #0"
    pin:
      sn74hc595: sn74hc595_hub
      # Use pin number 0
      number: 0
      inverted: false
```
1. no spi: section config
2. no spi_id
3. even with that it does not work with Wemos D1 Mini Lite