esphome/issues #5121: Can no longer use custom partitions table due to the recent fix " Generate partitions.csv based on flash size #5697 "

illigtr, 2023-11-16:
### The problem

Updating an existing ESP Home ESP32 device with 2MB memory now fails, due to the code now attempting to build a custom partition and ignoring an existing custom partition.

### Which version of ESPHome has the issue?

2023.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.2

### What platform are you using?

ESP32-IDF

### Board

esp32-c3-devkitm-1

### Component causing the issue

    esphome/esphome/components/esp32  /__init__.py

### Example YAML snippet

```yaml
esphome:
  name: "water"
  friendly_name: Hot Water
  platformio_options:
    board_upload.flash_size: 2MB
    board_upload.maximum_size: 2097152
    board_build.partitions: "/config/esphome/minimal.csv"
    board_build.flash_mode: dio
#    build_flags: 
#      - "-include \"fix.h\""

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
    version: recommended
    # Custom sdkconfig options
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
    # Advanced tweaking options
    advanced:
      ignore_efuse_mac_crc: false
```

### Anything in the logs that might be useful for us?

```txt
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1040, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1027, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 457, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 193, in write_cpp
    return write_cpp_file()
           ^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 211, in write_cpp_file
    writer.write_cpp(code_s)
  File "/esphome/esphome/writer.py", line 342, in write_cpp
    copy_src_tree()
  File "/esphome/esphome/writer.py", line 295, in copy_src_tree
    copy_files()
  File "/esphome/esphome/components/esp32/__init__.py", line 610, in copy_files
    get_idf_partition_csv(
  File "/esphome/esphome/components/esp32/__init__.py", line 550, in get_idf_partition_csv
    app_partition_size = APP_PARTITION_SIZES[flash_size]
                         ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
KeyError: '2MB'
```

### Additional information

Need to find a way to allow custom partition configurations like we used to have. Also, the auto-partition configuration needs to include support for 2MB ESP-32 boards, like this one:
https://www.waveshare.com/esp-c3-32s-kit.htm

kbx81, 2023-11-16:
Ah, sorry about that -- I didn't realize there were non-8266/ESP32 devices out there that included a flash size less than 4MB. We can easily add a 2MB option to the code and I'll see what I can do so you can still use a custom partition table if you prefer (although, I must admit, I'm curious about the use case for this). 😄

kbx81, 2023-11-16:
Wait -- were you using the custom partition table because we were assuming all devices were 4MB before? 😅

illigtr, 2023-11-16:
Thanks for replying…

A custom partition is handy if one wants to reserve an exact amount of
memory for SPIFFS and/or LittleFS, or potentially remove OTA and use direct
programming of devkit due to excessive size of program…

Hope this helps.

T

*From:* Keith Burzinski ***@***.***
*Sent:* November 16, 2023 15:52
*To:* esphome/issues ***@***.***>
*Cc:* illigtr ***@***.***>; Author ***@***.***>
*Subject:* Re: [esphome/issues] Can no longer use custom partitions table
due to the recent fix " Generate partitions.csv based on flash size #5697 "
(Issue #5121)

Ah, sorry about that -- I didn't realize there were non-8266/ESP32 devices
out there that included a flash size less than 4MB. We can easily add a 2MB
option to the code and I'll see what I can do so you can still use a custom
partition table if you prefer (although, I must admit, I'm curious about
the use case for this). 😄

—
<https://github.com/esphome/issues/issues/5121#issuecomment-1815294121>, or
unsubscribe

illigtr, 2023-11-16:
Correct! There was no support for a 2MB FLASH esp32 device.

*From:* Keith Burzinski ***@***.***
*Sent:* November 16, 2023 15:57
*To:* esphome/issues ***@***.***>
*Cc:* illigtr ***@***.***>; Author ***@***.***>
*Subject:* Re: [esphome/issues] Can no longer use custom partitions table
due to the recent fix " Generate partitions.csv based on flash size #5697 "
(Issue #5121)

Wait -- were you using the custom partition table because we were assuming
all devices were 4MB before? 😅

—
<https://github.com/esphome/issues/issues/5121#issuecomment-1815299932>, or
unsubscribe

kbx81, 2023-11-16:
https://github.com/esphome/esphome/pull/5779 will allow you to specify `2MB` for your device. You can test this with:

```yaml
external_components:
  - source: github://pr#5779
    components: [ esp32 ]
```

illigtr, 2023-11-16:
> esphome/esphome#5779 <https://github.com/esphome/esphome/pull/5779> will

Will try it...thx

illigtr, 2023-11-17:
My first attempt failed after changing the external components per your
instructions:

-----------------------------------------------------------------------------------------------------

INFO ESPHome 2023.11.1

INFO Reading configuration /config/esphome/baseboard.yaml...

Failed config

esp32: None

  Please specify flash_size within esp32 configuration only.

  board: esp32-c3-devkitm-1

  framework:

    version: 4.4.5

    sdkconfig_options:

      CONFIG_COMPILER_OPTIMIZATION_SIZE: y

    advanced:

      ignore_efuse_mac_crc: False

    components: []

    source: ~3.40405.0

    platform_version: ***@***.***

    type: esp-idf

  flash_size: 2MB

  variant: ESP32C3

-----------------------------------------------------------------------------------------------------

So I changed my configuration as follows: I added flash_size and variant to
the esp32: section.

I remarked out the board_upload.flash_size in the platformio options.

-----------------------------------------------------------------------------------------------------

esphome:

  name: "baseboard"

  friendly_name: Solarium

  platformio_options:

#    board_upload.flash_size: 2MB

    board_upload.maximum_size: 2097152

    board_build.partitions: "/config/esphome/minimal.csv"

    board_build.flash_mode: dio

esp32:

  board: esp32-c3-devkitm-1

  framework:

    type: esp-idf

    version: recommended

    # Custom sdkconfig options

    sdkconfig_options:

      CONFIG_COMPILER_OPTIMIZATION_SIZE: y

    # Advanced tweaking options

    advanced:

      ignore_efuse_mac_crc: false

  flash_size: 2MB

  variant: ESP32C3

external_components:

  - source: github://pr#5779

    components: [esp32]

#  - source:

#      type: local

#      path: my_components

-----------------------------------------------------------------------------------------------------

Success!

-----------------------------------------------------------------------------------------------------

Processing baseboard (board: esp32-c3-devkitm-1; framework: espidf;
platform: ***@***.***)

--------------------------------------------------------------------------------

Library Manager: Installing esphome/noise-c @ 0.1.4

INFO Installing esphome/noise-c @ 0.1.4

Unpacking  [####################################]  100%

Library Manager: ***@***.*** has been installed!

INFO ***@***.*** has been installed!

Library Manager: Resolving dependencies...

INFO Resolving dependencies...

Library Manager: Installing esphome/libsodium @ 1.10018.1

INFO Installing esphome/libsodium @ 1.10018.1

Unpacking  [####################################]  100%

Library Manager: ***@***.*** has been installed!

INFO ***@***.*** has been installed!

HARDWARE: ESP32C3 160MHz, 320KB RAM, 2MB Flash

- framework-espidf @ 3.40405.230623 (4.4.5)

 - tool-cmake @ 3.16.9

 - tool-ninja @ 1.10.2

 - toolchain-esp32ulp @ 2.35.0-20220830

 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5

Reading CMake configuration...

Generating assembly for certificate bundle...

*From:* Keith Burzinski ***@***.***
*Sent:* November 16, 2023 16:34
*To:* esphome/issues ***@***.***>
*Cc:* illigtr ***@***.***>; Author ***@***.***>
*Subject:* Re: [esphome/issues] Can no longer use custom partitions table
due to the recent fix " Generate partitions.csv based on flash size #5697 "
(Issue #5121)

esphome/esphome#5779 <https://github.com/esphome/esphome/pull/5779> will
allow you to specify 2MB for your device. You can test this with:

external_components:

  - source: github://pr#5779

    components: [ esp32 ]

—
<https://github.com/esphome/issues/issues/5121#issuecomment-1815349771>, or
unsubscribe

kbx81, 2023-11-18:
> My first attempt failed after changing the external components per your instructions:

Could you please elaborate? What exactly did/did not happen? How did you change your device's configuration/yaml?

kbx81, 2023-11-18:
So it worked -- sorry, the formatting of your message is weird, probably because you're using email to correspond. In any event, I'm glad it is working for you now. 😄

kbx81, 2023-11-18:
Note that there is also a new `partitions` key you can add within `esp32:` to specify the file instead of doing so via `platformio_options`.

closed 2023-11-18 by jesserockz