esphome/issues #5123: Become unavailable every day at the same time time all esp devices

AnnoniFaustoOutlook, 2023-11-16:
### The problem

Hi , i have 15 devices esop-01s connected to esphome in ha ( various functionality : bme280 sensor, thermostat ; swithc light)
every day at the same time (8:10:xx cet winter time, 9:10:xx cet summer time) . in summer there is minor problem due to thermosta is not in use, in winter day's when some esp are unavailable the temperature go immadiatly to unavailable and the average temperature of home go down . at this time thermostat ster warm and alexa say in all speaker that the waming machine si started and after few second stopped  (esp send correct temperature).

2023-11-16 08:10:46.829 WARNING (MainThread) [aioesphomeapi.connection] climabagno @ 192.168.1.18: Connection error occurred: climabagno @ 192.168.1.18: EOF received
2023-11-16 08:10:50.664 WARNING (MainThread) [aioesphomeapi.connection] climacorridoio @ 192.168.1.10: Connection error occurred: [Errno 104] Connection reset by peer
2023-11-16 08:10:50.676 WARNING (MainThread) [aioesphomeapi.connection] climastudio @ 192.168.1.24: Connection error occurred: climastudio @ 192.168.1.24: EOF received
2023-11-16 08:10:51.066 WARNING (MainThread) [aioesphomeapi.connection] climacamera @ 192.168.1.17: Connection error occurred: [Errno 104] Connection reset by peer
2023-11-16 08:10:51.259 WARNING (MainThread) [aioesphomeapi.connection] lavandino @ 192.168.1.16: Connection error occurred: lavandino @ 192.168.1.16: EOF received
2023-11-16 08:10:51.273 WARNING (MainThread) [aioesphomeapi.connection] climacucina @ 192.168.1.22: Connection error occurred: climacucina @ 192.168.1.22: EOF received
2023-11-16 08:10:53.705 WARNING (MainThread) [aioesphomeapi.connection] climabagnetto @ 192.168.1.23: Connection error occurred: climabagnetto @ 192.168.1.23: EOF received
2023-11-16 08:10:57.806 WARNING (MainThread) [aioesphomeapi.connection] comodinofausto @ 192.168.1.13: Connection error occurred: [Errno 104] Connection reset by peer
2023-11-16 08:10:58.056 WARNING (MainThread) [aioesphomeapi.connection] comodinosimona @ 192.168.1.14: Connection error occurred: comodinosimona @ 192.168.1.14: EOF received
2023-11-16 08:10:58.109 WARNING (MainThread) [aioesphomeapi.connection] climabcucina @ 192.168.1.19: Connection error occurred: climabcucina @ 192.168.1.19: EOF received
2023-11-16 08:10:58.129 WARNING (MainThread) [aioesphomeapi.connection] climabsala @ 192.168.1.20: Connection error occurred: climabsala @ 192.168.1.20: EOF received
2023-11-16 08:10:58.583 WARNING (MainThread) [aioesphomeapi.connection] pensile @ 192.168.1.15: Connection error occurred: pensile @ 192.168.1.15: EOF received
2023-11-16 08:11:38.949 WARNING (MainThread) [aioesphomeapi.connection] climasala @ 192.168.1.21: Connection error occurred: [Errno 104] Connection reset by peer
2023-11-16 08:24:55.617 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised when updating state of sensor.gn191gm_vehicleinfo_tripsinfo_trips_array_1_totaldistance_value, topic: 'homeassistant/sensor/ZFAEFAA24NX126703_car_vehicleInfo_tripsInfo_trips_array_1_totalDistance_value/state' with payload: b'null'

and

2023-11-16 08:10:46.829 WARNING (MainThread) [aioesphomeapi.connection] climabagno @ 192.168.1.18: Connection error occurred: climabagno @ 192.168.1.18: EOF received
2023-11-16 08:10:50.664 WARNING (MainThread) [aioesphomeapi.connection] climacorridoio @ 192.168.1.10: Connection error occurred: [Errno 104] Connection reset by peer
2023-11-16 08:10:50.676 WARNING (MainThread) [aioesphomeapi.connection] climastudio @ 192.168.1.24: Connection error occurred: climastudio @ 192.168.1.24: EOF received
2023-11-16 08:10:51.066 WARNING (MainThread) [aioesphomeapi.connection] climacamera @ 192.168.1.17: Connection error occurred: [Errno 104] Connection reset by peer
2023-11-16 08:10:51.259 WARNING (MainThread) [aioesphomeapi.connection] lavandino @ 192.168.1.16: Connection error occurred: lavandino @ 192.168.1.16: EOF received
2023-11-16 08:10:51.273 WARNING (MainThread) [aioesphomeapi.connection] climacucina @ 192.168.1.22: Connection error occurred: climacucina @ 192.168.1.22: EOF received
2023-11-16 08:10:53.705 WARNING (MainThread) [aioesphomeapi.connection] climabagnetto @ 192.168.1.23: Connection error occurred: climabagnetto @ 192.168.1.23: EOF received
2023-11-16 08:10:57.806 WARNING (MainThread) [aioesphomeapi.connection] comodinofausto @ 192.168.1.13: Connection error occurred: [Errno 104] Connection reset by peer
2023-11-16 08:10:58.056 WARNING (MainThread) [aioesphomeapi.connection] comodinosimona @ 192.168.1.14: Connection error occurred: comodinosimona @ 192.168.1.14: EOF received
2023-11-16 08:10:58.109 WARNING (MainThread) [aioesphomeapi.connection] climabcucina @ 192.168.1.19: Connection error occurred: climabcucina @ 192.168.1.19: EOF received
2023-11-16 08:10:58.129 WARNING (MainThread) [aioesphomeapi.connection] climabsala @ 192.168.1.20: Connection error occurred: climabsala @ 192.168.1.20: EOF received
2023-11-16 08:10:58.583 WARNING (MainThread) [aioesphomeapi.connection] pensile @ 192.168.1.15: Connection error occurred: pensile @ 192.168.1.15: EOF received
2023-11-16 08:11:38.949 WARNING (MainThread) [aioesphomeapi.connection] climasala @ 192.168.1.21: Connection error occurred: [Errno 104] Connection reset by peer
2023-11-16 08:24:55.617 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised when updating state of sensor.gn191gm_vehicleinfo_tripsinfo_trips_array_1_totaldistance_value, topic: 'homeassistant/sensor/ZFAEFAA24NX126703_car_vehicleInfo_tripsInfo_trips_array_1_totalDistance_value/state' with payload: b'null'

i have three zone with 3 different wifi and three group of esp , all with static ip address , wich is the common problem for  this situation at the same time everyday?

Thank you
Annoni Fausto
Italy

### Which version of ESPHome has the issue?

2023-11-0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023-11-0

### What platform are you using?

ESP8266

### Board

esp-01s

### Component causing the issue

api

### Example YAML snippet

```yaml
[... 200 lines cut ...]
2023-11-16 08:10:57.806 WARNING (MainThread) [aioesphomeapi.connection] comodinofausto @ 192.168.1.13: Connection error occurred: [Errno 104] Connection reset by peer
2023-11-16 08:10:58.056 WARNING (MainThread) [aioesphomeapi.connection] comodinosimona @ 192.168.1.14: Connection error occurred: comodinosimona @ 192.168.1.14: EOF received
2023-11-16 08:10:58.109 WARNING (MainThread) [aioesphomeapi.connection] climabcucina @ 192.168.1.19: Connection error occurred: climabcucina @ 192.168.1.19: EOF received
2023-11-16 08:10:58.129 WARNING (MainThread) [aioesphomeapi.connection] climabsala @ 192.168.1.20: Connection error occurred: climabsala @ 192.168.1.20: EOF received
2023-11-16 08:10:58.583 WARNING (MainThread) [aioesphomeapi.connection] pensile @ 192.168.1.15: Connection error occurred: pensile @ 192.168.1.15: EOF received
2023-11-16 08:11:38.949 WARNING (MainThread) [aioesphomeapi.connection] climasala @ 192.168.1.21: Connection error occurred: [Errno 104] Connection reset by peer
2023-11-16 08:24:55.617 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised when updating state of sensor.gn191gm_vehicleinfo_tripsinfo_trips_array_1_totaldistance_value, topic: 'homeassistant/sensor/ZFAEFAA24NX126703_car_vehicleInfo_tripsInfo_trips_array_1_totalDistance_value/state' with payload: b'null'

and

2023-11-16 08:10:46.829 WARNING (MainThread) [aioesphomeapi.connection] climabagno @ 192.168.1.18: Connection error occurred: climabagno @ 192.168.1.18: EOF received
2023-11-16 08:10:50.664 WARNING (MainThread) [aioesphomeapi.connection] climacorridoio @ 192.168.1.10: Connection error occurred: [Errno 104] Connection reset by peer
2023-11-16 08:10:50.676 WARNING (MainThread) [aioesphomeapi.connection] climastudio @ 192.168.1.24: Connection error occurred: climastudio @ 192.168.1.24: EOF received
2023-11-16 08:10:51.066 WARNING (MainThread) [aioesphomeapi.connection] climacamera @ 192.168.1.17: Connection error occurred: [Errno 104] Connection reset by peer
2023-11-16 08:10:51.259 WARNING (MainThread) [aioesphomeapi.connection] lavandino @ 192.168.1.16: Connection error occurred: lavandino @ 192.168.1.16: EOF received
2023-11-16 08:10:51.273 WARNING (MainThread) [aioesphomeapi.connection] climacucina @ 192.168.1.22: Connection error occurred: climacucina @ 192.168.1.22: EOF received
[... cut ...]
2023-11-16 08:10:58.583 WARNING (MainThread) [aioesphomeapi.connection] pensile @ 192.168.1.15: Connection error occurred: pensile @ 192.168.1.15: EOF received
2023-11-16 08:11:38.949 WARNING (MainThread) [aioesphomeapi.connection] climasala @ 192.168.1.21: Connection error occurred: [Errno 104] Connection reset by peer
2023-11-16 08:24:55.617 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised when updating state of sensor.gn191gm_vehicleinfo_tripsinfo_trips_array_1_totaldistance_value, topic: 'homeassistant/sensor/ZFAEFAA24NX126703_car_vehicleInfo_tripsInfo_trips_array_1_totalDistance_value/state' with payload: b'null'
```

ssieb, 2023-11-16:
If it happens to all of them at the same time, that strongly suggests that it's a problem with your wifi or network.

AnnoniFaustoOutlook, 2023-11-16:
Hi thanks , but I have three separated sites with 3 different access point (brand different and said  different) how many possibility that all In different sites become  unavailable at the same time? For me the problem is in the home assistant / esphome , or a function not created by me that run at the same time from more and more day’s and firmware version. Every day winter at same time and summer at the same time.( 9:10 am,8:10 am) I don’t have automation at. This time.

closed 2024-03-31 by github-actions[bot]