esphome/issues #5124: Build dependencies failure after recent update for armv7l

Tonymac32, 2023-11-17:
### The problem

No builds succeed, I haven't tried to step back in time to see where the break first occurred.  Odroid XU4 host.

### Which version of ESPHome has the issue?

2023.11.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

all

### Component causing the issue

Build dependencies

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.11.1
INFO Reading configuration /config/esphome/cat-feeder.yaml...
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing cat-feeder (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=False to SConscript calls.
Missing SConscript '/data/cache/platformio/packages/framework-arduinoespressif32/tools/platformio-build.py'
File "/data/cache/platformio/platforms/espressif32/builder/frameworks/arduino.py", line 41, in <module>
Library Manager: Installing ESPmDNS
Warning! Could not find the package with 'ESPmDNS' requirements for your system 'linux_armv7l'
Library Manager: Installing Update
Warning! Could not find the package with 'Update' requirements for your system 'linux_armv7l'
Dependency Graph
|-- WiFi @ 1.2.7
|-- noise-c @ 0.1.4
Compiling .pioenvs/cat-feeder/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/cat-feeder/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/cat-feeder/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/cat-feeder/src/esphome/components/api/api_pb2_service.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/cat-feeder/src/esphome/components/api/api_server.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/cat-feeder/src/esphome/components/api/list_entities.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/cat-feeder/src/esphome/components/api/proto.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
*** [.pioenvs/cat-feeder/src/esphome/components/api/api_connection.cpp.o] Error 127
*** [.pioenvs/cat-feeder/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [.pioenvs/cat-feeder/src/esphome/components/api/api_pb2.cpp.o] Error 127
*** [.pioenvs/cat-feeder/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
*** [.pioenvs/cat-feeder/src/esphome/components/api/api_server.cpp.o] Error 127
*** [.pioenvs/cat-feeder/src/esphome/components/api/list_entities.cpp.o] Error 127
Compiling .pioenvs/cat-feeder/src/esphome/components/api/subscribe_state.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
*** [.pioenvs/cat-feeder/src/esphome/components/api/proto.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [.pioenvs/cat-feeder/src/esphome/components/api/subscribe_state.cpp.o] Error 127
========================== [FAILED] Took 3.71 seconds ==========================
```

roofburner, 2023-11-17:
Same issue here. I use docker run to upgrade my esp board.

I followed the guide below the error, but no luck. Reboot did not help either.

```
INFO ESPHome 2023.11.1
INFO Reading configuration ./relay.yml...
INFO Generating C++ source...
INFO Compiling app...
Processing heatingrelais (board: esp12e; framework: arduino; platform: platformio/espressif8266@3.2.0)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
RuntimeError: Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/platformio/__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "/usr/local/lib/python3.11/dist-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/run/cli.py", line 145, in cli
    process_env(
  File "/usr/local/lib/python3.11/dist-packages/platformio/run/cli.py", line 208, in process_env
    ).process()
      ^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/run/processor.py", line 90, in process
    ).run(build_vars, build_targets, self.silent, self.verbose, self.jobs)
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/platform/_run.py", line 64, in run
    result = self._run_scons(variables, targets, jobs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/platform/_run.py", line 117, in _run_scons
    stdout=proc.BuildAsyncPipe(
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/proc.py", line 65, in __init__
    super().__init__()
  File "/usr/local/lib/python3.11/dist-packages/platformio/proc.py", line 38, in __init__
    self._thread.start()
  File "/usr/lib/python3.11/threading.py", line 957, in start
    _start_new_thread(self._bootstrap, ())
RuntimeError: can't start new thread

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

jmguitars, 2023-11-17:
The problem

No builds succeed

Which version of ESPHome has the issue?

2023.11.1

What type of installation are you using?

ESP Home assist all methods

Which version of Home Assistant has the issue?

Core
2023.11.2
Supervisor
2023.11.3
Operating System
11.1
Frontend
20231030.2

What platform are you using?

esp8366
nodemcu

Component causing the issue

Build dependencies

Example YAML snippet

INFO ESPHome 2023.11.1
INFO Reading configuration /config/esphome/esphome-web-5017f0.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-5017f0 (board: nodemcu; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
Compiling .pioenvs/esphome-web-5017f0/src/esphome/components/api/api_connection.cpp.o
In file included from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/api_frame_helper.h:2:10: fatal error: cstdint: No such file or directory
    2 | #include <cstdint>
      |          ^~~~~~~~~
compilation terminated.
*** [.pioenvs/esphome-web-5017f0/src/esphome/components/api/api_connection.cpp.o] Error 1
========================== [FAILED] Took 7.25 seconds ==========================

Anything in the logs that might be useful for us?

srusinov, 2023-11-17:
Same issue on XU4:

INFO ESPHome 2023.11.1
INFO Reading configuration /config/esphome/esp32-cam.yaml...
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-cam (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
Compiling .pioenvs/esp32-cam/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/esp32-cam/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/esp32-cam/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/esp32-cam/src/esphome/components/api/api_pb2_service.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/esp32-cam/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/esp32-cam/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/esp32-cam/src/esphome/components/api/proto.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
sh: 1: xtensa-esp32-elf-g++: not found
*** [.pioenvs/esp32-cam/src/esphome/components/api/api_connection.cpp.o] Error 127
*** [.pioenvs/esp32-cam/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [.pioenvs/esp32-cam/src/esphome/components/api/api_pb2.cpp.o] Error 127
*** [.pioenvs/esp32-cam/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
Compiling .pioenvs/esp32-cam/src/esphome/components/api/subscribe_state.cpp.o
*** [.pioenvs/esp32-cam/src/esphome/components/api/list_entities.cpp.o] Error 127
*** [.pioenvs/esp32-cam/src/esphome/components/api/api_server.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [.pioenvs/esp32-cam/src/esphome/components/api/proto.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [.pioenvs/esp32-cam/src/esphome/components/api/subscribe_state.cpp.o] Error 127
========================== [FAILED] Took 8.20 seconds ==========================

Tonymac32, 2023-11-18:
Fixed by https://github.com/esphome/esphome/pull/5781

closed 2023-11-18 by Tonymac32

roofburner, 2023-11-21:
Dear Tonymac32,

I still have the issue I mentioned above.

I try to "compile" ESPHome 2023.11.2 using docker (Ubuntu Server on Dell hardware).

This is what I execute:
```
docker pull esphome/esphome
docker run --rm -v /usr/share/esphome:/config -it esphome/esphome run ./relay.yml
```

Please advise...

Tonymac32, 2023-11-21:
@roofburner looking at your debug info, I'd say your issue is unrelated to the one I described, mine was caused by a broken symlink making the compiler unavailable, your appears to be quite different.  I'd suggest opening a new ticket