esphome/issues #5125: Nullptr topic causes panic with ESP32 when MQTT message is read chunked

aaliddell, 2023-11-17:
### The problem

When a received MQTT message is read in multiple chunks under esp32 MQTT backend, any messages after the first will have a nullptr topic pointer. This then causes a std::logic_error to be raised when creating a std::string for the topic in mqtt_client, leading to a panic and reboot.

Topic is null on subsequent chunks from here: https://github.com/espressif/esp-mqtt/blob/7894dd0ace9ef8b035e3686d2767324540c1cfea/mqtt_client.c#L1090
Nullptr gets propagated here due to empty topic: https://github.com/esphome/esphome/blame/32e3f2623973f45bd37b6b0dc956adcd82a45e77/esphome/components/mqtt/mqtt_backend_esp32.cpp#L146
Nullptr causes std::logic_error here in std::string constructor (implicit cast): https://github.com/esphome/esphome/blob/32e3f2623973f45bd37b6b0dc956adcd82a45e77/esphome/components/mqtt/mqtt_client.cpp#L48

The topic needs to be stored from the first segment and then used once all segments have been read. Will send a PR shortly.

### Which version of ESPHome has the issue?

2023.9.1 but still present in latest

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32-IDF

### Board

esp32dev

### Component causing the issue

mqtt

### Example YAML snippet

```yaml
esphome:
  name: Something

esp32:
  board: esp32dev
  framework:
    type: esp-idf

mqtt:
  id: mqtt_client
  broker: 1.2.3.4
  username: username
  password: password
  on_message:
  # Subscription required to cause bug
    - topic: some/topic
      then:
        - logger.log: "Message"

wifi:
  ssid: ssid
  password: password
```

### Anything in the logs that might be useful for us?

```txt
0x40084236: panic_abort at /config/.esphome/platformio/packages/framework-espidf/components/esp_system/panic.c:408
0x4008c32d: esp_system_abort at /config/.esphome/platformio/packages/framework-espidf/components/esp_system/esp_system.c:137
0x40092306: abort at /config/.esphome/platformio/packages/framework-espidf/components/newlib/abort.c:46
0x40164a0a: __wrap__Unwind_RaiseException at ??:?
0x401636b7: __cxa_throw at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:90
0x40163f5b: std::__throw_logic_error(char const*) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/functexcept.cc:66 (discriminator 2)
0x400d7cfa: void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char const*>(char const*, char const*, std::forward_iterator_tag) at /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.tcc:212
 (inlined by) void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct_aux<char const*>(char const*, char const*, std::__false_type) at /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:236
 (inlined by) void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char const*>(char const*, char const*) at /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:255
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(char const*, std::allocator<char> const&) at /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:516
0x400d8310: std::_Function_handler<void (char const*, char const*, unsigned int, unsigned int, unsigned int), esphome::mqtt::MQTTClientComponent::setup()::{lambda(char const*, char const*, unsigned int, unsigned int, unsigned int)#1}>::_M_invoke(std::_Any_data const&, char const*&&, std::_Any_data const&, unsigned int&&, char const*&&, char const*&&) at /config/.esphome/build/<snip>/src/esphome/components/mqtt/mqtt_client.cpp:48
 (inlined by) _M_invoke at /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:297
0x400d6e31: std::function<void (char const*, char const*, unsigned int, unsigned int, unsigned int)>::operator()(char const*, char const*, unsigned int, unsigned int, unsigned int) const at /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687
 (inlined by) esphome::CallbackManager<void (char const*, char const*, unsigned int, unsigned int, unsigned int)>::call(char const*, char const*, unsigned int, unsigned int, unsigned int) at /config/.esphome/build/<snip>/src/esphome/core/helpers.h:482
 (inlined by) esphome::mqtt::MQTTBackendESP32::mqtt_event_handler_(esphome::mqtt::Event const&) at /config/.esphome/build/<snip>/src/esphome/components/mqtt/mqtt_backend_esp32.cpp:146
0x400d6f87: esphome::mqtt::MQTTBackendESP32::loop() at /config/.esphome/build/<snip>/src/esphome/components/mqtt/mqtt_backend_esp32.cpp:103
 (inlined by) esphome::mqtt::MQTTBackendESP32::loop() at /config/.esphome/build/<snip>/src/esphome/components/mqtt/mqtt_backend_esp32.cpp:98
0x400d8643: esphome::mqtt::MQTTClientComponent::loop() at /config/.esphome/build/<snip>/src/esphome/components/mqtt/mqtt_client.cpp:295
0x4017a169: esphome::Component::call_loop() at /config/.esphome/build/<snip>/src/esphome/core/component.cpp:76
0x4017a255: esphome::Component::call() at /config/.esphome/build/<snip>/src/esphome/core/component.cpp:98
0x400e0218: esphome::Application::loop() at /config/.esphome/build/<snip>/src/esphome/core/application.cpp:74 (discriminator 2)
0x400e40c6: loop() at /config/<snip>.yaml:48
```

aaliddell, 2024-03-17:
.

spuder, 2025-07-06:
I believe I'm running into this same issue.

When connecting to a bambu 3d printer at /device/<serial>/report, there are occasional messages that are about 3.7k bytes.

Those messages always cause a crash.

closed 2025-07-10 by jesserockz