esphome/issues #5126: The `wifi_info` platform randomly returns either the IPv4 or IPv6 address of a node

kquinsland, 2023-11-18:
### The problem

A picture is worth a thousand words.

Basically, I have an instance of `wifi_info` so the IP address of an ESP can be displayed through the home assistant web UI.

When I have `network.enable_ipv6` set to `true`, the device gets two IP addresses. This is intentional; I operate a dual-stack network.

However, the IP address that gets reported ... alternates between V6 and V4 as shown here:

[image]

### Which version of ESPHome has the issue?

2023.11

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Component causing the issue

wifi_info

### Example YAML snippet

```yaml
esphome:
  name: ${hostname}
  friendly_name: ${friendly_name}
  # Will boot loop w/o this when using the ESP-IDF and the C3
  platformio_options:
    board_build.flash_mode: dio

preferences:
  flash_write_interval: 15min

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

# Enable logging
logger:
  # TODO: revert when done testing
  level: DEBUG

  # The ESP32-C3 has built in USB/JTAG support
  # Only with ESP-IDF :/
  hardware_uart: USB_SERIAL_JTAG

# See: https://esphome.io/guides/configuration-types.html#packages
packages:
  mqtt: !include ../packages/mqtt.yaml
  ota: !include ../packages/ota.yaml
  time: !include ../packages/ntp.yaml
  web_server: !include ../packages/web_server.yaml
  wifi: !include ../packages/wifi.yaml
  friendly_uptime: !include ../packages/sensor/friendly_uptime.yaml

# Enable IPv6 support
# Disable for now to prevent spam on the IP address sensor(s)
# See: https://esphome.io/components/network.html
#network:
#  enable_ipv6: true

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "IPv4 Address"
      entity_category: "diagnostic"
      icon: "mdi:ip-network"
```

### Anything in the logs that might be useful for us?

```txt
no, sadly
```

HeMan, 2024-03-04:
Could you test with recent dev-branch? Or would you rather wait until it's released?

kquinsland, 2024-03-04:
> Could you test with recent dev-branch? Or would you rather wait until it's released?

release makes things easy but i can clone esphome/dev branch and use that to build a test firmware sometime today.

kquinsland, 2024-03-04:
Sorry for the "stream of conscious" vibe, using this input to draft notes as I do the test in real time:

I started with the latest stable ESPHome release (`v2024.2.1`)  on this device:

```yaml
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
```

This is using WiFI and MQTT and I could confirm that as soon as I enabled IPv6 support, the `wifi_info` component would flip/flop between the IPv4 and V6 address.

Then after building using `dev` branch - `v2024.3.0-dev`- I don't appear to have a flip-flop issues.
I tracked down [your changes](https://github.com/esphome/esphome-docs/pull/3234/files#diff-9590086b44623edf1b09d2b5ae3f3e80e8e3b2ac5a946c137e47bccca5e4baa8) to the `{wifi,eth}_info` sensors and updated the package to:

```yaml
- platform: wifi_info
  ip_address:
    name: ESP IP Address
    address_0:
      name: "IP Address 0"
    address_1:
      name: "IP Address 1"
    address_2:
      name: "IP Address 2"
    address_3:
      name: "IP Address 3"
```

And got this result:
[image]

Where the value of `IP Address 3` is blank/unknown (which is to be expected in my network)

closed 2024-03-21 by kquinsland