esphome/issues #5127: voice_assistant restarts every 5 seconds and does not respond to wake word

joshuaboniface, 2023-11-18:
### The problem

I'm running what seems like a fairly common INMP441-based wake-word enabled `voice_assistant` setup.

Every 5 seconds without fail, the voice_assistant component restarts, I presume because it's ending a Wake Word cycle.

The problem is, every time it does this, the assistant is unresponsive to the wake word for 0.5-1 seconds after the assistant restarts. This happens basically as soon as the assist pipeline restarts, and persists for a little bit after the last `Event Type: 9` is posted. It may seem like an insignificantly short time, but with a 5s restart interval that means that close to 10-20% of the time I try to wake it, on average, it ignores me, and I've really noticed this in trying to use it.

I'm not sure if the bug is the fact that the voice_assistant is restarting so often, and if this is somehow configurable, or if the actual bug is deeper and is related to the fact the mic isn't listening to the wake word for some time after the restart.

For what it's worth, it's also really slow to start listening to the wake word again after a response or on boot-up, by about the same 0.5-1s. I can ignore it there but I can't really ignore it when it happens all the time.

### Which version of ESPHome has the issue?

2023.11.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2023.11.2

### What platform are you using?

ESP32-IDF

### Board

esp32dev

### Component causing the issue

voice_assistant

### Example YAML snippet

```yaml
i2s_audio:
  i2s_lrclk_pin: GPIO25
  i2s_bclk_pin: GPIO26

microphone:
  - platform: i2s_audio
    id: mic
    adc_type: external
    i2s_din_pin: GPIO27
    pdm: false

voice_assistant:
  microphone: mic 
  use_wake_word: false
  noise_suppression_level: 2
  auto_gain: 31dBFS
  volume_multiplier: 3.0 
  id: assist
  on_wake_word_detected:
  on_listening:
  on_stt_end:
  on_tts_start:
```

Do note that all the `on_` entries there have things in them related to my feedback system, but that doesn't seem relevant to this issue. Also note that these are indeed the only ones I have defined.
```

### Anything in the logs that might be useful for us?

```txt
[00:54:30][D][voice_assistant:502]: Event Type: 0   
[00:54:30][D][voice_assistant:502]: Event Type: 2                                                                      
[00:54:30][D][voice_assistant:589]: Assist Pipeline ended                                                              
[00:54:30][D][voice_assistant:395]: State changed from STREAMING_MICROPHONE to IDLE
[00:54:30][D][voice_assistant:401]: Desired state set to IDLE                                         
[00:54:30][D][voice_assistant:395]: State changed from IDLE to START_PIPELINE   
[00:54:30][D][voice_assistant:401]: Desired state set to START_MICROPHONE                                              
[00:54:30][D][voice_assistant:206]: Requesting start...
[00:54:30][D][voice_assistant:395]: State changed from START_PIPELINE to STARTING_PIPELINE                             
[00:54:30][D][voice_assistant:416]: Client started, streaming microphone                                               
[00:54:30][D][voice_assistant:395]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[00:54:30][D][voice_assistant:401]: Desired state set to STREAMING_MICROPHONE
[00:54:30][D][voice_assistant:502]: Event Type: 1                                                                      
[00:54:30][D][voice_assistant:505]: Assist Pipeline running                                                            
[00:54:30][D][voice_assistant:502]: Event Type: 9     
[00:54:35][D][voice_assistant:502]: Event Type: 0
[00:54:35][D][voice_assistant:502]: Event Type: 2
[00:54:35][D][voice_assistant:589]: Assist Pipeline ended
[00:54:35][D][voice_assistant:395]: State changed from STREAMING_MICROPHONE to IDLE
[00:54:35][D][voice_assistant:401]: Desired state set to IDLE
[00:54:35][D][voice_assistant:395]: State changed from IDLE to START_PIPELINE
[00:54:35][D][voice_assistant:401]: Desired state set to START_MICROPHONE
[00:54:35][D][voice_assistant:206]: Requesting start...
[00:54:35][D][voice_assistant:395]: State changed from START_PIPELINE to STARTING_PIPELINE
[00:54:35][D][voice_assistant:416]: Client started, streaming microphone
[00:54:35][D][voice_assistant:395]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[00:54:35][D][voice_assistant:401]: Desired state set to STREAMING_MICROPHONE
[00:54:35][D][voice_assistant:502]: Event Type: 1
[00:54:35][D][voice_assistant:505]: Assist Pipeline running
[00:54:35][D][voice_assistant:502]: Event Type: 9
```

Log of two restart events. The failure to listen to the wake word persists after the final `Event Type: 9` by 0.5-1s each time.
```

joshuaboniface, 2023-11-18:
Trying to dig in a bit to the codebase, I've become fairly suspicious that this is actually coming from the HomeAssistant side rather than ESPHome. But the wake word timeout there seems to be 3 seconds by default, not 5 seconds, so I'm a bit puzzled where it's getting that 5s value.

andrejsoucek, 2023-11-21:
@joshuaboniface any progress? I'm facing exactly the same issue on 
```
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
```

joshuaboniface, 2023-11-21:
Not particularly. Still been digging into the code trying to figure it out (I'm no C++ programmer by any means), but I've turned up nothing much except that *maybe* an ADF-supported board would help.

I'm also suspicious that my issue might not be code related, but might instead be due to poor WiFi connectivity. That might be something worth checking on your side too. But I'm not entirely sure how to debug that with so many moving parts.

andrejsoucek, 2023-11-21:
My microphone probably doesn't work at all, IDK why. Are you using SPH0645 by any chance? I will buy INMP441 and try again I guess because I can't make it work even through Arduino IDE and plot monitor. :/ There's this problem with nonstandard I2S interface, described here https://www.esp32.com/viewtopic.php?t=35402

joshuaboniface, 2023-11-21:
No I'm using an INMP441. The mic seems to work flawlessly otherwise so I don't think that has anytihng to do with this.

hanfreakingsolo, 2023-11-22:
+1 for this issue. I'm using a SPH0645 as I could not get 3 different INMP441's to work.
The 5 second restart has been very frustrating, and on my end I can confirm there is no difference even with a perfect wifi signal

andrejsoucek, 2023-11-22:
@hanfreakingsolo wdym you could not get 3 different INMP441s to work? how did you know they are not working? and how do you know SPH0645 is working? were you able to test it somehow? are you sure it can hear you? how? I would like to do the same test

hanfreakingsolo, 2023-11-24:
@andrejsoucek yeah I should clarify that a bit. I tried 3 to make absolutely sure I didn't fry the mic at some point, so I'm sure it's not a hardware issue. I tried multiple configurations in esphome and wiring left and right channel, without luck. Troubleshooting is the really difficult part because from what I can tell there is no easy way to hear what home assistant hears (maybe someone in this thread can help?). So I don't know if the mic is sending nothing at all, or if the signal is just too noisy/quiet/loud.
One thing I learned though with my SPH0645, is that the volume multiplier is extremely important - set it to 3, won't work at all. Set it to 1, also no luck. It has to be at 2 in my config for openwakeword to recognize anything at all. That could be something I had wrong with the INMP441 - I didn't try different values, never thought it would be that sensitive

joshuaboniface, 2023-11-26:
>there is no easy way to hear what home assistant hears (maybe someone in this thread can help?)

You can do this, add:

```
assist_pipeline:
    debug_recording_dir: /share/assist_pipeline
```

to your `configuration.yaml`, and you'll get recordings of the wake word activations in that dir. If you get no sound, it's definitely not recording. I ended up doing this to debug my mics at first, as I had some wiring issues.

One thing I'd suggest is to try using different connectors. I was using shoddy wires and a cheap breadboard at first, but using better wires caused it to suddenly start working again.

That said, all of this discussion is unrelated to the original issue I posted, which is purely about the backend ESPHome handling of the pipeline. To avoid this getting too far off-topic I'd suggest you ask for additional help in the main HomeAssistant forums.

sparkydave1981, 2023-11-30:
I'm having the same issue too which posted about on the HA forum [here](https://community.home-assistant.io/t/year-of-the-voice-chapter-4-wake-words/625584/262?u=sparkydave).  I got a very complicated response that perhaps someone will understand if you want to go read it.  

> You can do this, add:
>
> ```
> assist_pipeline:
>     debug_recording_dir: /share/assist_pipeline
> ```
>
> to your `configuration.yaml`, and you'll get recordings of the wake word activations in that dir. If [...]

I'll give this a try.

> One thing I'd suggest is to try using different connectors. I was using shoddy wires and a cheap breadboard at first, but using better wires caused it to suddenly start working again.

I definitely don't have this issue as everything is soldered.

> about the backend ESPHome handling of the pipeline.

I'm guessing the 'assist pipeline ended' should never happen since the ESP needs to relay all sound for analysis by the handler on HA...?

sparkydave1981, 2023-11-30:
I also just discovered that my ESP seems to give up completely on the voice monitoring / pipeline connection after a while.  I have to power cycle to get it going again.  Once I did that I finally started seeing recordings in the share directory as mentioned above, however it's getting spammed with 5 second audio files that so far all seem to be silent...

joshuaboniface, 2023-11-30:
Folks, none of you are reporting the same issue I am. Please read more than just the title.

My wake word detection is *just fine*, most of the time. My issue is *entirely* about *the small gap* in detection after each wake word cycle completes. If you're not getting ANY wake word detection, that is not this issue and please make your own. This issue is complicated enough to explain correctly without it being sidetracked; please respect that.

joshuaboniface, 2023-11-30:
To add substantiatively to the original issue, this has now persisted on:

ESP-IDF and Arduino firmwares
ESPHome versions 2023.11.1 through 2023.11.6
HomeAssistant versions 2023.11.2 through 2023.11.3
With or without various other sensors or configurations of ESPHome present.

I have added a video demonstration of the issue to the bottom of the original post in the hopes that it might add some clarity to my description.

I did substantially more digging into the workings of the wake words and the voice assist pipelines in HASS more generally, and have still found nothing obvious as to why this works the way it does. I can offer no good explanation.

Subjectively, to me, the issue is fairly clear: there is a cycle where the `voice_assist` goes through the various stages of IDLE to START_PIPELINE to STARTING_PIPELINE to STREAMING_MICROPHONE, and the total time this takes, plus whatever extra time is needed for the microphone stream to actually reach homeassistant, is too long to avoid missing wake words at random times given the very short period involved (5 seconds).

I can see at least 3 conceptual solutions:

1. Increase the wake word detection time to something like 60s. It would be far less likely to be missed for 0.5 out of every 60 seconds (1/120th of the time) versus for 0.5 of every 5 seconds (1/10th of the time). 60s of WAV audio, even at the maximum possible PCM rates (which we are not using) would only be ~35MB which should be fine.
2. Adjust when the microphone stops and starts. Honestly, the fact that there's any noticeable time when the microphone is not listening seems to me to be the underlying "bug" here. The 4-stage cycle simply seems to be far too long and indirect.
3. Buffer the incoming audio somehow. I don't think this is really possible given the ESP32's limited RAM, but I figured I'd mention it for completeness sake.

Now obviously all that said, I'm not a C++ expert and I do not understand this code beyond a surface level. I'm simply providing these ideas as a jumping-off point for further discussion. I do hope this issue receives some attention soon as it does make using wake words quite cumbersome.

sparkydave1981, 2023-11-30:
> Folks, none of you are reporting the same issue I am. Please read more than just the title, which I have now adjusted to make it clear that "does not respond to wake word" is NOT the operative part of my issue.

So you don't think the fact that the logs are constantly switching from `Assist Pipeline running` to `Assist Pipeline ended` is an issue?  Sorry for my ignorance but I would have thought it should just stay 'running'.  I take it from what you are saying, the 5 second clips are created by that cycle of 'running', ended' and this is normal.

andrejsoucek, 2023-12-01:
@sparkydave1981 it is fine, it works like that

joshuaboniface, 2023-12-08:
@andrejsoucek You know I completely forgot to do that, despite actually telling people to do it earlier in the thread.

Ignoring for a moment how much of a PITA it is to piece together these 5 second chunks, I was able to replicate the issue but listening back to the audio, there is a clear recording of me saying the wakeword twice, but only the second one was picked up.

This just has me even more confused now, as I'm not sure where to add further debugging to this to see what the HASS side is actually doing. But I guess I know it definitely isn't an ESPHome issue.

closed 2023-12-08 by joshuaboniface