esphome/issues #5129: Can't compile esphome project

szymucha94, 2023-11-18:
### The problem

I'm getting a segfault during update of esphome project.
`--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/data/build/lr-bedside-lamp/.pioenvs/lr-bedside-lamp/CMakeFiles/CMakeOutput.log".
See also "/data/build/lr-bedside-lamp/.pioenvs/lr-bedside-lamp/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /data/build/lr-bedside-lamp/.pioenvs/lr-bedside-lamp/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_34238 && [1/2] Building C object CMakeFiles/cmTC_34238.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_34238
    FAILED: cmTC_34238 
    : && /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_34238.dir/testCCompiler.c.obj  -o cmTC_34238   && :
    xtensa-esp32-elf-gcc: internal compiler error: Segmentation fault signal terminated program collect2
    Please submit a full bug report,
    with preprocessed source if appropriate.
    See <https://gcc.gnu.org/bugs/> for instructions.
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:298 (__project)
  CMakeLists.txt:3 (project)`

### Which version of ESPHome has the issue?

2023.11.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.2

### What platform are you using?

ESP32-IDF

### Example YAML snippet

```yaml
# --------------------------------------------------------------------------
# Substitutions
#
# These are substitutions as used by the configuration packages from below.
# You can uncomment and update the ones that you want to modify.
# --------------------------------------------------------------------------

substitutions:
  name: lr-bedside-lamp
  friendly_name: LR Bedside Lamp
  #light_name: ${friendly_name} RGBWW Light
  #light_mode_text_sensor_name: ${friendly_name} Light Mode
  #default_transition_length: 800ms

# --------------------------------------------------------------------------
# Load configuration packages
#
# These provide a convenient way to compose your device configuration from
# some functional building blocks. Pick and mix the blocks that you need.
#
# For customization you can override options in your config or you can
# copy the contents of these packages directly in your config file as
# an example for your own customizations.
#
# Available packages are:
# - core.yaml                : core components & hardware setup
# - behavior_default.yaml    : default device behavior
# - ota_feedback.yaml        : enable visual feedback during OTA updates
# - activate_preset_svc.yaml : 'activate_preset' service for Home Assistant
# --------------------------------------------------------------------------

packages:
  bslamp2:
    url: https://github.com/mmakaay/esphome-xiaomi_bslamp2
    ref: dev
    files:
      - packages/core.yaml
      - packages/behavior_default.yaml
      - packages/ota_feedback.yaml
      - packages/activate_preset_svc.yaml
    refresh: 0s

# --------------------------------------------------------------------------
# Use your own preferences for these components.
# --------------------------------------------------------------------------

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

api:
  encryption:
    key: "hidden"

ota:
  password: "hidden"
  
logger:
  level: NONE

sensor:
  - platform: internal_temperature
    name: "Internal Temperature"
```

### Anything in the logs that might be useful for us?

```txt
This happened (not sure if directly) after server upgrade involving change from single core allocated to HAOS VM to dual core. There were multiple HA and esphome upgrades since server change.
```

jsuanet, 2023-11-18:
Did you try to clean the build files before you started the compilation? I didn't have any problems to build the bslamp2 with Esphome 2023.11.2 (at this moment the latest version).

szymucha94, 2023-11-18:
@jsuanet yes, I'm cleaning build files before every attempt to compile.
```
INFO ESPHome 2023.11.2
INFO Reading configuration /config/esphome/lr-bedside-lamp.yaml...
INFO Updating https://github.com/mmakaay/esphome-xiaomi_bslamp2@dev
INFO Updating https://github.com/mmakaay/esphome-xiaomi_bslamp2@dev
INFO Deleting /data/build/lr-bedside-lamp/.pioenvs
INFO Deleting /data/build/lr-bedside-lamp/.piolibdeps
INFO Done!
```

```
INFO ESPHome 2023.11.2
INFO Reading configuration /config/esphome/lr-bedside-lamp.yaml...
INFO Updating https://github.com/mmakaay/esphome-xiaomi_bslamp2@dev
INFO Generating C++ source...
INFO Compiling app...
Processing lr-bedside-lamp (board: esp32doit-devkit-v1; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.2") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/data/build/lr-bedside-lamp/.pioenvs/lr-bedside-lamp/CMakeFiles/CMakeOutput.log".
See also "/data/build/lr-bedside-lamp/.pioenvs/lr-bedside-lamp/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /data/build/lr-bedside-lamp/.pioenvs/lr-bedside-lamp/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_9d86a && [1/2] Building C object CMakeFiles/cmTC_9d86a.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_9d86a
    FAILED: cmTC_9d86a 
    : && /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_9d86a.dir/testCCompiler.c.obj  -o cmTC_9d86a   && :
    xtensa-esp32-elf-gcc: internal compiler error: Segmentation fault signal terminated program collect2
    Please submit a full bug report,
    with preprocessed source if appropriate.
    See <https://gcc.gnu.org/bugs/> for instructions.
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:298 (__project)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 1.96 seconds ==========================
```

szymucha94, 2023-11-18:
okay, I just reinstalled esphome and it's now compiling. Very strange issue.

closed 2023-11-18 by szymucha94