esphome/issues #5130: esp8266: Socket could not enable tcp nodelay, errno: 104

angmac, 2023-11-18:
### The problem

Only on the esp8266, the esp32 is not affected: 10 minutes after reboot, a bunch of warning appear when I run the "esphome logs" command:

INFO ESPHome 2023.11.2
INFO Reading configuration dolgozo-lampa.yaml...
INFO Starting log output from 192.168.2.10 using esphome API
INFO Successfully connected to 192.168.2.10 in 0.005s
INFO Successful handshake with 192.168.2.10 in 3.150s
[21:41:53][I][app:102]: ESPHome version 2023.11.2 compiled on Nov 18 2023, 21:36:50
[21:41:53][C][wifi:559]: WiFi:
[21:41:53][C][wifi:391]:   Local MAC: DC:4F:22:C7:13:4B
[21:41:53][C][wifi:396]:   SSID: 't2ha'
[21:41:53][C][wifi:397]:   IP Address: 192.168.2.10
[21:41:53][C][wifi:398]:   BSSID: FC:5C:45:7F:AD:A1
[21:41:53][C][wifi:400]:   Hostname: 'dolgozo-lampa'
[21:41:53][C][wifi:402]:   Signal strength: -34 dB ▂▄▆█
[21:41:53][C][wifi:406]:   Channel: 6
[21:41:53][C][wifi:407]:   Subnet: 255.255.255.0
[21:41:53][C][wifi:408]:   Gateway: 192.168.2.1
[21:41:53][C][wifi:409]:   DNS1: 192.168.2.1
[21:41:53][C][wifi:410]:   DNS2: 0.0.0.0
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][C][logger:416]: Logger:
[21:41:53][C][logger:417]:   Level: DEBUG
[21:41:53][C][logger:418]:   Log Baud Rate: 115200
[21:41:53][C][logger:420]:   Hardware UART: UART0
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][C][gpio.output:010]: GPIO Binary Output:
[21:41:53][C][gpio.output:011]:   Pin: GPIO12
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][C][gpio.output:010]: GPIO Binary Output:
[21:41:53][C][gpio.output:011]:   Pin: GPIO13
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'button_1'
[21:41:53][C][gpio.binary_sensor:016]:   Pin: GPIO0
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][C][light:103]: Light 'dolgozó lámpa'
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][C][light:103]: Light 'led'
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][C][restart:068]: Restart Switch 'dolgozó lámpa Restart'
[21:41:53][C][restart:070]:   Icon: 'mdi:restart'
[21:41:53][C][restart:090]:   Restore Mode: always OFF
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][C][captive_portal:088]: Captive Portal:
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][C][mdns:115]: mDNS:
[21:41:53][C][mdns:116]:   Hostname: dolgozo-lampa
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][C][ota:097]: Over-The-Air Updates:
[21:41:53][C][ota:098]:   Address: 192.168.2.10:8266
[21:41:53][C][ota:101]:   Using Password.
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][C][api:139]: API Server:
[21:41:53][C][api:140]:   Address: 192.168.2.10:6053
[21:41:53][C][api:142]:   Using noise encryption: YES
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104ms that issue #3693 is the same. There were no solution there.
[21:41:53][W][ota:147]: Socket could not enable tcp nodelay, errno: 104
...

- Using the esphome device from home assistant works
- OTA upgrade fails. I can OTA upgrade immediately after reboot.
- It seems that issue #3693 is the same. There were no solution there.

### Which version of ESPHome has the issue?

2023.11.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2023.11.2

### What platform are you using?

ESP8266

### Board

Sonoff touch, Sonoff 4ch, nodemcu

### Example YAML snippet

```yaml
base.yaml:
-----------------------------
wifi:
  power_save_mode: none
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: ${ip}
  manual_ip:
    static_ip: ${ip}
    gateway: !secret gateway
    subnet: !secret subnet
    dns1: !secret dns1
  # Enable fallback hotspot (captive portal) in case wifi connection fails  
  ap:
    ssid: "${device}"
    password: !secret fallback_password

captive_portal:

switch:
  - platform: restart
    name: "${device_name} Restart"
    id: reset
-----------------------------

dolgozo-lampa.yaml:
-----------------------------
substitutions:
  ip: 192.168.2.10
  device_name: "dolgozó lámpa"
  device: "dolgozo-lampa"

esphome:
  name: ${device}

esp8266:
  board: esp01_1m

api:
  encryption:
    key: *** MY KEY ***

logger:

ota:
  password: *** MY PASSWORD ***

packages:
  base: !include base.yaml

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    id: button_1
    on_press:
      then:
        - light.toggle: light_1

output:
  - platform: gpio
    pin: GPIO12
    id: relay_1
  - platform: gpio
    pin: GPIO13
    id: led_relay_1

light:
  - platform: binary
    name: "${device_name}"
    id: light_1
    output: relay_1
    on_turn_on:
      - light.turn_on: led_1
    on_turn_off:
      - light.turn_off: led_1
  - platform: binary
    name: "led"
    id: led_1
    internal: True
    output: led_relay_1
-----------------------------
```

angmac, 2023-11-18:
Python 3.11.5, Arch linux, Python venv install

bbccdd, 2023-12-20:
Same problem here.
Dozens of lines of this error every second and failing to control the device or update using OTA.

```
16:40:48 | [W] | [ota:147] | Socket could not enable tcp nodelay, errno: 104
16:40:50 | [W] | [ota:147] | Socket could not enable tcp nodelay, errno: 104
```

A clue, anyone?

angmac, 2023-12-21:
Seems 2023.11.6 solved the issue for me.

bbccdd, 2023-12-22:
Just refreshed with 2023.12.3 but the issue remains. It is on a shelly 1 btw.

hahlic, 2023-12-24:
It's happens to me too. Only help is to disconnect power and reconnect. My suspicion is on sketch update but not sure.
When it's happen all my ESPs start to warning.

angmac, 2023-12-25:
Some additional information:

- esphome version 2023.11.6; device firmware 2023.11.6 and never try to connect to the device with different version of esphome: seems good
- esphome version 2023.12.5; device firmware 2023.11.6: problem occured
- esphome version 2023.11.6 (downgraded); device firmware 2023.11.6 and tried to connect connect to it with different esphome version before: problem occured
- After reset the device: seems good
- esphome version: 2023.12.5; device firmware 2023.12.5: seems good

It seems to me the problem occured only when I try to connect to the esp8266 device which has different firmware version than the esphome version. This is normal on esphome upgrade. After this connection the device remain in bad state until reset.