esphome/issues #5132: Can't install OTA to device after update to 2023.11.2 docker

nabheet, 2023-11-19:
### The problem

I am getting the following error when trying to update OTA:

```
Successfully installed cachecontrol-0.13.1 certifi-2023.11.17 cffi-1.16.0 charset-normalizer-3.3.2 click-8.1.7 colorama-0.4.6 contextlib2-21.6.0 cryptography-3.4.8 filelock-3.13.1 future-0.18.3 idf-component-manager-1.4.1 idna-3.4 kconfiglib-13.7.1 msgpack-1.0.7 packaging-23.2 pycparser-2.21 pyparsing-2.3.1 pyyaml-6.0.1 requests-2.31.0 requests-file-1.5.1 requests-toolbelt-1.0.0 schema-0.7.5 six-1.16.0 tqdm-4.66.1 urllib3-1.26.18
Reading CMake configuration...
Error: Couldn't find target config target-_project_elf_src-e2b2d9bb6bbfee578368.json
========================= [FAILED] Took 18.98 seconds =========================
```

and:

```
Wrote 0xd4ed0 bytes to file /config/.esphome/build/sunroom/.pioenvs/sunroom/firmware-factory.bin, ready to flash to offset 0x0
======================== [SUCCESS] Took 820.70 seconds ========================
Error: Couldn't find target config target-_project_elf_src-fdeedeb809cb748ec373.json
========================== [FAILED] Took 2.36 seconds ==========================
ERROR Could not match idedata, please report this error
ERROR Stdout: Processing sunroom (board: nodemcu-32s; framework: espidf; platform: platformio/espressif32@5.4.0)
```

### Which version of ESPHome has the issue?

2023.11.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

stabe

### What platform are you using?

ESP32-IDF

### Board

nodemcu

### Component causing the issue

Not sure

### Example YAML snippet

```yaml
esphome:
  name: garage

esp32:
  board: nodemcu-32s
  framework:
    type: esp-idf

# Enable logging
logger:
#  level: INFO

api:
  encryption:
    key: !secret encryption_key

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
  ap:
    ssid: "Garage Fallback Hotspot"
    password: "PASSWORD"

switch:
  - platform: gpio
    name: Garage Door Controller
    id: garage_door_controller
    internal: True
    pin: 13
  - platform: shutdown
    name: "Garage Door Shutdown"

button:
  - platform: shutdown
    id: garage_door_controller_shutdown
    name: "Garage Door Controller Shutdown"
  
binary_sensor:
  - platform: gpio
    name: Garage Wall Switch
    id: garage_wall_switch
    internal: True
    pin:
      number: 27
      inverted: True
      mode:
        input: True
        pullup: True
    on_release:
      then:
        - cover.open: garage_door
    on_multi_click:
    - timing:
        - ON for at least 10s
      then:
        - button.press: garage_door_controller_shutdown

  - platform: gpio
    name: Garage Door Top End Stop
    id: garage_door_stop_end_stop
    internal: True
    pin:
      number: 26
      inverted: True
      mode:
        input: True
        pullup: True
    
cover:
  - platform: template
    name: "Garage Door"
    id: garage_door
    device_class: garage
    entity_category: config
    optimistic: True
    lambda: |-
      if (id(garage_door_stop_end_stop).state) {
        return COVER_OPEN;
      } else {
        return COVER_CLOSED;
      }
    open_action:
      - switch.turn_on: garage_door_controller
      - delay: 0.5s
      - switch.turn_off: garage_door_controller

    close_action:
      - switch.turn_on: garage_door_controller
      - delay: 0.5s
      - switch.turn_off: garage_door_controller

    stop_action:
      - switch.turn_on: garage_door_controller
      - delay: 0.5s
      - switch.turn_off: garage_door_controller

sensor:
  - platform: internal_temperature
    name: "Temperature"
```

nabheet, 2023-11-19:
BTW - versio 2023.10 works just fine.

Kwiatjk, 2023-11-20:
I have that same problem back to 2023.10

nabheet, 2023-12-03:
Now even 2023.10 is broken for me and I have no idea why. And the frustrating thing is that the latest version worked this morning for one my garage.yaml. But it didn't work for any others. Then it is not working on garage.yaml either. 😰

nabheet, 2023-12-03:
Actually, garage worked again: 

```
INFO Successfully compiled program.
INFO Resolving IP address of garage.local
INFO  -> 192.168.86.60
INFO Uploading .esphome/build/garage/.pioenvs/garage/firmware.bin (797616 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from garage.local using esphome API
WARNING Can't connect to ESPHome API for garage: Error resolving IP address: [Errno -3] Temporary failure in name resolution (APIConnectionError)
INFO Trying to connect to garage in the background
INFO Successfully connected to garage in 0.021s
INFO Successful handshake with garage in 0.153s
[23:10:36][I][app:102]: ESPHome version 2023.11.6 compiled on Dec  3 2023, 22:46:31
[23:10:36][C][wifi:559]: WiFi:

```

nabheet, 2023-12-03:
And this one failed:

```
uilding .pioenvs/sunroom/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/sunroom/firmware.bin"], [".pioenvs/sunroom/firmware.elf"])
Wrote 0xd4ed0 bytes to file /config/.esphome/build/sunroom/.pioenvs/sunroom/firmware-factory.bin, ready to flash to offset 0x0
=============================================================== [SUCCESS] Took 353.87 seconds ===============================================================
Error: Couldn't find target config target-_project_elf_src-e2ea3826d3b64a93059b.json
================================================================= [FAILED] Took 1.16 seconds =================================================================
ERROR Could not match idedata, please report this error
ERROR Stdout: Processing sunroom (board: nodemcu-32s; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 (5.4.0) > NodeMCU-32S
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
```

nabheet, 2023-12-04:
And framework: arduino just worked with the same sunroom.yaml. I am so confused.

nabheet, 2023-12-18:
This is still a problem with version 2023.11.6.

```
Creating esp32c3 image...
Successfully created esp32c3 image.
esp32_create_combined_bin([".pioenvs/sunroom-top-end-stop/firmware.bin"], [".pioenvs/sunroom-top-end-stop/firmware.elf"])
Wrote 0xdb800 bytes to file /config/.esphome/build/sunroom-top-end-stop/.pioenvs/sunroom-top-end-stop/firmware-factory.bin, ready to flash to offset 0x0
=============================================================== [SUCCESS] Took 367.13 seconds ===============================================================
Error: Couldn't find target config target-_project_elf_src-cd8227038e57213fe0cd.json
================================================================= [FAILED] Took 1.46 seconds =================================================================
ERROR Could not match idedata, please report this error
ERROR Stdout: Processing sunroom-top-end-stop (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (5.4.0) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40400.0 (4.4.0) 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Reading CMake configuration...

ERROR 
```