esphome/issues #5135: esphome does not generate unique IDs

fcorrao1, 2023-11-20:
### The problem

I have an air gradient air quality sensor which has a name and friendly_name of ag1.  It contains a pms5003 particulate matter sensor which provides pm1.0, pm2.5 and pm10 values. There seems to be a clash between the ID’s that are chosen for the pm1.0 and pm10 entities.  The names for the pm1.0 and pm10 entities are getting munged such that they are identical.  This creates a conflict which prevents one of them from being used.  Deleting and re-adding the entities and device doesn't seem to resolve the issue.

Log message:

ERROR (MainThread) [homeassistant.components.sensor] Platform esphome does not generate unique IDs. ID 84:F3:EB:16:77:BC-sensor-pm_10m is already used by sensor.ag1_pm_10m - ignoring sensor.ag1_pm_10m

Relevant sections of ag1’s config:

```
esphome:
  name: ag1
  friendly_name: ag1

sensor:
  - platform: pmsx003
    type: PMSX003
    uart_id: uartuno
    pm_1_0:
      id: pm11
      name: "pm 1.0µm"
    pm_2_5:
      id: pm251
      name: "pm 2.5µm"
    pm_10_0:
      id: pm101
      name: "pm 10µm"
    update_interval: 120s
```

### Which version of ESPHome has the issue?

2023.11.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.2

### What platform are you using?

ESP8266

### Board

d1_mini

### Component causing the issue

pmsx003

bdraco, 2023-11-20:
We discussed this on discord. It looks like the `.` is dropped which causes the issue

https://ptb.discord.com/channels/429907082951524364/524177279270780928/1176277404558303262

Jesse suggested replacing the `.` with a `_` to avoid the conflict

jesserockz, 2023-11-20:
My suggestion was more around altering the name sanitizing code to replace any `.` with `_`

bdraco, 2023-11-20:
> My suggestion was more around altering the name sanitizing code to replace any `.` with `_`

We are on the same page, I just did a bad job of explaining the intent above

closed 2023-11-23 by bdraco