esphome/issues #5136: MQTT Client with default config for HomeAssistant broken in ESPHome 2023.11.1

KidA001, 2023-11-21:
### The problem

Using basic MQTT Client config for HomeAssistant as described [here](https://esphome.io/components/mqtt.html#:~:text=.-,Using%20with%20Home%20Assistant) causes ESPHome to throw an error when compiling `error: 'const struct esp_mqtt_event_t' has no member named 'qos' qos(event.qos)` (Full stack trace below)

### Which version of ESPHome has the issue?

2023.11.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.1

### What platform are you using?

ESP32

### Board

ESP32S3

### Component causing the issue

MQTT

### Example YAML snippet

```yaml
Using basic MQTT Config. This config has not changed and has been used in previous versions without issue

mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_un
  password: !secret mqtt_pw
```

### Anything in the logs that might be useful for us?

```txt
Full ESP YAML

esphome:
  name: propane-tank-main-house
  friendly_name: Propane Tank Main House
  on_boot:
    then:
      - output.turn_on: pwr_pin
      - delay: 1s
  on_shutdown:
    then:
      - output.turn_off: pwr_pin
      - logger.log: "Goodnight sweet prince"

esp32:
  board: esp32-s3-devkitc-1
  variant: ESP32S3
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 5.1.1

# Deep sleep
deep_sleep:
  run_duration: 10s
  sleep_duration: 6h

# Enable logging
logger:

ota:
  password: redacted

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  power_save_mode: none

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Propane-Tank-Main-House"
    password: redacted

captive_portal:

mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_un
  password: !secret mqtt_pw

output:
  - platform: gpio
    pin: GPIO35
    id: pwr_pin

    
sensor:
  - platform: adc
    pin: GPIO10
    name: "Remote Hall"
    update_interval: 2s
    attenuation: 11db
    unit_of_measurement: "%"
    filters:
      # (𝑛 − min / max−min) × 100
      # full: 1.48
      # empty: 1.9
      - lambda: return ((1.48 - 1.9) / -0.419999 ) * 100;
```

### Additional information

Full stack trace
```
Compiling .pioenvs/water-tank-sensor-v3/src/esphome/components/mqtt/custom_mqtt_device.cpp.o
Compiling .pioenvs/water-tank-sensor-v3/src/esphome/components/mqtt/mqtt_backend_esp32.cpp.o
Compiling .pioenvs/water-tank-sensor-v3/src/esphome/components/mqtt/mqtt_client.cpp.o
Compiling .pioenvs/water-tank-sensor-v3/src/esphome/components/mqtt/mqtt_component.cpp.o
In file included from src/esphome/components/mqtt/mqtt_backend_esp32.cpp:4:
src/esphome/components/mqtt/mqtt_backend_esp32.h: In constructor 'esphome::mqtt::Event::Event(const esp_mqtt_event_t&)':
src/esphome/components/mqtt/mqtt_backend_esp32.h:38:19: error: 'const struct esp_mqtt_event_t' has no member named 'qos'
         qos(event.qos),
                   ^~~
src/esphome/components/mqtt/mqtt_backend_esp32.h:39:19: error: 'const struct esp_mqtt_event_t' has no member named 'dup'
         dup(event.dup),
                   ^~~
*** [.pioenvs/water-tank-sensor-v3/src/esphome/components/mqtt/mqtt_backend_esp32.cpp.o] Error 1
In file included from src/esphome/components/mqtt/mqtt_client.h:13,
                 from src/esphome/components/mqtt/custom_mqtt_device.h:7,
                 from src/esphome/components/mqtt/custom_mqtt_device.cpp:1:
src/esphome/components/mqtt/mqtt_backend_esp32.h: In constructor 'esphome::mqtt::Event::Event(const esp_mqtt_event_t&)':
src/esphome/components/mqtt/mqtt_backend_esp32.h:38:19: error: 'const struct esp_mqtt_event_t' has no member named 'qos'
         qos(event.qos),
                   ^~~
src/esphome/components/mqtt/mqtt_backend_esp32.h:39:19: error: 'const struct esp_mqtt_event_t' has no member named 'dup'
         dup(event.dup),
                   ^~~
In file included from src/esphome/components/mqtt/mqtt_client.h:13,
                 from src/esphome/components/mqtt/mqtt_component.h:11,
                 from src/esphome/components/mqtt/mqtt_component.cpp:1:
src/esphome/components/mqtt/mqtt_backend_esp32.h: In constructor 'esphome::mqtt::Event::Event(const esp_mqtt_event_t&)':
src/esphome/components/mqtt/mqtt_backend_esp32.h:38:19: error: 'const struct esp_mqtt_event_t' has no member named 'qos'
         qos(event.qos),
                   ^~~
src/esphome/components/mqtt/mqtt_backend_esp32.h:39:19: error: 'const struct esp_mqtt_event_t' has no member named 'dup'
         dup(event.dup),
                   ^~~
In file included from src/esphome/components/mqtt/mqtt_client.h:13,
                 from src/esphome/components/mqtt/mqtt_client.cpp:1:
src/esphome/components/mqtt/mqtt_backend_esp32.h: In constructor 'esphome::mqtt::Event::Event(const esp_mqtt_event_t&)':
src/esphome/components/mqtt/mqtt_backend_esp32.h:38:19: error: 'const struct esp_mqtt_event_t' has no member named 'qos'
         qos(event.qos),
                   ^~~
src/esphome/components/mqtt/mqtt_backend_esp32.h:39:19: error: 'const struct esp_mqtt_event_t' has no member named 'dup'
         dup(event.dup),
                   ^~~
*** [.pioenvs/water-tank-sensor-v3/src/esphome/components/mqtt/custom_mqtt_device.cpp.o] Error 1
*** [.pioenvs/water-tank-sensor-v3/src/esphome/components/mqtt/mqtt_component.cpp.o] Error 1
*** [.pioenvs/water-tank-sensor-v3/src/esphome/components/mqtt/mqtt_client.cpp.o] Error 1
========================== [FAILED] Took 4.12 seconds ==========================
```

ssieb, 2023-11-21:
Do you have a `custom_components` directory?  I tested compiling and there's no issue.

KidA001, 2023-11-21:
> Do you have a `custom_components` directory? I tested compiling and there's no issue.

Not in the `config/esphome` dir, but I do have `config/custom_components` - would that effect ESPHome?

KidA001, 2023-11-21:
Seems like this issue was specific to the board settings I was using, I updated to:
```yaml
esp32:
  board: esp32-s3-devkitc-1
  variant: ESP32S3
  framework:
    type: arduino
    version: 2.0.5
    platform_version: 5.2.0
```

And that fixed my issues

closed 2023-11-21 by KidA001