esphome/issues #5137: problems flashing my lights that have an esp01 and don't have USB

compujunk, 2023-11-21:
### The problem

I have the following problem, I have about 30 WiFi lamps with an esp01 in them and ESPHome on them. Now a few things have changed in the config and I want to flash the new firmware on the lamps, but I keep getting the message:

"ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB"

The problem is that I don't have a USB connection on those lamps, so opening them and then flashing them via serial is not an option either.
How can I still flash these lamps with the current ESPHome? Previously I never had this problem with those lamps.
My Sonoff Basics also have that message, but I can still flash them via serial, even though they are often in hard-to-reach places.

### Which version of ESPHome has the issue?

2023.11.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.3

### What platform are you using?

ESP8266

### Board

esp01

### Example YAML snippet

```yaml
substitutions:
  devicename: eettafellamp1
  friendly_name: eettafellamp1

packages:
  device_base: !include common/device_base.yaml

esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m
  build_path: ./build/${devicename}

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
  ap:
    ssid: "Fallback $devicename"
    password: !secret wifi_password

captive_portal:

web_server:
 port: 80

light:
  - platform: cwww
    name: $friendly_name
    id: $devicename
    cold_white: output_component1
    warm_white: output_component2
    cold_white_color_temperature: 3500 K
    warm_white_color_temperature: 2000 K

output:
  - platform: esp8266_pwm
    id: output_component1
    pin: 14
  - platform: esp8266_pwm
    id: output_component2
    pin: 12

sensor:
  - platform: homeassistant
    entity_id: sensor.licht_sensor
    id: licht_sensor
    on_value_range:
        - above: 1000
          then:
            - light.turn_on: $devicename
        - below: 950
          then:
            - light.turn_off: $devicename
```

ssieb, 2023-11-22:
How did you initially flash them and have you ever updated them?

compujunk, 2023-11-23:
[image]
I just flash them over ota from the esphome in HA
And yes, I have re-fed them before. But I only do that if there are real changes in my yaml file, not with every esphome update

ssieb, 2023-11-23:
Have you ever changed the board setting?
What happens if you download the file and use the device web server to upload it?

compujunk, 2023-11-23:
do you mean "Have you ever changed the board setting?" this sometimes:

esphome:
   name: $devicename
   platform: ESP8266
   board: esp01_1m
yes, as it was in the beginning, but have been using this method on these devices for a long time.

If I upload the bin file via the device's web server, I get the following
[image]

ssieb, 2024-02-10:
That looks like the firmware file was compiled expecting 4MB of flash, but that doesn't make sense.  Try the latest release and see if that makes any difference.
Can you flash one using serial and see if that fixes it?

compujunk, 2024-02-11:
they are lights so I can't access them serially because I can't open them.
I had the same problem with my Sonoff's basic, but I was able to re-flash them via serial and they now work well again, but with these lamps that is not possible and since there are about 30 lamps, that is a problem. They are lamps from the LSC brand with an old Tuya chip (ESP8266) that I had also hacked via OTA at the time and installed ESP on it.

but I tried again with this result,

Via ESPHome itself: ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB

Via the web interface: Update Failed: ERROR[8]: Flash config wrong real: 1048576 IDE: 4194304

still the same errors.

compujunk, 2024-04-26:
In other words, I can no longer update about 30 lamps and so I am getting rid of them because you have changed something that means I can no longer update? I think that is a bad thing and hope that you will look further for a solution.

ssieb, 2024-04-26:
We didn't change anything.  There was a bug in the Arduino core.