esphome/issues #5138: Non-functional ENT key on Keypad

stomko11, 2023-11-21:
### The problem

Hello. I have wiegand keypad that, aside from 0-9 keys, has ENT and ESC keys which are recognized as keys 10 and 11.
My config is as follows:
`
key_collector:
  - id: wg_pincode
    source_id: wggate
    min_length: 4
    max_length: 4
    end_keys: "#"
    end_key_required: true
    back_keys: "*"
    clear_keys: "C"
    allowed_keys: "0123456789"
    timeout: 5s
`

While ESC works as "clear" key while I am already in the sequence to enter PIN, ENT does not confirm anything. I have tried defining `end_keys: "10"` as well, but that does not seem to work either.

How can I define it to be able to use it as enter key?
Funny enough - if I flip definition, ESC does not work for accepting the key either. It simply always clears, and ENT does nothing.
If I am not in sequence for entering the pin, both just get identified as keys 10 and 11

### Which version of ESPHome has the issue?

2023.11

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11

### What platform are you using?

ESP8266

### Component causing the issue

key_collector

ssieb, 2023-11-21:
They are functional since as you say they are getting recognized. 10 and 11 are `*` and `#`.
If you have more questions like this, come ask on discord.

stomko11, 2023-11-21:
I would if I knew how to.
Anyhow, I must have been doing something wrong before, because I have tested now and it works as expected. I apologize for unnecessary post.

closed 2023-11-21 by stomko11