esphome/issues #5139: M5Stack ATOM Echo media player stops working, reboots

savage7, 2023-11-21:
### The problem

I've flashed the media player template from https://esphome.io/projects/?type=media
The media playback works but stops after some playbacks. The device restart after this playback is possible again, but stops working again after 2-3 playbacks.

### Which version of ESPHome has the issue?

2023.11.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.2

### What platform are you using?

ESP32

### Board

M5Stack ATOM Echo

### Component causing the issue

i2s_audio

### Example YAML snippet

```yaml
Nothing has been changed from the template.
```

### Anything in the logs that might be useful for us?

```txt
[19:36:14][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.54 s).
[19:36:14][W][component:215]: Components should block for at most 20-30ms.
[19:36:17][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.37 s).
[19:36:17][W][component:215]: Components should block for at most 20-30ms.
[19:36:18][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.36 s).
[19:36:18][W][component:215]: Components should block for at most 20-30ms.
[19:40:59][I][ota:117]: Boot seems successful, resetting boot loop counter.
[19:40:59][D][esp32.preferences:114]: Saving 1 preferences to flash...
[19:40:59][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
```

SanderKo85, 2023-12-05:
Hi!

Just received two Atom Echo M5's, and I have almost exactly the same problem, also at Playback, but after using it to control the lights. After two or three "confirms" the whole thing locks up.

super-qua, 2023-12-06:
> Hi!
> 
> Just received two Atom Echo M5's, and I have almost exactly the same problem, also at Playback, but after using it to control the lights. After two or three "confirms" the whole thing locks up.

Same for me, but only on one device. Didn't find time to check the logs so far