esphome/issues #5141: Connection trigger compile error on old Arduino framework (2023.11 regression)

TheJulianJES, 2023-11-22:
### The problem

Using Arduino framework 2.7.4 causes a compile error in the api component for the newly added connection triggers.
See the provided YAML example and the logs below.

Issue was **not** present on 2023.10.x and earlier. Seems to be introduced in 2023.11 with:
- https://github.com/esphome/esphome/pull/5628 

(FastLED requires Arduino framework 2.7.4 on ESP8266, as mentioned in the docs.)

### Which version of ESPHome has the issue?

2023.11.0 and 2023.11.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

NodeMCU

### Component causing the issue

api

### Example YAML snippet

```yaml
esphome:
  name: temptest
  friendly_name: temptest

esp8266:
  board: nodemcuv2
  framework:
    version: 2.7.4 # Check https://esphome.io/components/light/fastled.html later

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "u4NLNSYdNMIOc9/i399FqkrdkZVigfn/Rqr1fFQrLxg="

ota:
  password: "7dd38b241e389af140fbcc0b57008c55"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
In file included from src/esphome/components/api/api_connection.h:6:0,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/api_server.h:118:96: error: invalid use of '::'
   Trigger<std::string, std::string> *client_connected_trigger_ = new Trigger<std::string, std::string>();
                                                                                                ^
src/esphome/components/api/api_server.h:118:96: error: expected ';' at end of member declaration
src/esphome/components/api/api_server.h:118:102: error: expected unqualified-id before '>' token
   Trigger<std::string, std::string> *client_connected_trigger_ = new Trigger<std::string, std::string>();
                                                                                                      ^
src/esphome/components/api/api_server.h:119:99: error: invalid use of '::'
   Trigger<std::string, std::string> *client_disconnected_trigger_ = new Trigger<std::string, std::string>();
                                                                                                   ^
src/esphome/components/api/api_server.h:119:99: error: expected ';' at end of member declaration
src/esphome/components/api/api_server.h:119:105: error: expected unqualified-id before '>' token
   Trigger<std::string, std::string> *client_disconnected_trigger_ = new Trigger<std::string, std::string>();
                                                                                                         ^
src/esphome/components/api/api_server.h:118:83: error: template argument 1 is invalid
   Trigger<std::string, std::string> *client_connected_trigger_ = new Trigger<std::string, std::string>();
                                                                                   ^
src/esphome/components/api/api_server.h:119:86: error: template argument 1 is invalid
   Trigger<std::string, std::string> *client_disconnected_trigger_ = new Trigger<std::string, std::string>();
                                                                                      ^
Compiling .pioenvs/temptest/src/esphome/components/api/api_server.cpp.o
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [.pioenvs/temptest/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from src/esphome/components/api/api_server.cpp:1:0:
src/esphome/components/api/api_server.h:118:96: error: invalid use of '::'
   Trigger<std::string, std::string> *client_connected_trigger_ = new Trigger<std::string, std::string>();
                                                                                                ^
src/esphome/components/api/api_server.h:118:96: error: expected ';' at end of member declaration
src/esphome/components/api/api_server.h:118:102: error: expected unqualified-id before '>' token
   Trigger<std::string, std::string> *client_connected_trigger_ = new Trigger<std::string, std::string>();
                                                                                                      ^
src/esphome/components/api/api_server.h:119:99: error: invalid use of '::'
   Trigger<std::string, std::string> *client_disconnected_trigger_ = new Trigger<std::string, std::string>();
                                                                                                   ^
src/esphome/components/api/api_server.h:119:99: error: expected ';' at end of member declaration
src/esphome/components/api/api_server.h:119:105: error: expected unqualified-id before '>' token
   Trigger<std::string, std::string> *client_disconnected_trigger_ = new Trigger<std::string, std::string>();
                                                                                                         ^
src/esphome/components/api/api_server.h:118:83: error: template argument 1 is invalid
   Trigger<std::string, std::string> *client_connected_trigger_ = new Trigger<std::string, std::string>();
                                                                                   ^
src/esphome/components/api/api_server.h:119:86: error: template argument 1 is invalid
   Trigger<std::string, std::string> *client_disconnected_trigger_ = new Trigger<std::string, std::string>();
                                                                                      ^
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [.pioenvs/temptest/src/esphome/components/api/api_server.cpp.o] Error 1
========================== [FAILED] Took 2.62 seconds ==========================
```

ssieb, 2023-11-22:
Using an old arduino framework version is not supported as mentioned in the compile log.  There is some work to get a newer fastled library, but until then, try using neopixel instead.

closed 2023-11-22 by ssieb

TheJulianJES, 2023-11-22:
The docs say to use Arduino framework version 2.7.4 for FastLED, so it should still be supported.
Just some weeks ago another compile error was solved (that only happened on 2.7.4): https://github.com/esphome/issues/issues/5044

It also looks like the [linked PR](https://github.com/esphome/esphome/pull/5628) broke support by accident. It doesn't look intentional to me.
I'd understand if FastLED support is removed completely for ESP8266 and the docs are updated to reflect this, but at the moment, this looks like a bug.

TheJulianJES, 2023-11-23:
@jesserockz Since the connection triggers PR [#5628](https://github.com/esphome/esphome/pull/5628) is yours, can you say if this is expected to break with the older framework version required for FastLED?

I'm not exactly sure why, but it seems like the template parameter pack used in the automation `Trigger` isn't supported with two string parameters on that version.

If this breakage is expected, let me know and I'll PR an update to [the docs](https://esphome.io/components/light/fastled.html) to remove the mention of the Arduino 2.7.4 framework version. I think this just causes confusion if it's supported according to the docs, but not supported in reality.

Ideally, the change should also be mentioned in the 2023.11.0 breaking changes section then (that FastLED is "removed" for ESP8266).

---

For anyone else finding this issue, there's an effort made in this PR to update FastLED (and seemingly no longer require an old framework version):
- https://github.com/esphome/esphome/pull/4257

ssieb, 2023-11-23:
I suspect the problem is that the old Arduino version also pulls in an older compiler.
I've delayed updating the docs because of the update PR, but it might have been better to have marked it as not working since there have been several issues caused by this.
Is there a reason you don't want to use the neopixel component instead?

TheJulianJES, 2023-11-23:
> I've delayed updating the docs because of the update PR

Ah, I see. But yeah, even if it's finished soon, it might be a good idea to add a note/warning in the docs for now like you said.

> old Arduino version also pulls in an older compiler.

Hmm, I wonder if C++11 features aren't even supported then. I'll check later what version that is.

>  reason you don't want to use the neopixel component instead?

The main reason is probably that I'm using some custom FastLED effects (which I've shared somewhere some years back).
They directly use FastLED commands. I'll probably just switch over anyway. They're outdated too, but kept working.

The introduced issue also looked to be easily fixable at first, so I took a shot but couldn't figure it out on the older version. The new connection triggers are the only thing that cause the compilation error on 2023.11.x. When removed, it compiles at least.

darox, 2023-12-11:
I see the same error on one device.

heizie, 2024-01-01:
in case someone just want to solve this without specific requirement of the version.
just change `version` to `recommended` in the config. e.g.
```
esp8266:
  board: YOURBOARD
  framework:
    version: recommended
```

ssieb, 2024-01-01:
That's the default, so it's the same as not specifying anything at all.