esphome/issues #5142: Unable to compile esphome via docker run

roofburner, 2023-11-22:
### The problem

Hi all,

I try to "compile" ESPHome 2023.11.2 using docker (Ubuntu Server on Dell hardware).

This is what I execute:
```
docker pull esphome/esphome
docker run --rm -v /usr/share/esphome:/config -it esphome/esphome run ./relay.yml
```

Please advise...

Same issue here. I use docker run to upgrade my esp board.

I followed the guide below the error, but no luck. Reboot did not help either.

```
INFO ESPHome 2023.11.1
INFO Reading configuration ./relay.yml...
INFO Generating C++ source...
INFO Compiling app...
Processing heatingrelais (board: esp12e; framework: arduino; platform: platformio/espressif8266@3.2.0)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
RuntimeError: Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/platformio/__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "/usr/local/lib/python3.11/dist-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/run/cli.py", line 145, in cli
    process_env(
  File "/usr/local/lib/python3.11/dist-packages/platformio/run/cli.py", line 208, in process_env
    ).process()
      ^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/run/processor.py", line 90, in process
    ).run(build_vars, build_targets, self.silent, self.verbose, self.jobs)
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/platform/_run.py", line 64, in run
    result = self._run_scons(variables, targets, jobs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/platform/_run.py", line 117, in _run_scons
    stdout=proc.BuildAsyncPipe(
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/proc.py", line 65, in __init__
    super().__init__()
  File "/usr/local/lib/python3.11/dist-packages/platformio/proc.py", line 38, in __init__
    self._thread.start()
  File "/usr/lib/python3.11/threading.py", line 957, in start
    _start_new_thread(self._bootstrap, ())
RuntimeError: can't start new thread

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

Kind regards

### Which version of ESPHome has the issue?

2023.11.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

esp12e

ssieb, 2023-11-22:
What version of docker do you have?

ssieb, 2023-11-23:
Try upgrading docker.  It appears to be a docker issue.

ssieb, 2023-11-23:
Or possibly add `privileged: true` to the docker compose config.

roofburner, 2023-11-23:
I have upgraded docker to 24.0.6. Compile was successful. Thanks @ssieb!

closed 2023-11-23 by ssieb