esphome/issues #5144: Hidden SSIDs do not work on LibreTiny Devices

rishabmehta7, 2023-11-23:
### The problem

Beken devices do not connect to Hidden SSIDs, even after them being specified. The same config works fine on ESP32 and ESP8266.

### Which version of ESPHome has the issue?

2023.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.3

### What platform are you using?

BK72XX

### Component causing the issue

WiFi

### Example YAML snippet

```yaml
wifi:
  networks:
  - ssid: "SSID1"
    password: !secret wifi_password
    priority: 0.0
  - ssid: "SSID2"
    password: !secret wifi_password
    priority: 1.0
  - ssid: "SSID3"
    password: !secret wifi_password
    priority: 2.0
    hidden: True
  use_address: xxx
  ap:
    ssid: "xxx"
    password: !secret wifi_password
```

ssieb, 2023-11-23:
Does it work if it's the only ssid?

rishabmehta7, 2023-11-23:
> Does it work if it's the only ssid?

No just tried that as well.

ssieb, 2024-03-25:
Can you try asking the libretiny project either in github or discord about this?