esphome/issues #5148: Unable to use font in esphome addon installed inside HAAS OS

meteosat007, 2023-11-23:
### The problem

I have searched all of the issues and the internet and I see the issue been a problem for docker install, yet on the esphome docs for font, it states that if esphome is added as an add-on, the pillow should be installed and correct.

### Which version of ESPHome has the issue?

Latest DEV version and also I had latest prod version

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.3

### What platform are you using?

ESP32

### Board

ESPDEV Board

### Component causing the issue

font

### Example YAML snippet

```yaml
font:
  - file:
      type: gfonts
      family: Roboto
      weight: 900
    id: font2
    size: 16
display:
  - platform: waveshare_epaper
    id: eink_display
    cs_pin: 15
    dc_pin: 27
    busy_pin: 25
    reset_pin: 26
    model: 7.50inV2alt
    update_interval: never
    reset_duration: 2ms
    lambda: |-
      auto red = Color(255, 0, 0);
      auto green = Color(0, 255, 0);
      auto blue = Color(0, 0, 255);
      auto white = Color(255, 255, 255);
      it.rectangle(20, 50, 30, 30, white);
      it.rectangle(25, 55, 30, 30, red);
      it.rectangle(30, 60, 30, 30, green);
      it.rectangle(35, 65, 30, 30, blue);
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.12.0-dev
INFO Reading configuration /config/esphome/e-weather.yaml...
INFO Detected timezone 'Europe/London'
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

font: [source /config/esphome/e-weather.yaml:690]
  
  Please downgrade your pillow installation to below 10.0.0. (pip install pillow">4.0.0,<10.0.0").
  - file: 
      type: gfonts
      family: Roboto
      weight: 900
    id: font2
    size: 16
```

### Additional information

I have tried everything and cannot now create any displays using font due to this error.

ssieb, 2023-11-24:
Do you have a `custom_components` directory?  That version check was changed in 2023.10 so there's no way you should be seeing that in dev.

meteosat007, 2023-11-24:
I solved the problem by adding a custom_components folder and after that I can again use font without any issues.
Using the normal standard installation of DEV or production I would get the error.
Happy with the work around as it took a lage many hours to resolve the problem.

closed 2023-11-24 by meteosat007

ssieb, 2023-11-24:
You added that folder?  What did you put in it?  That's likely to cause you more issues later.